openwrt/mt76 #572: [mt7613] Archer A6 V3 cannot have two SSIDs running on 5 GHz radio

graysky2, 2021-09-01:
The Archer A6 V3 is unable to have more than a single SSID up on the 5 GHz radio (MeidaTek MT7613BE).  Bringing up a 2nd one causes both to crash.

Initially, both SSIDs are defined but disabled.  Here is `logread -f` as I bring up one of them:
```
Wed Sep  1 07:32:57 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Wed Sep  1 07:32:58 2021 kern.info kernel: [   70.229448] br-lan: port 7(wlan1) entered blocking state
Wed Sep  1 07:32:58 2021 kern.info kernel: [   70.234758] br-lan: port 7(wlan1) entered disabled state
Wed Sep  1 07:32:58 2021 kern.info kernel: [   70.241214] device wlan1 entered promiscuous mode
Wed Sep  1 07:32:58 2021 kern.info kernel: [   70.343127] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Wed Sep  1 07:32:58 2021 kern.info kernel: [   70.349914] br-lan: port 7(wlan1) entered blocking state
Wed Sep  1 07:32:58 2021 kern.info kernel: [   70.355309] br-lan: port 7(wlan1) entered forwarding state
Wed Sep  1 07:32:58 2021 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->ENABLED
Wed Sep  1 07:32:58 2021 daemon.notice hostapd: wlan1: AP-ENABLED
Wed Sep  1 07:32:59 2021 daemon.notice netifd: Network device 'wlan1' link is up
```

And here it is bringing up the 2nd one where the error is triggered:
```
Wed Sep  1 07:33:23 2021 daemon.notice hostapd: Remove interface 'wlan1'
Wed Sep  1 07:33:23 2021 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Wed Sep  1 07:33:23 2021 daemon.notice hostapd: wlan1: AP-DISABLED
Wed Sep  1 07:33:23 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Wed Sep  1 07:33:23 2021 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Wed Sep  1 07:33:23 2021 kern.info kernel: [   95.303050] device wlan1 left promiscuous mode
Wed Sep  1 07:33:23 2021 kern.info kernel: [   95.307762] br-lan: port 7(wlan1) entered disabled state
Wed Sep  1 07:33:23 2021 daemon.notice netifd: Network device 'wlan1' link is down
Wed Sep  1 07:33:24 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.357077] br-lan: port 7(wlan1) entered blocking state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.362404] br-lan: port 7(wlan1) entered disabled state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.368526] device wlan1 entered promiscuous mode
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.373699] br-lan: port 7(wlan1) entered blocking state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.379105] br-lan: port 7(wlan1) entered forwarding state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.406385] br-lan: port 7(wlan1) entered disabled state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.476646] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.483252] br-lan: port 7(wlan1) entered blocking state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.488615] br-lan: port 7(wlan1) entered forwarding state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.503044] br-lan: port 8(wlan1-1) entered blocking state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.508606] br-lan: port 8(wlan1-1) entered disabled state
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.514756] device wlan1-1 entered promiscuous mode
Wed Sep  1 07:33:24 2021 daemon.err hostapd: Could not set interface wlan1-1 flags (UP): No such file or directory
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.606855] device wlan1-1 left promiscuous mode
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.611511] br-lan: port 8(wlan1-1) entered disabled state
Wed Sep  1 07:33:24 2021 daemon.err hostapd: Failed to add BSS (BSSID=cc:32:e5:01:eb:ba)
Wed Sep  1 07:33:24 2021 daemon.err hostapd: Interface initialization failed
Wed Sep  1 07:33:24 2021 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->DISABLED
Wed Sep  1 07:33:24 2021 daemon.notice hostapd: wlan1: AP-DISABLED
Wed Sep  1 07:33:24 2021 daemon.err hostapd: wlan1: Unable to setup interface.
Wed Sep  1 07:33:24 2021 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.853087] device wlan1 left promiscuous mode
Wed Sep  1 07:33:24 2021 kern.info kernel: [   96.857968] br-lan: port 7(wlan1) entered disabled state
Wed Sep  1 07:33:24 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Wed Sep  1 07:33:24 2021 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Wed Sep  1 07:33:24 2021 daemon.notice hostapd: wlan1-1: CTRL-EVENT-TERMINATING
Wed Sep  1 07:33:24 2021 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1-1 wasn't started
Wed Sep  1 07:33:24 2021 daemon.notice netifd: radio1 (3569): Command failed: Invalid argument
Wed Sep  1 07:33:24 2021 daemon.notice netifd: radio1 (3569): Device setup failed: HOSTAPD_START_FAILED
```

I am happy to provide additional info/log upon request.

I also found https://github.com/openwrt/mt76/issues/518 but I am not seeing the `daemon.err hostapd: Could not set interface wlan1-1 flags (UP): No such file or directory` there so I do not know if this is related.

ptpt52, 2021-09-06:
Please try this:
```
rm /lib/firmware/mt7663_n9_v3.bin
rm /lib/firmware/mt7663pr2h.bin
```
and reboot
this may fix the issue?

graysky2, 2021-09-06:
Interestingly, I do not have them to begin with:
```
# ls -l /lib/firmware/
drwxr-xr-x    2 root     root           185 Aug 31 18:20 mediatek
-rwxr-xr-x    1 root     root         64132 Aug 31 18:20 mt7603_e1.bin
-rwxr-xr-x    1 root     root         72180 Aug 31 18:20 mt7603_e2.bin
-rw-r--r--    1 root     root          4000 Aug 31 18:20 regulatory.db
```

ptpt52, 2021-09-06:
update path.. /lib/firmware/mediatek/

graysky2, 2021-09-06:
Amazing.  Removing those two files and rebooting fixed the issue.  Will I have any missing features without those files?

ptpt52, 2021-09-06:
package issue in openwrt, it should not use `kmod-mt7663-firmware-sta`, and instead use `kmod-mt7663-firmware-ap`, but by default the device select both, and the mt76 driver use the sta one first.

graysky2, 2021-09-06:
Can we open an issues against the correct package/packages to fix this?  I think this would affect other users.

graysky2, 2021-09-06:
Thanks very much @ptpt52 :heart: 
Removing `kmod-mt7663-firmware-sta` followed by a reboot fixed this for me.  I will close this issue.

closed 2021-09-06 by graysky2

Borromini, 2021-09-08:
Nice catch @ptpt52. I grepped the tree and (at least in 21.02) it looks like it's just two devices pulling in the mt7663-firmware-sta package. I checked my EAP235-Wall v1 and that uses mt7663-firmware-ap like it's supposed to.