openwrt/mt76 #574: MT7615dn:  5G  max power is 6dBm

siwind, 2021-09-05:
This router is Raisecom MSG1500 X.00,  which is customized by China Telecom.
SOC  is  mediatek mt7621,  wifi is  a single MT7615DN chipset which provides 2.4G & 5G communication.
The hardware of 5G has a LAN and PA.

After installing openwrt master  firmware,  5G configuation shows that max power is  5dBm.

Following is the openwrt version:

```
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.142 #0 SMP Tue Aug 31 14:15:59 2021 mips GNU/Linux
root@OpenWrt:~# opkg list | grep mt76
kmod-mt76-connac - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt76-core - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt7615-common - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt7615-firmware - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt7615e - 5.4.142+2021-07-15-bbebea7d-4
root@OpenWrt:~# 
```

Borromini, 2021-09-05:
Is this a NAND device? If it has bad blocks radio firmware might have been relocated. That would explain the low power.

siwind, 2021-09-05:
> Is this a NAND device? If it has bad blocks radio firmware might have been relocated. That would explain the low power.

Yes,  it has a  128MB nand flash,  and it has not bad blocks in firmware.  
I think that the problem may be caused by mt-76 driver?

siwind, 2021-09-07:
> Is this a NAND device? If it has bad blocks radio firmware might have been relocated. That would explain the low power.

I change to another stock's  eeprom file, and it show normal now.  
Thanks for your help!

Airomeo, 2021-09-07:
@siwind I have the same problem, can you share your EEPROM?

DragonBluep, 2022-04-27:
https://github.com/openwrt/mt76/issues/633#issuecomment-1107465562 After my test, the output power is normal, just the displayed value is not correct.