openwrt/mt76 #576: 2.4GHz WiFi disappears after seeing mt7603_filter_tx and ieee80211_ocb_leave warnings

easyteacher, 2021-09-06:
I have two ssids on the 2.4G WiFi, and they both disappeared after seeing these warnings. I have disabled a ssid to see if this bug still happens.

#488 #551 #560 

```
# opkg info kmod-mt76
Package: kmod-mt76
Version: 5.10.52+2021-07-15-bbebea7d-4
Depends: kernel (= 5.10.52-1-7653798df7a28770a4a905db429ff7bb), kmod-mt76-core, kmod-mt76x2, kmod-mt7603
Status: install user installed
Architecture: mipsel_24kc
Installed-Time: 1627246808
# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r17216+1-8c2509dc5f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r17216+1-8c2509dc5f'
DISTRIB_TAINTS='no-all busybox override'
```

> [272540.999849] ------------[ cut here ]------------
> [272541.004626] WARNING: CPU: 3 PID: 1068 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-07-15-bbebea7d/mt7603/mac.c:206 mt7603_filter_tx+0x16c/0x174 [mt7603e]
> [272541.019741] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 lz4 libchacha20poly1305 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211 cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota2 xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_ipp2p xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm tcp_bbr slhc sch_cake poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_flow_table nf_conntrack_netlink nf_conncount macvlan lz4_decompress lz4_compress
> [272541.020662]  libcurve25519_generic libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables compat chacha_mips fuse sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_csum libcrc32c act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4
> [272541.108931]  ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel tun nls_utf8 nls_cp950 nls_cp936 zram zsmalloc sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 mbcache jbd2 usbcore nls_base usb_common mii crc32c_generic crc32_generic
> [272541.248108] CPU: 3 PID: 1068 Comm: napi/phy0-7 Not tainted 5.10.52 #0
> [272541.254651] Stack : 80756cb0 839a0000 00000001 80081c5c 80860000 80757c10 00000000 00000000
> [272541.263142]         82f79c64 809b0000 81c261c8 807ecde7 80725f68 00000001 82f79c08 ff4e4248
> [272541.271605]         00000000 00000000 80725f68 82f79aa8 ffffefff 808567e4 00000000 00000000
> [272541.280061]         00000000 000034f7 00000000 0003c92c 00000000 80860000 00000000 80720000
> [272541.288515]         00000009 00000005 00000005 839a0000 00000000 803ef7dc 0000000c 809b000c
> [272541.296970]         ...
> [272541.299508] Call Trace:
> [272541.302070] [<80008108>] show_stack+0x30/0x100
> [272541.306604] [<80372bac>] dump_stack+0xa4/0xe0
> [272541.311061] [<8002d884>] __warn+0xc0/0xe8
> [272541.315151] [<8002d908>] warn_slowpath_fmt+0x5c/0xac
> [272541.320237] [<839944ec>] mt7603_filter_tx+0x16c/0x174 [mt7603e]
> [272541.326244] [<83994688>] mt7603_wtbl_set_ps+0x130/0x138 [mt7603e]
> [272541.332420] [<83991a34>] mt7603_sta_ps+0x38/0x430 [mt7603e]
> [272541.338098] [<839b4744>] mt76_rx_poll_complete+0x2cc/0x59c [mt76]
> [272541.344280] [<839b15b8>] mt76_dma_rx_poll+0x3c4/0xd7c [mt76]
> [272541.350056] [<804d7f8c>] __napi_poll+0x3c/0x154
> [272541.354669] [<804d8164>] napi_threaded_poll+0xc0/0x128
> [272541.359888] [<8004d478>] kthread+0x13c/0x144
> [272541.364239] [<80003038>] ret_from_kernel_thread+0x14/0x1c
> [272541.369709] 
> [272541.371442] ---[ end trace 77f95f2224e36093 ]---
> [284968.476930] sched: RT throttling activated
> [434259.443402] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> [434259.448982] mt76x2e 0000:01:00.0: Build: 1
> [434259.453533] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [434259.482091] mt76x2e 0000:01:00.0: Firmware running!
> [434259.492362] ieee80211 phy1: Hardware restart was requested
> [700144.654471] ------------[ cut here ]------------
> [700144.659652] WARNING: CPU: 1 PID: 1097 at backports-5.10.42-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x55c/0x63c [mac80211]
> [700144.671922] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 lz4 libchacha20poly1305 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211 cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota2 xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_ipp2p xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm tcp_bbr slhc sch_cake poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_flow_table nf_conntrack_netlink nf_conncount macvlan lz4_decompress lz4_compress
> [700144.672950]  libcurve25519_generic libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables compat chacha_mips fuse sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_csum libcrc32c act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4
> [700144.761858]  ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel tun nls_utf8 nls_cp950 nls_cp936 zram zsmalloc sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 mbcache jbd2 usbcore nls_base usb_common mii crc32c_generic crc32_generic
> [700144.902295] CPU: 1 PID: 1097 Comm: mt76-tx phy1 Tainted: G        W         5.10.52 #0
> [700144.910316] Stack : 80756cb0 00000002 00000000 80081c5c 80860000 80757c10 00000000 00000000
> [700144.918824]         83ab9c94 809b0000 816ab2c8 807ecde7 80725f68 00000001 83ab9c38 ff4e4248
[... cut ...]
> [700144.944279]         00000009 00000014 00000004 00000002 00000000 803ef7dc 00000004 809b0004
> [700144.952790]         ...
> [700144.955345] Call Trace:
> [700144.957908] [<80008108>] show_stack+0x30/0x100
> [700144.962496] [<80372bac>] dump_stack+0xa4/0xe0
> [700144.967001] [<8002d884>] __warn+0xc0/0xe8
> [700144.971101] [<8002d908>] warn_slowpath_fmt+0x5c/0xac
> [700144.976323] [<83959380>] ieee80211_ocb_leave+0x55c/0x63c [mac80211]
> [700144.982788] 
> [700144.984724] ---[ end trace 77f95f2224e36094 ]---

easyteacher, 2021-09-12:
https://github.com/openwrt/mt76/commit/fc31457cd99cb85c8cea9329eedc5edd80038f29

Call trace leads to this commit. I am wondering if mt7603e does not handle SMPS well.

easyteacher, 2021-09-16:
I run a loop script to continuously toggle power saving status on my laptop WiFi card,

```
CCOUNT=1
while :
do
iwconfig wlp2s0 power on
sleep 1
iwconfig wlp2s0 power off
echo On/Off Cycle ${CCOUNT}
sleep 1
CCOUNT=$((${CCOUNT} + 1))
done
```

and monitor `STA-OPMODE-SMPS-MODE-CHANGED` on the router side using `logread -e SMPS -f`, which makes me find something interesting.

If something bad is going to happen, hostapd will not log `STA-OPMODE-SMPS-MODE-CHANGED` message in time when running the loop script, even when the client is still connected to the 2.4G AP.

My assumption is that sometimes the PS status of a client is not updated successfully, and the router side still keeps the old PS status, which could potentially make WiFi connection unstable. So setting htmode to Legacy/None to disable SMPS feature introduced by 802.11n could really help to solve the issue.

easyteacher, 2021-09-17:
I tried this patch and 2.4G AP is very stable now, except #26.

```
diff --git a/mt7603/main.c b/mt7603/main.c
index 8edea1e7..c72264ef 100644
--- a/mt7603/main.c
+++ b/mt7603/main.c
@@ -633,7 +633,7 @@ mt7603_sta_rate_tbl_update(struct ieee80211_hw *hw, struct ieee80211_vif *vif,
 	mt7603_wtbl_set_rates(dev, msta, NULL, msta->rates);
 	msta->rate_probe = false;
 	mt7603_wtbl_set_smps(dev, msta,
-			     sta->smps_mode == IEEE80211_SMPS_DYNAMIC);
+			     1 == 0);
 	spin_unlock_bh(&dev->mt76.lock);
 }
```

But as long as 2.4G is mostly for smart home devices, it's worth to revert https://github.com/openwrt/mt76/commit/fc31457cd99cb85c8cea9329eedc5edd80038f29 to get a stable AP.

easyteacher, 2021-09-20:
As mt7615d is not affected by this bug, I am going to throw away mt7603e for less suffering.

JFtico, 2021-09-27:
@easyteacher Which of the above commits gave the best results?

I understand you are moving on, but some of us are stuck with a bunch of units with this problem.

Thanks In Advance

easyteacher, 2021-09-28:
> @easyteacher Which of the above commits gave the best results?
> 
> I understand you are moving on, but some of us are stuck with a bunch of units with this problem.
> 
> Thanks In Advance

Please try this.

https://github.com/openwrt/mt76/pull/583

DragonBluep, 2021-10-30:
[FS#873 - problems of mt7628 wifi driver on miwifi nano](
https://bugs.openwrt.org/index.php?do=details&task_id=873&string=mt7628&search_name=&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)
[FS#3325 - MT7628an: lose radio whilst downloading jarge data](
https://bugs.openwrt.org/index.php?do=details&task_id=3325&string=7628&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)
It seems that this problem has existed for a long time. When using a laptop for high-load transmission like testing net speed, it is easy to cause problems. It has been verified on the RTL8822CE and 7265AC rev C wireless network cards. 

> > @easyteacher Which of the above commits gave the best results?
> > I understand you are moving on, but some of us are stuck with a bunch of units with this problem.
> > Thanks In Advance
> 
> Please try this.
> 
> #583

Thank you for trying to fix it, but it seems that the effect is not obvious and the problem still exists.

easyteacher, 2021-10-30:
> Thank you for trying to fix it, but it seems that the effect is not obvious and the problem still exists.

It could be a hardware defect. I don't remember if the closed-source driver has the same problem.

DragonBluep, 2021-11-11:
There are some closed source driver in coolsnowwolf's repository hasfor openwrt, which is worth a try.