openwrt/mt76 #578: Mt76x0e low txpower (Tplink Archer c20 V5)

newmem12, 2021-09-09:
I just installed openwrt 21.02 for Tplink Archer C20 v5. But the maximum txpower is only 14dBm and the 5G signal does not seem to be as strong as the original firmware. This device uses the MT76x0e 5G chip. Anyone have the same situation as me?

sanitariu, 2021-09-09:
Yes everyone has the same. I think it is reading wrong data from the eeprom. You can manually edit your file:
./build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2021-07-15-bbebea7d/mt76x0/eeprom.c
Find this: val = mt76x02_eeprom_get(dev, 0x132);
and change to val = 23;
I hope it helps.

newmem12, 2021-09-09:
Thank you, let me try again

stm32repo, 2021-09-27:
`[   45.689094] WARNING: CPU: 0 PID: 725 at backports-5.10.68-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x578/0x658 [mac80211]
[   45.701016] Modules linked in: rt2800soc rt2800mmio rt2800lib pppoe ppp_async mt76x0e mt76x0_common iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD rt2x00soc rt2x00mmio rt2x00lib pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio ohci_platform ohci_hcd rtl8367b rtl8366_smi fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[   45.772487] CPU: 0 PID: 725 Comm: mt76-tx phy0 Not tainted 5.4.143 #0
[   45.779070] Stack : 80600000 8059ddd8 00000000 00000000 8059ced0 83041cb4 83d9aee0 805d6d63
[   45.787614]         80548cd4 000002d5 807333bc 00000014 00000004 00000001 83041c68 0bfd8c34
[   45.796759]         00000000 00000000 80760000 00000000 30232033 000000e1 65746e69 2e352064
[   45.805380]         00000000 00000002 00000000 000bc987 00000000 00000009 00000000 83556170
[   45.813927]         00000009 00000014 00000004 00000002 00000000 802a656c 00000000 80730000
[   45.822472]         ...
[   45.824966] Call Trace:
[   45.827499] [<80009c90>] show_stack+0x30/0x100
[   45.832059] [<80023bb0>] __warn+0xc0/0x10c
[   45.836257] [<80023c58>] warn_slowpath_fmt+0x5c/0xac
[   45.841493] [<83556170>] ieee80211_ocb_leave+0x578/0x658 [mac80211]
[   45.847986] ---[ end trace 00077af99fb32f36 ]---`