openwrt/mt76 #579: driver mt76x0e crash

sanitariu, 2021-09-09:
openwrt crash:

[82649.095410] ------------[ cut here ]------------
[82649.100146] WARNING: CPU: 0 PID: 684 at backports-5.10.42-1/net/mac80211/airtime.c:456 0x83155944 [mac80211@fd8c66ac+0x61320]
[82649.111646] Modules linked in: pppoe ppp_async mt76x0e mt76x0_common iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt compat asn1_decoder nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic
[82649.111826]  gf128mul gcm ctr cmac ccm leds_gpio gpio_button_hotplug aead cryptomgr crypto_null crypto_hash
[82649.210553] CPU: 0 PID: 684 Comm: mt76-tx phy1 Not tainted 5.4.143 #0
[82649.217088] Stack : 00000000 80058154 80490000 8049698c 00000000 00000000 00000000 00000000
[82649.225571]         00000000 00000000 00000000 00000000 00000000 00000001 82825c68 00000000
[82649.234043]         82825d00 00000000 00000000 00000000 00000038 804321c4 65746e69 2e352064
[82649.242575]         00000000 82825ba5 ffffffff ffffffff 82825c48 00000000 00000000 83155944
[82649.251072]         00000009 00000014 00000004 00000002 00000000 8025efc0 00000000 80600000
[82649.259553]         ...
[82649.262028] Call Trace:
[82649.262038] [<80058154>] 0x80058154
[82649.268084] [<804321c4>] 0x804321c4
[82649.271644] [<83155944>] 0x83155944 [mac80211@fd8c66ac+0x61320]
[82649.277652] [<8025efc0>] 0x8025efc0
[82649.281200] [<800095e8>] 0x800095e8
[82649.284731] [<800095f0>] 0x800095f0
[82649.288319] [<80022a18>] 0x80022a18
[82649.291867] [<83155944>] 0x83155944 [mac80211@fd8c66ac+0x61320]
[82649.297890] [<80022ac0>] 0x80022ac0
[82649.301457] [<83155944>] 0x83155944 [mac80211@fd8c66ac+0x61320]
[82649.307493] [<83103410>] 0x83103410 [mac80211@fd8c66ac+0x61320]
[82649.313498] [<83155b7c>] 0x83155b7c [mac80211@fd8c66ac+0x61320]
[82649.319517] [<831363a8>] 0x831363a8 [mac80211@fd8c66ac+0x61320]
[82649.325553] [<83155f34>] 0x83155f34 [mac80211@fd8c66ac+0x61320]
[82649.331627] [<830f2970>] 0x830f2970 [mt76x02_lib@48eec363+0x92e0]
[82649.337964] [<832d0488>] 0x832d0488 [mt76@a27b359b+0x91e0]
[82649.343537] [<830f4efc>] 0x830f4efc [mt76x02_lib@48eec363+0x92e0]
[82649.349745] [<832d0488>] 0x832d0488 [mt76@a27b359b+0x91e0]
[82649.355311] [<80433618>] 0x80433618
[82649.358884] [<832d0518>] 0x832d0518 [mt76@a27b359b+0x91e0]
[82649.364447] [<804338d0>] 0x804338d0
[82649.368030] [<8003e8a0>] 0x8003e8a0
[82649.371578] [<8003e768>] 0x8003e768
[82649.375133] [<80004ed8>] 0x80004ed8
[82649.378668]
[82649.380175] ---[ end trace 2a4f7b458698aeec ]---

openwrt version:
OpenWrt SNAPSHOT, r17486-cdc646c032

newmem12, 2021-09-09:
Have you tried the 21.02 release version?

sanitariu, 2021-09-09:
I am using master branch - r17486-cdc646c032
From 3-4 days behind. I will update to latest again.... but there are no changes i think.

mrkiko, 2021-09-14:
Please try updating to mt76 HEAD.
A manual update in the mt76's openwrt package is needed at the moment.

siwind, 2021-09-25:
mt7603 alse crash  when using master code:

https://github.com/openwrt/mt76/issues/551

sanitariu, 2021-11-16:
Now at r18070-49f615022c. Mediatek 7628 does not crash but it looses wifi on heavy traffic like torrent. It is easy to test. Just connect wifi to the router. Grab some torrent and do speeds like 5-8MB/s via wifi. After 5-10 minutes you will be disconnected. WIFI ssid is not discoverable for 1 minute then you can reconnect again. There is nothing in the log files. Still lot of work must be done for mediatek.