openwrt/mt76 #588: MT7615D 5G STA scan causes 2G AP clients to disconnect.

colinw-smartrg, 2021-09-21:
**SETUP:** Running OpenWRT master on MT7621 MTK Eval board with MT7615D PCIe card.

**opkg versions**:
kmod-mt76 - 5.10.52+2021-08-13-8b3d8e2d-4
kmod-mt7615-common - 5.10.52+2021-08-13-8b3d8e2d-4
kmod-mt7615-firmware - 5.10.52+2021-08-13-8b3d8e2d-4
kmod-mt7615e - 5.10.52+2021-08-13-8b3d8e2d-4

**Configuration**
Configure one 2G interface for AP. ssid=Openwrt
Configure one 5G interface for STA to ssid=none, network=uplink
Configure both with:
encryption='psk2+ccmp'
key='12345678'
Connect laptop client to the AP, Openwrt.

**Results**
Laptop client disconnects and reconnects often when 5G scan happens.
Scan can be when Searching for the ssid or manually initiated with:
wpa_cli scan

logread shows the client connecting and disconnecting:
This repeats over and over.

> Tue Sep 21 21:35:38 2021 daemon.warn wpa_supplicant[1717]: wlan1: Failed to initiate sched scan
> Tue Sep 21 21:35:41 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:24:d7:77:10:50
> Tue Sep 21 21:35:41 2021 daemon.info hostapd: wlan0: STA 00:24:d7:77:10:50 IEEE 802.11: authenticated
> Tue Sep 21 21:35:41 2021 daemon.info hostapd: wlan0: STA 00:24:d7:77:10:50 IEEE 802.11: associated (aid 1)
> Tue Sep 21 21:35:41 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:24:d7:77:10:50
> Tue Sep 21 21:35:41 2021 daemon.info hostapd: wlan0: STA 00:24:d7:77:10:50 RADIUS: starting accounting session DA5BC1E503855455
> Tue Sep 21 21:35:41 2021 daemon.info hostapd: wlan0: STA 00:24:d7:77:10:50 WPA: pairwise key handshake completed (RSN)
> Tue Sep 21 21:35:41 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 00:24:d7:77:10:50

If I disable the wpa network
wpa_cli disable_network 0
The laptop client connection becomes stable.

Initiating a manual scan sometimes causes a disconnect.
wpa_cli scan
If the client disconnects, we get the same log as above.

Re-enabling the network
wpa_cli enable_network 0
Causes the wpa log message:
> Tue Sep 21 21:35:38 2021 daemon.warn wpa_supplicant[1717]: wlan1: Failed to initiate sched scan
to repeat every 5 seconds 
After about 1 minute, the 2G station begins disconnecting and repeating the logs shown above.

**Client side log**
Laptop 2G Client wpa_cli shows these messages when 5G is scanning.

> <3>CTRL-EVENT-SCAN-STARTED 
> <3>CTRL-EVENT-SCAN-RESULTS 
> <3>WPS-AP-AVAILABLE 
> <3>SME: Trying to authenticate with 00:0c:43:28:a6:d6 (SSID='OpenWrt' freq=2412 MHz)
> <3>Trying to associate with 00:0c:43:28:a6:d6 (SSID='OpenWrt' freq=2412 MHz)
> <3>Associated with 00:0c:43:28:a6:d6
> <3>CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
> <3>WPA: Key negotiation completed with 00:0c:43:28:a6:d6 [PTK=CCMP GTK=CCMP]
> <3>CTRL-EVENT-CONNECTED - Connection to 00:0c:43:28:a6:d6 completed [id=0 id_str=]
> <3>CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-26 noise=9999 txrate=6000
> <3>CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-26 noise=9999 txrate=6000
> <3>CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-26 noise=9999 txrate=6000
> <3>CTRL-EVENT-BEACON-LOSS 
> <3>CTRL-EVENT-DISCONNECTED bssid=00:0c:43:28:a6:d6 reason=4 locally_generated=1
> <3>CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
> <3>CTRL-EVENT-SCAN-STARTED 
> <3>CTRL-EVENT-SCAN-RESULTS 
> <3>WPS-AP-AVAILABLE 
> <3>SME: Trying to authenticate with 00:0c:43:28:a6:d6 (SSID='OpenWrt' freq=2412 MHz)
> <3>Trying to associate with 00:0c:43:28:a6:d6 (SSID='OpenWrt' freq=2412 MHz)
> <3>Associated with 00:0c:43:28:a6:d6
> <3>CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
> <3>WPA: Key negotiation completed with 00:0c:43:28:a6:d6 [PTK=CCMP GTK=CCMP]
> <3>CTRL-EVENT-CONNECTED - Connection to 00:0c:43:28:a6:d6 completed [id=0 id_str=]
> <3>CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-33 noise=9999 txrate=6000
> <3>CTRL-EVENT-BEACON-LOSS 
> <3>CTRL-EVENT-DISCONNECTED bssid=00:0c:43:28:a6:d6 reason=4 locally_generated=1
> <3>CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
> <3>CTRL-EVENT-SCAN-STARTED 
> <3>CTRL-EVENT-SCAN-RESULTS 
> <3>WPS-AP-AVAILABLE 
> <3>SME: Trying to authenticate with 00:0c:43:28:a6:d6 (SSID='OpenWrt' freq=2412 MHz)
> <3>Trying to associate with 00:0c:43:28:a6:d6 (SSID='OpenWrt' freq=2412 MHz)
> <3>Associated with 00:0c:43:28:a6:d6
> <3>CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
> <3>WPA: Key negotiation completed with 00:0c:43:28:a6:d6 [PTK=CCMP GTK=CCMP]
> <3>CTRL-EVENT-CONNECTED - Connection to 00:0c:43:28:a6:d6 completed [id=0 id_str=]
> <3>CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-33 noise=9999 txrate=6000
>

ryderlee1110, 2021-09-22:
I don't have 7615D right now, but some others can do -  2.4g AP (wlan0) pings a connected station without problems while band1 station "iw dev wlan1 scan". Will this problem only happens on wpa_cli scan?

colinw-smartrg, 2021-09-22:
I am also seeing the disconnection issue on the 2.4G with 
iw wlan1 scan
Not every time, but about every other time.

ptpt52, 2021-09-23:
mark