openwrt/mt76 #590: Mesh (802.11s) on MT7622AV [BPI-R64]

JonnyTischbein, 2021-09-23:
Hi,

I'm having trouble using 802.11s on my Banana Pi R64. After setting everything up, I do not see my Beacons with the same or other devices (another BPI R64 and multiple laptops). There is no `dmesg` log, warn or error.

My image was build using the current commit [da5bb885e17cf77caea70adcf473f1fb95448553](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=da5bb885e17cf77caea70adcf473f1fb95448553 ) of the openwrt git.
Connecting to an AP works fine, as well as joining an AP network.

My configs are:
```
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/18000000.wmac'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option country 'DE'
	option cell_density '2'
	option beacon_int '15'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option mode 'mesh'
	option encryption 'none'
	option mesh_id 'Avarange-Mesh'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
       option network 'mesh_network1'
```
and the network:

```
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd1e:8d69:e16e::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'

config interface 'mesh_network1'
	option proto 'static'
	option ipaddr '10.44.0.3'
	option netmask '255.255.255.0'
```

In the [banana pi forum](https://forum.banana-pi.org/t/bpi-r64-mesh-802-11s-on-internal-wifi-card-mt7622av/12610/4) the question came up, wether the chip does support 802.11s at all.

`iw` shows "mesh point", which should indicate, that it is supported. On the other hand the [mediatek site ](https://www.mediatek.com/products/homeNetworking/mt7622) only states 802.11b/g/n. (I'm not sure if the s is integrated in b/g/n or something)

```
root@OpenWrt:~# iw phy
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0xf RX 0xf
	Configured Antennas: TX 0xf RX 0xf
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-31
		Frequencies:
			* 2412 MHz [1] (6.0 dBm)
			* 2417 MHz [2] (6.0 dBm)
			* 2422 MHz [3] (6.0 dBm)
			* 2427 MHz [4] (6.0 dBm)
			* 2432 MHz [5] (6.0 dBm)
			* 2437 MHz [6] (6.0 dBm)
			* 2442 MHz [7] (6.0 dBm)
			* 2447 MHz [8] (6.0 dBm)
			* 2452 MHz [9] (6.0 dBm)
			* 2457 MHz [10] (6.0 dBm)
			* 2462 MHz [11] (6.0 dBm)
			* 2467 MHz [12] (6.0 dBm)
			* 2472 MHz [13] (6.0 dBm)
			* 2484 MHz [14] (disabled)
	valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 16,
		   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: 0
	max scan plan iterations: 0
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

Can you help me ?
Is 802.11s supported or did I miss something in my configuration / image building ?

Thanks! :)

PolynomialDivision, 2021-09-28:
I can also reproduce it.

PolynomialDivision, 2021-10-05:
I get following error: `kern.warn kernel: [ 1039.736745] netlink: 'iw': attribute type 302 has an invalid length.` Both adhoc and 802.11s.

closed 2021-11-25 by nbd168