openwrt/mt76 #594: crash log for mt7915

ptpt52, 2021-09-27:
```
<1>[615520.974224] CPU 3 Unable to handle kernel paging request at virtual address 0000003f, epc == 8dcd7140, ra == 8ddab670
<4>[615520.984927] Oops[#1]:
<4>[615520.987290] CPU: 3 PID: 1717 Comm: kworker/u9:3 Not tainted 5.4.143 #0
<4>[615520.993915] Workqueue: napi_workq napi_workfn
<4>[615520.998351] $ 0   : 00000000 00000001 00000000 8dccb250
<4>[615521.003654] $ 4   : 8dcc9f80 00000000 8e965d88 8e965d88
<4>[615521.008959] $ 8   : 8e2a3840 00000001 8dccb250 00000064
<4>[615521.014256] $12   : 80720000 00000003 ffffe000 00000040
<4>[615521.019555] $16   : 89823028 8dcc9f80 89823028 8f2e98a8
<4>[615521.024856] $20   : 00000000 00000002 8ddaf884 8dccc150
<4>[615521.030156] $24   : 8b63c130 8dcd01a4                  
<4>[615521.035456] $28   : 8e964000 8e965d20 00000000 8ddab670
<4>[615521.040755] Hi    : 000000da
<4>[615521.043709] Lo    : 00000000
<4>[615521.046697] epc   : 8dcd7140 mt76_tx_status_skb_done+0x0/0x90 [mt76]
<4>[615521.053150] ra    : 8ddab670 mt7915_queue_rx_skb+0xc0/0x1528 [mt7915e]
<4>[615521.059742] Status: 11007c03	KERNEL EXL IE 
<4>[615521.064003] Cause : 40800008 (ExcCode 02)
<4>[615521.068075] BadVA : 0000003f
<4>[615521.071023] PrId  : 0001992f (MIPS 1004Kc)
<4>[615521.075179] Modules linked in: pppoe ppp_async batman_adv wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_hybla tcp_bbr slhc sch_cake poly1305_mips nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp
<4>[615521.075358]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libblake2s_generic ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables hwmon crc_ccitt compat_xtables compat chacha_mips br_netfilter asn1_decoder natflow natcap sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port
<4>[615521.162719]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ip6_gre ip_gre gre ifb nat46 ip6_udp_tunnel udp_tunnel sit ip6_tunnel tunnel6 tunnel4 ip_tunnel tun nls_utf8 nls_iso8859_1 nls_base sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ecb ctr cmac ccm arc4 leds_gpio softdog gpio_button_hotplug crc32c_generic
<4>[615521.304734] Process kworker/u9:3 (pid: 1717, threadinfo=fd0c0a6a, task=9390efda, tls=00000000)
<4>[615521.313397] Stack : 00000001 00000000 8dcc9f80 00000001 8d8869c0 8dccc148 00000005 8f2e9988
<4>[615521.321810]         8ddb0000 8ddaf0dc 8dccba48 00000000 8dcc9f80 00000000 000d0000 8dda0750
<4>[615521.330232]         8dcca1c0 89823000 00000800 89823000 89823000 8041bc50 00000000 00001a03
<4>[615521.338662]         00000000 00000000 8e965d88 8e965d88 00000000 89823000 8dcca250 40000000
<4>[615521.347092]         00000000 89823000 00000028 00000002 8dcca1c0 00000040 00000000 8dcd1310
<4>[615521.355505]         ...
<4>[615521.358038] Call Trace:
<4>[615521.360600] [<8dcd7140>] mt76_tx_status_skb_done+0x0/0x90 [mt76]
<4>[615521.366712] [<8ddab670>] mt7915_queue_rx_skb+0xc0/0x1528 [mt7915e]
<4>[615521.372971] [<8dcd1310>] mt76_dma_rx_poll+0x29c/0x784 [mt76]
<4>[615521.378727] [<804393dc>] __napi_poll+0x3c/0x11c
<4>[615521.383333] [<804394fc>] napi_workfn+0x40/0x90
<4>[615521.387853] [<800452d8>] process_one_work+0x214/0x470
<4>[615521.392980] [<80045b58>] worker_thread+0x194/0x5e8
<4>[615521.397864] [<8004b9bc>] kthread+0x130/0x138
<4>[615521.402220] [<800065d8>] ret_from_kernel_thread+0x14/0x1c
<4>[615521.407684] Code: 8fb00014  03e00008  27bd0028 <90a2003f> 24070003  30430003  34480002  34630002  1467001c 
<4>[615521.417485] 
<4>[615521.419354] ---[ end trace d4816f20797b77d9 ]---
```

code base is https://github.com/openwrt/mt76/commit/56b842b72a72e1ac12d306ef4a85f264a471065b

ryderlee1110, 2021-09-27:
can you run git bisect  ?

ryderlee1110, 2021-09-28:
Can you use the latest mt76 version?

ptpt52, 2021-09-28:
This happened by accident and I am testing the latest code