openwrt/mt76 #595: mt7615 driver incompatible with OpenWrt 21.02.0 release (netifd particularly)

Assader, 2021-09-27:
In every system load _except_ the first one after reflash/mtd erase only wlan0 (at phy0) is initialized.
Searching for the root cause, i found out that wireless routine for phy1 is being stopped at [find_phy](https://github.com/openwrt/openwrt/blob/v21.02.0/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh#L904) call. Placing debug call that lists /sys/class/ieee80211 directory near find_phy showed that only phy0 exist and no phy1.
I set some print debug points at the mt76 code and found out that phy1 is being created significantly later that phy0:
```
[   15.086684] mt7615e 0000:01:00.0: enabling device (0000 -> 0002)
[   15.098939] mt7615_init_hardware: calling INIT_WORK() for mt7615_pci_init_work at 14868759911 ns
[   15.139380] mt76_register_device: calling ieee80211_register_hw at 14909186535 ns
[   15.159769] mt7615_register_device: calling ieee80211_queue_work at 14929578585 ns
[   15.174977] mt7615_pci_init_work: calling mt7615_mcu_init at 14944780590 ns
[   15.204118] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   15.781773] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
[   15.881485] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
[   23.494426] mt7615_pci_init_work: mt7615_register_ext_phy at 23264215170 ns
[   23.508707] mt7615_register_ext_phy: changing mac for 5 ghz band
[   23.520806] mt76_register_phy: calling ieee80211_register_hw at 23290611515 ns
```
I guess, [_register_ext_phy](https://github.com/openwrt/mt76/blob/master/mt7615/pci_init.c#L32) must be called earlier.

Btw, i dont know how, but despite the fact that 5g mac-changing code is being executed, i have the same mac on both interfaces:
```
root@SNR-CPE-ME2-SFP-58:/# ip a
...
11: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether f8:f0:82:62:19:11 brd ff:ff:ff:ff:ff:ff
12: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether f8:f0:82:62:19:11 brd ff:ff:ff:ff:ff:ff

```
Keywords: openwrt 21, mt7615, dbdc, wifi autostart.

Assader, 2021-09-27:
Actually, yes. I just moved ext phy registration to the top of the caller function and now netifd behavior is stable.
No side effects are observed so far.
So you might want to merge [this](https://github.com/openwrt/mt76/pull/596).
The mac address issue is still persists, tho.

Assader, 2021-09-28:
Speaking about MAC addresses...
What is the point of overriding MAC for [bo](https://github.com/openwrt/mt76/blob/master/mt7615/eeprom.c#L349)-[th](https://github.com/openwrt/mt76/blob/master/mt7615/init.c#L489) phys with the _same_ value?
If i override MAC address via DTS, i expect one MAC to be i specified and the other, still, do not. Or, if mt7615 eeprom contains MAC address guaranteed, i may expect only phy1 MAC be overridden. In case if i don't want to mess OUI.
It would be nice if DTS would have `mtd-mac-address0` and `mtd-mac-address1` entries, but it has not.
Current logic is misleading and you may consider merging [this](https://github.com/openwrt/mt76/pull/597) (or do not override phy0 MAC) and do something with 7915, which situation looks alike.

nbd168, 2021-11-25:
Fixed in the latest version

closed 2021-11-25 by nbd168