openwrt/mt76 #598: broadcasts delivery problem

bol-van, 2021-09-29:
I experience the following glitch on my TP-Link Archer c6u (MT7621)
I still haven't figured out a condition to reproduce the probelm.
It just happens. In the evening it works, in the morning it doesn't. It stops working without any specific action from my side. Just leave it for enough long time and it will eventually happen.

wifi clients with power saving enabled (iw dev wlan0 set power_save on) connected to 2.4 GHz wifi (5 GHz not checked, dont have any compatible devices) stop receiving broadcast frames until they send something
broadcasts include arp and ndp frames

so, if you have a PC connected via LAN and android via wifi and they are bridged

arp -d
ping ip_of_android
<nothing>
<nothing>
<nothing>

android does not receive arps on wlan0 - checked with tcpdump in android root console
router sees outgoing arp on wlan interface without replies

but if you send something from android it starts receiving broadcasts for the short time
PC caches mac address and then ping goes normally until cache is cleaned or expired

restarting wifi interfaces does not help (openwrt command 'wifi')
only module reload helps or reboot

```
#!/bin/sh

wifi down
/etc/init.d/wpad stop
rmmod mt7603e
rmmod mt7615e
rmmod mt7615_common
rmmod mt76_connac_lib
rmmod mt76
rmmod mac80211
rmmod cfg80211
rmmod compat

kmodloader
/etc/init.d/wpad start
wifi
```

bol-van, 2021-09-29:
I tested this on 2 mediatek based androids. I'm unsure if its related to client wireless chipset.
I don't have any qualcomm based devices to test at the moment
I have a PC with wireless USB realtek adapter but PC does not power save

ryderlee1110, 2021-09-29:
Did you include this one https://patchwork.kernel.org/project/linux-wireless/patch/20210906083559.9109-1-nbd@nbd.name/ ?

bol-van, 2021-09-29:
I'm now on today's snapshot with 5.10 kernel
Yes, that patch is there

bol-van, 2021-09-30:
Typical androids always send/receive something because they are cloud connected via google apps
This may be the key factor not allowing to reproduce this issue
My both 2 androids are cleaned from google apps, do not run whatsapp and similar apps that always connected and use network.

bol-van, 2021-09-30:
I found that in "problematic" state  TX counter in "iwinfo wlan0 assoclist"
does not increase
In normal state it does increase on every sent arp

bol-van, 2021-10-10:
TPLINK c6u <> realtek 8812BU on windows pc  , using 5 ghz VHT 80

bad condition. AP client IP is 192.168.4.88
unicasts are delivered normally because pings are working if mac address is cached
but not broadcasts. after clearing neighbor cache pings not working anymore

```
root@router:~# ping 192.168.4.88
PING 192.168.4.88 (192.168.4.88) 56(84) bytes of data.
64 bytes from 192.168.4.88: icmp_seq=1 ttl=64 time=295 ms
64 bytes from 192.168.4.88: icmp_seq=2 ttl=64 time=314 ms
64 bytes from 192.168.4.88: icmp_seq=3 ttl=64 time=311 ms
64 bytes from 192.168.4.88: icmp_seq=4 ttl=64 time=329 ms
64 bytes from 192.168.4.88: icmp_seq=5 ttl=64 time=345 ms
^C
--- 192.168.4.88 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 294.801/318.808/345.332/17.048 ms

root@router:~# arping -b 192.168.4.88
ARPING 192.168.4.88 from 192.168.4.1 br-lan
^CSent 76 probes (76 broadcast(s))
Received 0 response(s)

root@router:~# ip neigh del 192.168.4.88 dev br-lan

root@router:~# ping 192.168.4.88
PING 192.168.4.88 (192.168.4.88) 56(84) bytes of data.
From 192.168.4.1 icmp_seq=1 Destination Host Unreachable
From 192.168.4.1 icmp_seq=2 Destination Host Unreachable
From 192.168.4.1 icmp_seq=3 Destination Host Unreachable
^C
```

good condition after wifi restart on the router
```
root@pest:~# arping -b 192.168.4.88
ARPING 192.168.4.88 from 192.168.4.1 br-lan
Unicast reply from 192.168.4.88 [XX:XX:XX:XX:XX:XX]  3.928ms
Unicast reply from 192.168.4.88 [XX:XX:XX:XX:XX:XX]  3.773ms
Unicast reply from 192.168.4.88 [XX:XX:XX:XX:XX:XX]  3.210ms
^CSent 3 probes (3 broadcast(s))
Received 3 response(s)

root@pest:~# ip neigh del 192.168.4.88 dev br-lan

root@pest:~# ping 192.168.4.88
PING 192.168.4.88 (192.168.4.88) 56(84) bytes of data.
64 bytes from 192.168.4.88: icmp_seq=1 ttl=128 time=10.0 ms
64 bytes from 192.168.4.88: icmp_seq=2 ttl=128 time=3.25 ms
64 bytes from 192.168.4.88: icmp_seq=3 ttl=128 time=3.66 ms
^C
--- 192.168.4.88 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 3.253/5.649/10.039/3.108 ms
```

bol-van, 2021-10-15:
I checked in monitor my mt7603 already broadcasts beacons with SMPS disabled in HT capabilities
As all of the neighbor APs do

ryderlee1110, 2021-10-22:
Does it work after enabling multicast_to_unicast ?

bol-van, 2022-11-01:
After upgrade openwrt to 22.03.02  problem seem to go away.
I haven't studied it deeply but numerous like this  

```
arp -d
ping my_device

```
cause devices to ping within seconds

rany2, 2023-06-01:
@bol-van `multicast_to_unicast` should be set on config of type wifi-iface not network device.

rany2, 2023-06-01:
also it's now `multicast_to_unicast_all` not `multicast_to_unicast`, `multicast_to_unicast` does not do anything anymore.

bol-van, 2023-06-02:
Anyway, problem is not fixed yet

teeedubb, 2025-08-26:
Hi,

I believe I am experiencing the same issue. Clients will connect to a bridged AP but will not obtain any network connectivity. Restarting network on primary router is the only thing that fixes it, and for a short time at that. Mai9n router is gl.inet MT6000 and bridged AP is a xiaomi ax3000t. Both are running 24.10 snapshot, I have also tried with master and the issue still presents itself.

Kind bummed by this bug as it really impacts my home network. Not sure how to proceed - lots of routers these days have MTK hardware and supported routers with other hardware can be difficult to find locally.