openwrt/mt76 #601: OpenWrt 21.02.0 r16279 R6220 Problem with 2.4 GHz Radio

ABusuuu, 2021-10-05:
Tue Sep 28 16:16:19 2021 daemon.notice netifd: radio0 (1899): Could not find PHY for device 'radio0'
Tue Sep 28 16:16:19 2021 daemon.notice netifd: radio0 (2019): WARNING: Variable 'data' does not exist or is not an array/object
Tue Sep 28 16:16:19 2021 daemon.notice netifd: radio0 (2019): Bug: PHY is undefined for device 'radio0'

In luci i have Wireless is not associated and in Radio 0 settings,  Operation frequency field  is empty, i can't choose anythink. 
I noticed when i change 5Ghz radio 1 channel to higher than 40 i have this same problem as radio 0.

WeiqiChenCN, 2021-11-13:
Openwrt Version: `OpenWrt 21.02.1 r16325-88151b8303`

System Log: 

```
Sat Nov 13 20:23:38 2021 daemon.notice netifd: radio1 (4423): Could not find PHY for device 'radio1'
Sat Nov 13 20:23:38 2021 daemon.notice netifd: radio1 (4438): WARNING: Variable 'data' does not exist or is not an array/object
Sat Nov 13 20:23:38 2021 daemon.notice netifd: radio1 (4438): Bug: PHY is undefined for device 'radio1'
```

(I want upload log file and screen shot file, but there is a bug from github, Therefore I cannot upload files)
(github bug: `POST https://github-production-upload-manifest-file-7fdce7.s3.amazonaws.com/ net::ERR_SSL_VERSION_OR_CIPHER_MISMATCH` )

robin92, 2022-03-18:
I got the following error too
```
Fri Mar 18 19:10:35 2022 daemon.notice netifd: radio0 (3331): Could not find PHY for device 'radio0'
Fri Mar 18 19:10:35 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Fri Mar 18 19:10:35 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Fri Mar 18 19:10:35 2022 daemon.notice netifd: radio0 (3343): WARNING: Variable 'data' does not exist or is not an array/object
Fri Mar 18 19:10:35 2022 daemon.notice netifd: radio0 (3343): Bug: PHY is undefined for device 'radio0'
Fri Mar 18 19:10:35 2022 daemon.notice netifd: Wireless device 'radio0' is now down
```

pierrepinon, 2022-06-25:
@robin92 I have the same issue, have you fix your issue ?

robin92, 2022-06-25:
No, not really. I wanted to buy another router anyway and haven't got
enough time to look deeper into the issue with this one.

> Message ID: ***@***.***>
>

DragonBluep, 2022-06-25:
https://forum.openwrt.org/t/r6220-could-not-find-phy-for-device-radio0-mt76x2-issue-continues/116665/4
@ligthpower said add these pcie reset pin definition to `mt7621_netgear_r6220.dts` solve the problem.
```
&pcie {
        status = "okay";

        reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>,
                      <&gpio 8 GPIO_ACTIVE_LOW>;
};
```

pierrepinon, 2022-06-25:
@DragonBluep I try solution and share my feedback. Thx

pierrepinon, 2022-06-26:
I add PR with fix in openwrt repository (https://github.com/openwrt/openwrt/pull/10125)

WeiqiChenCN, 2022-06-28:
> https://forum.openwrt.org/t/r6220-could-not-find-phy-for-device-radio0-mt76x2-issue-continues/116665/4 @ligthpower said add these pcie reset pin definition to `mt7621_netgear_r6220.dts` solve the [...]

I tried updated:
$ cat target/linux/ramips/dts/mt7621_netgear_sercomm_ayx.dtsi
```
...
&pcie {
        status = "okay";
        reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>,
                      <&gpio 8 GPIO_ACTIVE_LOW>;
};
...
```

or put changes into `target/linux/ramips/dts/mt7621_netgear_r6220.dts`

then created the image:

`make image PROFILE="netgear_r6220" FILES="files" PACKAGES="luci"`

```
root@netgear-openwrt:~# dmesg | grep pci
[    2.306509] mt7621-pci 1e140000.pcie: Parsing DT failed
[    2.523125] rt2880-pinmux pinctrl: pcie is already enabled
[    2.534238] mt7621-pci-phy 1e149000.pcie-phy: PHY for 0xbe149000 (dual port = 1)
[    2.549183] mt7621-pci-phy 1e14a000.pcie-phy: PHY for 0xbe14a000 (dual port = 0)
[    2.663730] mt7621-pci-phy 1e149000.pcie-phy: Xtal is 40MHz
[    2.674847] mt7621-pci-phy 1e14a000.pcie-phy: Xtal is 40MHz
[    2.785675] mt7621-pci 1e140000.pcie: pcie1 no card, disable it (RST & CLK)
[    2.799542] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
[    2.813416] mt7621-pci 1e140000.pcie: PCIE0 enabled
[    2.823137] mt7621-pci 1e140000.pcie: PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
[    2.841877] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
[    2.854558] pci_bus 0000:00: root bus resource [io  0x1e160000-0x1e16ffff]
[    2.868259] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
[    2.881960] pci_bus 0000:00: root bus resource [bus 00-ff]
[    2.892952] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
[    2.904956] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    2.917447] pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[    2.930012] pci 0000:00:00.0: supports D1
[    2.938003] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    2.950985] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    2.967172] pci 0000:01:00.0: [14c3:7662] type 00 class 0x028000
[    2.979205] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit]
[    2.992784] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    3.006238] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    3.019902] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    3.030322] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
[    3.042467] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff]
[    3.055998] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff pref]
[    3.070390] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    3.083616] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
[    3.096807] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
[    3.110680] pci 0000:00:00.0: BAR 8: assigned [mem 0x60000000-0x600fffff]
[    3.124212] pci 0000:00:00.0: BAR 9: assigned [mem 0x60100000-0x601fffff pref]
[    3.138604] pci 0000:00:00.0: BAR 1: assigned [mem 0x60200000-0x6020ffff]
[    3.152141] pci 0000:00:00.0: BAR 7: assigned [io  0x1e160000-0x1e160fff]
[    3.165673] pci 0000:01:00.0: BAR 0: assigned [mem 0x60000000-0x600fffff 64bit]
[    3.180250] pci 0000:01:00.0: BAR 6: assigned [mem 0x60100000-0x6010ffff pref]
[    3.194640] pci 0000:00:00.0: PCI bridge to [bus 01]
[    3.204543] pci 0000:00:00.0:   bridge window [io  0x1e160000-0x1e160fff]
[    3.218069] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    3.231594] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
[   13.563339] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=19
[   13.574253] pci 0000:00:00.0: enabling device (0004 -> 0007)
```

But still no 2.4g wireless...

WeiqiChenCN, 2022-06-28:
I tried imagebuilder 22.03.0-rc4, still no good.

```
root@OpenWrt:~# iw dev
phy#0
        Interface wlan0
                ifindex 8
                wdev 0x1
                addr cc:40:d0:9c:0c:44
                type managed
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0

```

```
root@OpenWrt:~# dmesg | grep pci
[    2.329732] mt7621-pci 1e140000.pcie: host bridge /pcie@1e140000 ranges:
[    2.343148] mt7621-pci 1e140000.pcie:   No bus range found for /pcie@1e140000, using [bus 00-ff]
[    2.360730] mt7621-pci 1e140000.pcie:      MEM 0x0060000000..0x006fffffff -> 0x0000000000
[    2.377062] mt7621-pci 1e140000.pcie:       IO 0x001e160000..0x001e16ffff -> 0x0000000000
[    2.393531] mt7621-pci 1e140000.pcie: Parsing DT failed
[    2.643767] rt2880-pinmux pinctrl: pcie is already enabled
[    2.654866] mt7621-pci 1e140000.pcie: host bridge /pcie@1e140000 ranges:
[    2.668238] mt7621-pci 1e140000.pcie:   No bus range found for /pcie@1e140000, using [bus 00-ff]
[    2.685770] mt7621-pci 1e140000.pcie:      MEM 0x0060000000..0x006fffffff -> 0x0000000000
[    2.702073] mt7621-pci 1e140000.pcie:       IO 0x001e160000..0x001e16ffff -> 0x0000000000
[    2.718483] mt7621-pci-phy 1e149000.pcie-phy: PHY for 0xbe149000 (dual port = 1)
[    2.733648] mt7621-pci-phy 1e14a000.pcie-phy: PHY for 0xbe14a000 (dual port = 0)
[    2.748746] mt7621-pci 1e140000.pcie: failed to parse bus ranges property: -22
[    2.863422] mt7621-pci-phy 1e149000.pcie-phy: Xtal is 40MHz
[    2.874547] mt7621-pci-phy 1e14a000.pcie-phy: Xtal is 40MHz
[    2.985890] mt7621-pci 1e140000.pcie: pcie1 no card, disable it (RST & CLK)
[    2.999754] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
[    3.013627] mt7621-pci 1e140000.pcie: PCIE0 enabled
[    3.023357] mt7621-pci 1e140000.pcie: PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
[    3.042072] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
[    3.054786] pci_bus 0000:00: root bus resource [io  0x1e160000-0x1e16ffff]
[    3.068497] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
[    3.082204] pci_bus 0000:00: root bus resource [bus 00-ff]
[    3.093147] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff] (bus address [0x00000000-0x0fffffff])
[    3.113476] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
[    3.125467] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    3.137971] pci 0000:00:00.0: reg 0x14: [mem 0x60200000-0x6020ffff]
[    3.150524] pci 0000:00:00.0: supports D1
[    3.158519] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    3.171935] pci 0000:01:00.0: [14c3:7662] type 00 class 0x028000
[    3.183997] pci 0000:01:00.0: reg 0x10: initial BAR value 0x00000000 invalid
[    3.198076] pci 0000:01:00.0: reg 0x10: [mem size 0x00100000 64bit]
[    3.210603] pci 0000:01:00.0: reg 0x30: initial BAR value 0x00000000 invalid
[    3.224647] pci 0000:01:00.0: reg 0x30: [mem size 0x00010000 pref]
[    3.237065] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    3.250859] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    3.261317] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
[    3.273459] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    3.286989] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
[    3.301387] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    3.314618] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
[    3.327800] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
[    3.341672] pci 0000:00:00.0: BAR 8: assigned [mem 0x60000000-0x600fffff]
[    3.355202] pci 0000:00:00.0: BAR 9: assigned [mem 0x60100000-0x601fffff pref]
[    3.369595] pci 0000:00:00.0: BAR 1: assigned [mem 0x60200000-0x6020ffff]
[    3.383132] pci 0000:00:00.0: BAR 7: assigned [io  0x1e160000-0x1e160fff]
[    3.396671] pci 0000:01:00.0: BAR 0: assigned [mem 0x60000000-0x600fffff 64bit]
[    3.411248] pci 0000:01:00.0: BAR 6: assigned [mem 0x60100000-0x6010ffff pref]
[    3.425638] pci 0000:00:00.0: PCI bridge to [bus 01]
[    3.435546] pci 0000:00:00.0:   bridge window [io  0x1e160000-0x1e160fff]
[    3.449069] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    3.462597] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
[   14.706474] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=18
[   14.717319] pci 0000:00:00.0: enabling device (0006 -> 0007)
```

WeiqiChenCN, 2022-06-30:
I compiled openwrt by source cdoe. And I noticed that `wireless` is missing from `Network`.

Is there some options I not enabled in menuconfig or kernel_menuconfig?
Who could help me out :rofl:  ?

pierrepinon, 2022-06-30:
@WeiqiChenCN I post an message on openwrt PR, you are right, the fix doesn't works as expected

pierrepinon, 2022-06-30:
I have a [WAC104](https://openwrt.org/toh/netgear/wac104) and [R6220](https://openwrt.org/toh/netgear/r6220) in my network. It's build on same architecture with same chipsets, the only differences that R6220 have WAN interface and USB port

I have no issue with WAC104 after many reboot or electric issue, the 2.4Ghz card is always up

So, I flash R6220 with WAC104 image, but the issue is already there 😓

WeiqiChenCN, 2022-06-30:
> I compiled openwrt by source cdoe. And I noticed that `wireless` is missing from `Network`.
> 
> Is there some options I not enabled in menuconfig or kernel_menuconfig? Who could help me out rofl ?

@pierrepinon Do you konw which options/switch to enable wifi drivers or somethings?

pierrepinon, 2022-06-30:
Right now I build new image with some fix pull from wac104 dts file

weiqi-chen, 2022-06-30:
> @weiqi-chen follow the [build wiki](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem)
> 
> Don't forgot to download official build config to build system with same official build parameters : https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem
> 
> Great day

I did compile openwrt follw the wiki [Build system usage](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#build_system_usage).

I also download `https://downloads.openwrt.org/releases/22.03.0-rc4/targets/ramips/mt7621/config.buildinfo` and overwrited `.config` to get the same configure from official 22.03.0-rc4 build.

I also used 
```
$ ./scripts/feeds update packages luci
$ ./scripts/feeds install -a -p luci
make menuconfig
```
to include luci(web frontend).

I also check `make menuconfig` and `make kernel_menuconfig` and check if drivers or mac80211 or somethings are not selected.

after all this, there is no wifi or radio0 show up in my openwrt built. :rofl:

weiqi-chen, 2022-06-30:
I used this steps to build my image

```
./scripts/feeds update -a
./scripts/feeds install -a
wget https://downloads.openwrt.org/releases/22.03.0-rc4/targets/ramips/mt7621/config.buildinfo -O .config
make -j $(nproc) defconfig 
make -j $(nproc) menuconfig
make -j $(nproc) world
```

Still no wireless interfaces...

weiqi-chen, 2022-07-02:
Updated:

```bash
make distclean
export all_proxy=socks5://rpi-ubuntu:1090
git checkout -- .
./scripts/feeds update -a
./scripts/feeds install -a

wget https://downloads.openwrt.org/releases/22.03.0-rc4/targets/ramips/mt7621/config.buildinfo -O .config

make -j $(nproc) defconfig

make -j $(nproc) download clean world

#flash firmware, then reboot, and reboot again.
#Now I can see wireless(radio0, 5G).

#patch ./target/linux/ramips/dts/mt7621_netgear_r6220.dts

make -j $(nproc) world

#flash firmware, then reboot, and reboot again.
#Still no wireless(radio1, 2.4G).

This patch is not work as expected.

```

pierrepinon, 2022-07-03:
FYI, when the device is switch off and start after a few minutes, 2.4Ghz pcie card works

weiqi-chen, 2022-07-04:
> FYI, when the device is switch off and start after a few minutes, 2.4Ghz pcie card works

But After time of about between a half of a day and a day, the 2.4G wireless is down. And could not start up again.

```
on Jul  4 17:54:06 2022 user.info : luci: accepted login on / for root from 192.168.10.147
Mon Jul  4 17:54:44 2022 daemon.notice netifd: radio1 (5926): Phy not found
Mon Jul  4 17:54:44 2022 daemon.notice netifd: radio1 (5926): Could not find PHY for device 'radio1'
Mon Jul  4 17:54:44 2022 daemon.notice netifd: Wireless device 'radio1' set retry=0
Mon Jul  4 17:54:44 2022 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Mon Jul  4 17:54:44 2022 daemon.notice netifd: radio1 (5937): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jul  4 17:54:44 2022 daemon.notice netifd: radio1 (5937): Bug: PHY is undefined for device 'radio1'
Mon Jul  4 17:54:44 2022 daemon.notice netifd: Wireless device 'radio1' is now down
Mon Jul  4 17:57:31 2022 daemon.notice netifd: radio1 (6075): Phy not found
Mon Jul  4 17:57:31 2022 daemon.notice netifd: radio1 (6075): Could not find PHY for device 'radio1'
Mon Jul  4 17:57:31 2022 daemon.notice netifd: Wireless device 'radio1' set retry=0
Mon Jul  4 17:57:31 2022 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Mon Jul  4 17:57:31 2022 daemon.notice netifd: radio1 (6086): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jul  4 17:57:31 2022 daemon.notice netifd: radio1 (6086): Bug: PHY is undefined for device 'radio1'
Mon Jul  4 17:57:31 2022 daemon.notice netifd: Wireless device 'radio1' is now down
```

I'm afraid that I need to switch off the router and wait minutes and turn it back on to make wifi work again...

pierrepinon, 2022-07-04:
@weiqi-chen I doesn’t have the same issue. But can you try to flash your device with WAC104 firmware (with force) ?

gj86, 2022-07-21:
> FYI, when the device is switch off and start after a few minutes, 2.4Ghz pcie card works

This solution works randomly. Sometime it never work, regardless of number of minutes or number of reboots.

weiqi-chen, 2022-07-21:
Yes. That is a sad story.

pierrepinon, 2022-07-21:
I just tried the last snapshot and I do not reproduce the problem for the moment

gj86, 2022-07-22:
> I just tried the last snapshot and I do not reproduce the problem for the moment

Same for me. Even if the 2.4G not detected, next reboot (without any wait) fixes it.

jacgl, 2022-08-04:
The same issue on one of mine R6220 after upgrading 19.07->22.03 (probably no matter which build). I have to restart several times until 2.4G radio is discovered.
Apparently, the 7603 chip disappears on PCIe, It is not visible in dmesg when enumeration is done, thus firmware not loaded.
mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)

pierrepinon, 2022-08-04:
@gj86 I have rebooted my device many times (for new settings or by electrical issues) and have no issues with radio 2.4GHZ

Do you have news issues or try new build ?

kgrodrigues, 2022-08-04:
From stock, I tried 21.02.3 yesterday, with 5 or so reboots, powering off for up to 5 mins I could not get my 2.4Ghz radio to show up. I downgraded to 19.07.9 and its been working fine.

gj86, 2022-08-05:
> @gj86 I have rebooted my device many times (for new settings or by electrical issues) and have no issues with radio 2.4GHZ
> 
> Do you have news issues or try new build ?

The issue is not fixed, tried with August 5 snapshot.

Hurricos, 2022-08-28:
The issue appears to be because the paths of the PCIe devices have changed -- this is at least true on the AP3825i.

pierrepinon, 2022-09-25:
https://github.com/openwrt/openwrt/issues/9374#issuecomment-1251506069

DragonBluep, 2022-10-17:
Does anyone want to try this fix? Look forward to your feedback.
openwrt image with fix: https://github.com/DragonBluep/openwrt/actions/runs/3260026677

Djfe, 2023-02-26:
@ABusuuu please close this issue, everything has been fixed :)
Even a new change that was made necessary by switching to Kernel 5.15 (Snapshot)
https://github.com/openwrt/openwrt/pull/12042

closed 2023-02-27 by nbd168