openwrt/mt76 #606: TP-LINK RE200 problem with the operation of the WLAN chipset. OpenWRT rel 21.02rc4

WodnyTraper, 2021-10-14:
Hello colleagues.

I have a TP-LINK RE200 v2.1 repeater based on the mediatek MT7628AN chipset and MT76x8AN and MT7610EN chipsets for WLAN. I am using OpenWRT image 21.02rc4. I came across a problem already described on many forums, the critical instability and the disappearance of SSID broadcasting in the 5GHz network. It is practically impossible to use now. The problems vary, but clearly indicate a driver error for this chipset. Maybe one of my fellow developers could help to solve this problem. Or one of you has found a solution to this problem.

WodnyTraper, 2021-10-16:
from what I read, this chip currently works best on the auto setting and then there is channel 36, and high channels such as No. 120 the chipset probably does not support and may be problematic for other devices

d-me3, 2021-11-14:
> Ok, I am sorry if I am little bit repetitive here but - may you try latest OpenWRt HEAD? I am hoping commit [645eac6](https://github.com/openwrt/mt76/commit/645eac64bece2770ec7b25515df3e30f1385e15d) fixes your issue. It would be great if you can report back! :)

Unfortunately this fix did not solve the 5Ghz issue with RE200 v2. I just did a snapshot build today (Nov 13th 2021) from master, and the 5Ghz radio is still capped at 9dBm and does not work, as can be seen below.

```
BusyBox v1.34.1 (2021-11-13 20:29:41 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r18084-b5ab6d0bc5
 -----------------------------------------------------
root@ap6:~# cat /proc/cpuinfo | grep machine
machine                 : TP-Link RE200 v2
root@ap6:~# iw phy phy1 channels
Band 2:
        * 5180 MHz [36]
          Maximum TX power: 9.0 dBm
          Channel widths: 20MHz HT40+ VHT80
        * 5200 MHz [40]
          Maximum TX power: 9.0 dBm
          Channel widths: 20MHz HT40- HT40+ VHT80
        * 5220 MHz [44]
          Maximum TX power: 8.0 dBm
          Channel widths: 20MHz HT40- HT40+ VHT80
        * 5240 MHz [48]
          Maximum TX power: 8.0 dBm
          Channel widths: 20MHz HT40- HT40+ VHT80
        * 5260 MHz [52]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5280 MHz [56]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5300 MHz [60]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5320 MHz [64]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5500 MHz [100]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5520 MHz [104]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5540 MHz [108]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5560 MHz [112]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5580 MHz [116]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5600 MHz [120]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5620 MHz [124]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5640 MHz [128]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5660 MHz [132]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5680 MHz [136]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5700 MHz [140]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- HT40+ VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5720 MHz [144]
          Maximum TX power: 8.0 dBm
          No IR
          Radar detection
          Channel widths: 20MHz HT40- VHT80
          DFS state: usable (for 779 sec)
          DFS CAC time: 60000 ms
        * 5745 MHz [149]
          Maximum TX power: 8.0 dBm
          No IR
          Channel widths: 20MHz HT40+ VHT80
        * 5765 MHz [153]
          Maximum TX power: 9.0 dBm
          No IR
          Channel widths: 20MHz HT40- HT40+ VHT80
        * 5785 MHz [157]
          Maximum TX power: 9.0 dBm
          No IR
          Channel widths: 20MHz HT40- HT40+ VHT80
        * 5805 MHz [161]
          Maximum TX power: 9.0 dBm
          No IR
          Channel widths: 20MHz HT40- HT40+ VHT80
        * 5825 MHz [165]
          Maximum TX power: 9.0 dBm
          No IR
          Channel widths: 20MHz HT40- VHT80
        * 5845 MHz [169] (disabled)
        * 5865 MHz [173] (disabled)
root@ap6:~#
```

phpaiva, 2022-10-09:
You can check this topic and my last posts, maybe is a useful hint for someone. I find diffent behaviors between suposed exactly same devices and firmwares, but by using that Oct 2020 snapshot (r14820-5cdc65f6d1) and leaving a task in crontab or a service which issues the commands I used every sec, 5 sec or per minute you get the device in fine operating conditions (also AP+STA mode in 5 GHz). My current script is:

    while true
    
	  do
  
		  sleep 1
		  cd /sys/kernel/debug/ieee80211/phy1/mt76
		  echo 0x13b0 > regidx
		  echo 0x2f2f1414 > regval
		  echo 0x2320 > regidx
		  echo 0x122c3cf2 > regval
  
	  done

It's a shame, but it seems that somewhen between that snapshot and the next OWRT release they ruined something which is more than just a Tx power issue.

The main problem in my case is that I do not understand what I am really doing, nobody seems yo be able to answer and I find a random behavior between devices which seem to be equal in hardware and firmware.

https://github.com/openwrt/mt76/issues/633#issuecomment-1272384213