openwrt/mt76 #617: mt7915e report support 160MHz

ptpt52, 2021-11-05:
but 160Mhz not actually supported by mt7915e, right?

ptpt52, 2021-11-05:
iw phy1 info
```
~# iw phy1 info

Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 4
	max scan IEs length: 2190 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 2:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x339979b1):
			Max MPDU length: 7991
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformer
			SU Beamformee
			MU Beamformer
			MU Beamformee
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
[... 242 lines cut ...]
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

majuss, 2021-11-11:
So this page states that the D variant is built into the Totolink for example, and only support HE80 so no 160 MHz. http://en.techinfodepot.shoutwiki.com/wiki/MediaTek

dfateyev, 2022-05-29:
I believe, it's specific to the chip used. I also haven't managed to get 160MHz band on Totolink devices on openwrt-21.02.

cryobry, 2022-12-31:
> So this page states that the D variant is built into the Totolink for example, and only support HE80 so no 160 MHz. http://en.techinfodepot.shoutwiki.com/wiki/MediaTek

No, the Totolink X5000R has the MT7915**E**, which does support 5Ghz 160Mhz channels. Of course, the usual caveats apply, you are limited to channel 100 (DFS) for 160Mhz in the US, the upper and lower channel groups only support 80Mhz channels.

```
#iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

DragonBluep, 2023-01-01:
It should have been fixed by https://github.com/openwrt/mt76/commit/daa30a31cf70ff290ab4d46cc132cdd8d35f88df.

Djfe, 2023-05-09:
and https://github.com/openwrt/mt76/commit/f9ca70d6367a16fecd80b93fec60fb0d4e3b9fd0 partially reintroduced the feature.

The comments under the commit list some issues with apple devices connecting to a Belkin RT-3200 though and reduced throughput.

mangkoran, 2023-05-21:
Currently I have Xiaomi CR6608 which use the same wifi chipset and living in MY.

```
# iw reg get
global
country MY: DFS-FCC
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 24), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5490 - 5650 @ 160), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 24), (N/A)
```
So does this mean 160Mhz channel width can only be used with channel 100, same as @cryobry [case above](https://github.com/openwrt/mt76/issues/617#issuecomment-1368219509)?

spamsool, 2024-08-23:
I also haven't managed to get 160MHz band on Totolink x500R on openwrt-23.05