openwrt/mt76 #620: MT7915 - no communication between ap clients

d3adme4t, 2021-11-11:
Hello!

I have a problem with my mt7915.
MT7915 card(tested both on dbdc and non-dbdc versions) card is set as an access point (hostapd).

All clients are able to connect and communicate to AP host, and wire-connected hosts, but not each other. 
The same hostapd.conf works great with MT7615, but not 7915. 
For me, it seems like hostapd is set to ap_isolate=1 (I've tried to explicitly set it to 0 - no effect)

my hostapd.conf:

```
interface=wlan0
driver=nl80211
ssid=test
hw_mode=g
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40]
ap_isolate=0
ieee80211n=1
country_code=US
channel=6
macaddr_acl=0
wmm_enabled=1
auth_algs=3
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
wpa_passphrase=password
```

Ready to provide any additional info.
Thank You for any advice.

ptpt52, 2021-11-16:
no such issue
```
client----AP----client
```
client to client talk ok for my test.

d3adme4t, 2021-11-16:
What kernel version do You use?