openwrt/mt76 #621: problem with MT7603U

ZibiM, 2021-11-19:
i have problem with usb wireless MT7603U,

lsusb:
Bus 001 Device 004: ID 0e8d:7603 MediaTek 802.11 n WLAN

kernel log:
kern.info kernel: [    2.660864] usb 1-1.1.3: new high-speed USB device number 4 using dwc_otg         
kern.info kernel: [    2.769926] usb 1-1.1.3: New USB device found, idVendor=0e8d, idProduct=7603, bcdDevice= 0.01
kern.info kernel: [    2.783415] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3     
kern.info kernel: [    2.796130] usb 1-1.1.3: Product: 802.11 n WLAN                  
kern.info kernel: [    2.803275] usb 1-1.1.3: Manufacturer: MediaTek                                                                        
kern.info kernel: [    2.810324] usb 1-1.1.3: SerialNumber: 1.0 

openwrt:
Firmware Version | OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.314.39479-e1ccb66

installed on Raspberry PI 3B+

installed kmod-mt76

in interface i see only wlan0 from raspberry, 
iw list display only Wiphy phy0

what's wrong with that

lukasz1992, 2021-11-19:
This driver does not support mt7603u

closed 2021-11-25 by nbd168

ChalesYu, 2022-04-15:
Just for reference , a modified vendor driver is [here](https://github.com/ChalesYu/buildroot_platform_hardware_wifi_mtk_drivers_mt7603/tree/pub-test-v20220304)
and , a half-complete(doesn't work) mt76 driver is [here](https://github.com/ChalesYu/buildroot_platform_hardware_wifi_mtk_drivers_mt7603/tree/mt76-03-usb)

qtopie, 2026-10-18:
for anyone may find it useful https://github.com/qtopie/mt7603u