openwrt/mt76 #622: 802.11s mesh broken

lynxthecat, 2021-11-19:
Presumably the recent mt76 feature updates are responsible for the reports surfacing of mesh having been broken in recent snapshots. 

To elaborate, I have x3 RT3200's setup in an 802.11s mesh. For months it has been 100% stable, but now recently either one of my two extension nodes becomes inaccessible even though the other stations list the inaccessible node as still connected. Rebooting my main router restores connectivity, but then eventually one of the two extensions nodes will again become inaccessible. By inaccessible I mean I cannot access the web interface or ping the affected node. 

I believe this issue has surfaced as a result of a change reflected in a fairly recent snapshot. 

The receive and send packets seem to increase during the period of inaccessibility (which I don't understand)? See the first station (presently inaccessible) rx/tx packets increase:

```
root@OpenWrt:~# iw dev wlan1 station dump |grep -i packets -A 2 -B 2
        inactive time:  50 ms
        rx bytes:       11694224
        rx packets:     98735
        tx bytes:       115203
        tx packets:     921
        tx retries:     0
        tx failed:      0
--
        inactive time:  10 ms
        rx bytes:       39163475
        rx packets:     184596
        tx bytes:       107933315
        tx packets:     100696
        tx retries:     0
        tx failed:      0
root@OpenWrt:~# iw dev wlan1 station dump |grep -i packets -A 2 -B 2
        inactive time:  90 ms
        rx bytes:       11698067
        rx packets:     98769
        tx bytes:       115203
        tx packets:     921
        tx retries:     0
        tx failed:      0
--
        inactive time:  0 ms
        rx bytes:       39168194
        rx packets:     184638
        tx bytes:       107934713
        tx packets:     100703
        tx retries:     0
        tx failed:      0
``` 

Dmesg output: 

```
BusyBox v1.34.1 (2021-11-16 09:48:45 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r18107-db34b93331
 -----------------------------------------------------
root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.79 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r18107-db34b93331) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Nov 16 21:02:45 2021
[    0.000000] Machine model: Linksys E8450 (UBI)
[    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[... 349 lines cut ...]
[   20.273213] x15: 0000000000000001 x14: 0000000000000070
[   20.278517] x13: 0000000000000080 x12: ffffffc010878040
[   20.283821] x11: 0000000000000001 x10: 0000000000000001
[   20.289124] x9 : ffffffc010e8b5f8 x8 : ffffff8000863700
[   20.294428] x7 : 0000000000000000 x6 : 0000000000000000
[   20.299731] x5 : 0000000000000000 x4 : 0000000000000000
[   20.305034] x3 : ffffffc010e8bb10 x2 : 0000000000000000
[   20.310339] x1 : 0000000000000000 x0 : 0000000000000000
[   20.315642] Call trace:
[   20.318083]  __skb_flow_dissect+0x1a0/0x13dc
[   20.322345]  __skb_get_hash+0x6c/0x130
[   20.326110]  ieee80211_schedule_txq+0x784/0x9dc [mac80211]
[   20.331597]  ieee80211_schedule_txq+0x974/0x9dc [mac80211]
[   20.337083]  ieee80211_tx_pending+0xe0/0x24c [mac80211]
[   20.342301]  tasklet_action_common.constprop.0+0x134/0x17c
[   20.347777]  tasklet_action+0x24/0x30
[   20.351430]  _stext+0x124/0x28c
[   20.354563]  do_softirq+0x74/0x80
[   20.357869]  __local_bh_enable_ip+0x88/0x90
[   20.362045]  napi_threaded_poll+0x94/0xf0
[   20.366048]  kthread+0x120/0x124
[   20.369267]  ret_from_fork+0x10/0x18
[   20.372833] ---[ end trace d98c09f8ca4ebf83 ]---
root@OpenWrt:~#
```

lynxthecat, 2021-11-20:
It seems that one of the extension nodes (i.e. nodes which connect to my main router acting as gateway) becomes inaccessible following a period of inactivity. Could this therefore be some kind of power saving or something?

d3adme4t, 2021-11-23:
Hello,
Looks like the related patch was published for the mainline kernel.

Pls see:
https://patchwork.kernel.org/project/linux-mediatek/patch/20211123033123.2684-1-xing.song@mediatek.com/

lynxthecat, 2021-11-24:
The patch addresses the crash trace? That was always there, even when the mesh functionality worked well. But what is new is that nodes on the mesh become inaccessible. I presume that is not something that has been addressed by the patch?

nbd168, 2021-11-25:
There was another patch that should improve mesh mode stability (allowing auth frames in blocked state). Please test the latest version

lynxthecat, 2022-01-07:
Same issue for me. If it helps when a node becomes inaccessible (although still listed as connected), I see a very low transmit rate:

[image]