openwrt/mt76 #625: MT7610E: STA mode not working correctly. Unable to authenticate to AP

DavideFioravanti, 2021-11-24:
I am having a lot of troubles using the MT7610E as STA. Currently I am testing these two devices using the latest code in master:

- TP-LINK TL-WR902AC (MT7628 + MT7610E)
- ASUS RT-AC51U (MT7620 + MT7610E)

Both devices are not able to connect as STA to any AP. I tried changing channel (36, 100, 165), with or without encryption, using Qualcomm or Mediatek HW as AP without any luck.
The log says:
```
wlan1: authenticate with XX:XX:XX:XX:XX:XX
wlan1: send auth to XX:XX:XX:XX:XX:XX (try 1/3)
wlan1: send auth to XX:XX:XX:XX:XX:XX (try 2/3)
wlan1: send auth to XX:XX:XX:XX:XX:XX (try 3/3)
wlan1: authentication with XX:XX:XX:XX:XX:XX timed out
```

Using another wifi adapter in monitor mode I see this (GLTechno_XX:XX:XX is the AP using QCA9887; Tp-LinkT_YY:YY:YY is the STA using the MT7610E):
```
2965	596.801882973	Tp-LinkT_YY:YY:YY	Broadcast		802.11	119	Probe Request, SN=6, FN=0, Flags=........, SSID=TEST
2966	596.802178823	GLTechno_XX:XX:XX	Broadcast		802.11	229	Beacon frame, SN=1006, FN=0, Flags=........, BI=200, SSID=TEST
2967	596.802965303	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1007, FN=0, Flags=........, BI=200, SSID=TEST
2968	596.803290283	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1007, FN=0, Flags=....R..., BI=200, SSID=TEST
2969	596.803619383	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1007, FN=0, Flags=....R..., BI=200, SSID=TEST
2970	596.803952793	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1007, FN=0, Flags=....R..., BI=200, SSID=TEST
2971	597.006971814	GLTechno_XX:XX:XX	Broadcast		802.11	229	Beacon frame, SN=1008, FN=0, Flags=........, BI=200, SSID=TEST
2972	597.188437285	Tp-LinkT_YY:YY:YY	Broadcast		802.11	119	Probe Request, SN=32, FN=0, Flags=........, SSID=TEST
2973	597.189427945	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1009, FN=0, Flags=........, BI=200, SSID=TEST
2974	597.189778145	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1009, FN=0, Flags=....R..., BI=200, SSID=TEST
2975	597.190191515	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1009, FN=0, Flags=....R..., BI=200, SSID=TEST
2976	597.190488715	GLTechno_XX:XX:XX	Tp-LinkT_YY:YY:YY	802.11	223	Probe Response, SN=1009, FN=0, Flags=....R..., BI=200, SSID=TEST
2977	597.211754664	GLTechno_XX:XX:XX	Broadcast		802.11	229	Beacon frame, SN=1010, FN=0, Flags=........, BI=200, SSID=TEST
```
The MT7610E seems to not send any authentication packet after the Probe Response. However if you look closely, the SN of the TP-LINK jumps from 6 to 32, so I think that router at least creates the packets but for some reason it's not able to transmit to the world. 

Additional infos:
**1)** When I restart the MT7610E I usually get this message
```
[ 1315.597177] mt76x0e 0000:01:00.0: TX DMA did not stop
[ 1321.598430] mt76x0e 0000:01:00.0: Warning: MAC TX did not stop!
```
and some packets are sent immediately after the second message. Probably there is an issue with the queues?

**2)** Rarely I am able to connect to the AP but only when I start two interfaces on the same MT7610E (one in STA and another in Monitor mode). Maybe the delay added by the second interface fixes something?

**3)** Interestingly MT7610U connected to the USB port of the same router works correctly, so the problem seems to be specific to the MT7610E.

Am I the only one with this problem? I tried to find the problem in the MT7610E specific part of the driver but it's too complex for me. I need at least some hints from the devs to narrow the search.

Thank you.

cgm999, 2022-08-30:
@DavideFioravanti is this issue still present for you?
 If yes check the value of regs 13b0 and 2320 when this is happening, I think I found the fix .. basically 

if I wrote into 0x2320  122c3cf2 it seems to get this going..

skramstad, 2023-08-22:
I've got the same issue with my TP-Link TL-WR902AC v3 and version 23.05 RC2, snapshot build etc... none of the latest builds work with STA/Client and 5GHz, however 

> @DavideFioravanti is this issue still present for you? If yes check the value of regs 13b0 and 2320 when this is happening, I think I found the fix .. basically
> 
> if I wrote into 0x2320 122c3cf2 it seems to get this going..

did work, and after a reboot it connected to my 5GHz wlan. You can have a look at the small fix here https://github.com/openwrt/mt76/issues/633#issuecomment-1445294226 .

I did the following btw:
```
# hexdump -ve '1/4 "%.8x\n"' /rom//lib/modules/$(uname -r)/mt76x0-common.ko|sed -e 's/^122c54f2/122c3cf2/' |sed -re 's/(..)(..)(..)(..)/\\\\x\4\\\\x\3\\\\x\2\\\\x\1/g'|  xargs -n1 printf >  /tmp/mt76x0-common.ko
# cp /tmp/mt76x0-common.ko /lib/modules/$(uname -r)/mt76x0-common.ko
```

Is there any way to make a permanent fix for this problem @nbd168 ?

Thanks in advance.

cgm999, 2023-08-22:
The fix from above changes 0x54 value to 0x3c , which I added some comments in #633 that we should take the value from EEPROM as starting point instead of 0x54.

skramstad, 2023-08-22:
Cool! This is beyond my knowledge @cgm999, but are we looking at a patch for MT7610E or device specific like TL-WR902AC v3? It seems that it might be more devices than just TL-WR902AC v3.

progvp, 2025-02-07:
I still have this issue with TP-Link TL-WR902AC v3 on both 23.05 and new 24.10 firmwares.

cgm999, 2025-02-16:
@progvp is possible you need a lower AGC value . As a test , you can try to test by setting the register in while 1 loop

`
echo 0x2320 > /sys/kernel/debug/ieee80211/phy1/mt76/regidx
val=20 # some vals is what you need to try, 2c 28 24 20 1c 18
while :; do
 echo "0x122c${val}f2" > /sys/kernel/debug/ieee80211/phy1/mt76/regval
done
during this time.. look at output: "grep false /sys/kernel/debug/ieee80211/phy1/mt76/agc" , if that is low.. means the value is good
`