openwrt/mt76 #627: 7915 5g sta client connect to wifi5 AP

ptpt52, 2021-11-26:
connected but only 54Mbps
```
root@X-WRT:~# iwinfo wlan1 assoclist
00:0F:43:26:60:00  -24 dBm / unknown (SNR -24)  0 ms ago
	RX: 54.0 MBit/s                               231448 Pkts.
	TX: 54.0 MBit/s                               172310 Pkts.
	expected throughput: unknown

```

AP is mt7613

logread on mt7915
```
Fri Nov 26 13:42:52 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Fri Nov 26 13:42:52 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Fri Nov 26 13:42:52 2021 daemon.notice hostapd: wlan1-1: ACS-COMPLETED freq=5745 channel=149
Fri Nov 26 13:42:52 2021 kern.info kernel: [ 1327.569574] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
Fri Nov 26 13:42:52 2021 kern.info kernel: [ 1327.576607] br-lan: port 5(wlan1-1) entered blocking state
Fri Nov 26 13:42:52 2021 kern.info kernel: [ 1327.582162] br-lan: port 5(wlan1-1) entered forwarding state
Fri Nov 26 13:42:52 2021 daemon.notice netifd: Network device 'wlan1-1' link is up
Fri Nov 26 13:42:52 2021 daemon.notice hostapd: wlan1-1: interface state ACS->ENABLED
Fri Nov 26 13:42:52 2021 daemon.notice hostapd: wlan1-1: AP-ENABLED
Fri Nov 26 13:42:54 2021 daemon.notice wpa_supplicant[3314]: wlan1: SME: Trying to authenticate with 00:0f:43:26:60:00 (SSID='X-WRT_5A71_5G' freq=5260 MHz)
Fri Nov 26 13:42:54 2021 kern.info kernel: [ 1329.430631] wlan1: authenticate with 00:0f:43:26:60:00
Fri Nov 26 13:42:54 2021 daemon.notice wpa_supplicant[3314]: wlan1: SME: Authentication request to the driver failed
Fri Nov 26 13:42:54 2021 daemon.notice netifd: Network device 'wlan1-1' link is down
Fri Nov 26 13:42:54 2021 kern.info kernel: [ 1329.438215] br-lan: port 5(wlan1-1) entered disabled state
Fri Nov 26 13:42:55 2021 daemon.notice wpa_supplicant[3314]: wlan1: SME: Trying to authenticate with 00:0f:43:26:60:00 (SSID='X-WRT_5A71_5G' freq=5260 MHz)
Fri Nov 26 13:42:55 2021 kern.info kernel: [ 1330.772062] wlan1: authenticate with 00:0f:43:26:60:00
Fri Nov 26 13:42:55 2021 kern.info kernel: [ 1330.875378] wlan1: send auth to 00:0f:43:26:60:00 (try 1/3)
Fri Nov 26 13:42:55 2021 kern.info kernel: [ 1330.882645] wlan1: authenticated
Fri Nov 26 13:42:55 2021 daemon.notice wpa_supplicant[3314]: wlan1: Trying to associate with 00:0f:43:26:60:00 (SSID='X-WRT_5A71_5G' freq=5260 MHz)
Fri Nov 26 13:42:55 2021 kern.info kernel: [ 1330.896211] wlan1: associate with 00:0f:43:26:60:00 (try 1/3)
Fri Nov 26 13:42:55 2021 kern.info kernel: [ 1330.905829] wlan1: RX AssocResp from 00:0f:43:26:60:00 (capab=0x11 status=0 aid=3)
Fri Nov 26 13:42:55 2021 daemon.notice netifd: Network device 'wlan1' link is up
Fri Nov 26 13:42:55 2021 daemon.notice netifd: Interface 'wwan' has link connectivity
Fri Nov 26 13:42:55 2021 daemon.notice netifd: Interface 'wwan' is setting up now
Fri Nov 26 13:42:55 2021 daemon.notice wpa_supplicant[3314]: wlan1: Associated with 00:0f:43:26:60:00
Fri Nov 26 13:42:55 2021 kern.info kernel: [ 1330.915127] wlan1: associated
Fri Nov 26 13:42:55 2021 daemon.notice wpa_supplicant[3314]: wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Fri Nov 26 13:42:55 2021 daemon.notice netifd: wwan (5529): udhcpc: started, v1.34.1
Fri Nov 26 13:42:55 2021 daemon.notice wpa_supplicant[3314]: wlan1: WPA: Key negotiation completed with 00:0f:43:26:60:00 [PTK=CCMP GTK=CCMP]
Fri Nov 26 13:42:55 2021 daemon.notice wpa_supplicant[3314]: wlan1: CTRL-EVENT-CONNECTED - Connection to 00:0f:43:26:60:00 completed [id=0 id_str=]
Fri Nov 26 13:42:55 2021 kern.info kernel: [ 1331.048071] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Fri Nov 26 13:42:55 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:42:55 2021 daemon.notice netifd: wwan (5529): udhcpc: broadcasting discover
Fri Nov 26 13:42:55 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:42:55 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:42:55 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:42:58 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:42:58 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:42:58 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:42:58 2021 daemon.notice netifd: wwan (5529): udhcpc: broadcasting discover
Fri Nov 26 13:42:59 2021 daemon.notice netifd: wwan (5529): udhcpc: broadcasting select for 192.168.15.213, server 192.168.15.1
Fri Nov 26 13:42:59 2021 daemon.notice netifd: wwan (5529): udhcpc: lease of 192.168.15.213 obtained from 192.168.15.1, lease time 43200
Fri Nov 26 13:42:59 2021 daemon.notice netifd: Interface 'wwan' is now up
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: reading /tmp/resolv.conf.d/resolv.conf.auto
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using nameserver 192.168.15.1#53
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using only locally-known addresses for test
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using only locally-known addresses for onion
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using only locally-known addresses for localhost
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using only locally-known addresses for local
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using only locally-known addresses for invalid
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using only locally-known addresses for bind
Fri Nov 26 13:42:59 2021 daemon.info dnsmasq[7240]: using only locally-known addresses for lan
Fri Nov 26 13:42:59 2021 user.notice mwan3-hotplug[5585]: mwan3 hotplug on wwan not called because interface disabled
Fri Nov 26 13:43:00 2021 user.notice firewall: Reloading firewall due to ifup of wwan (wlan1)
Fri Nov 26 13:43:01 2021 user.notice natcapd: Reloading cone_wan_ip due to ifup of wwan (wlan1)
Fri Nov 26 13:43:01 2021 user.notice natcapd: Reloading natcap_connected due to ifup of wwan (wlan1)
[... 69 lines cut ...]
Fri Nov 26 13:44:34 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:44:36 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:36 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:36 2021 daemon.notice hostapd: handle_probe_req: send failed
Fri Nov 26 13:44:40 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:40 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:40 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:44:46 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:46 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:46 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:44:52 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:52 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:52 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:44:58 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:58 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:44:58 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:45:04 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:45:04 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:45:04 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:45:10 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:45:10 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Nov 26 13:45:10 2021 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Fri Nov 26 13:45:16 2021 daemon.warn hostapd: Failed to check if DFS is required; ret=-1

```

nbd168, 2021-11-26:
Please show me your /etc/config/wireless and /var/run/hostapd-*.conf

ptpt52, 2021-11-26:
/etc/config/wireless
```

config wifi-device 'radio0'
	option type 'mac80211'
	option band '2g'
	option htmode 'HE20'
	option phy 'phy0'
	option hwmode '11g'
	option disabled '0'
	option country 'CN'
	option channel 'auto'
	option txpower '20'
	option cell_density '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option band '5g'
	option phy 'phy1'
	option hwmode '11a'
	option disabled '0'
	option country 'CN'
	option channel 'auto'
	option txpower '23'
	option cell_density '0'
	option htmode 'HE80'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'X-WRT_68FE'
	option encryption 'psk2'
	option skip_inactivity_poll '1'
	option wpa_group_rekey '0'
	option wpa_pair_rekey '0'
	option wpa_master_rekey '0'
	option disassoc_low_ack '0'
	option key '88888888'

config wifi-iface 'wifinet1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'X-WRT_68FE'
	option encryption 'psk2'
	option skip_inactivity_poll '1'
	option wpa_group_rekey '0'
	option wpa_pair_rekey '0'
	option wpa_master_rekey '0'
	option disassoc_low_ack '0'
	option key '88888888'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'sta'
	option network 'wwan'
	option ssid 'X-WRT_5A71_5G'
	option encryption 'psk2'
	option key '88888888'

```

 /var/run/hostapd-phy1.conf
```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=CN
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
dtim_period=2
channel=acs_survey

noscan=1
tx_queue_data2_burst=2.0
ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=-6
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-2][BF-ANTENNA-4][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]
ieee80211ax=1
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=-6
he_default_pe_duration=4
he_rts_threshold=1023
he_mu_edca_qos_info_param_count=0
he_mu_edca_qos_info_q_ack=0
he_mu_edca_qos_info_queue_request=0
he_mu_edca_qos_info_txop_request=0
he_mu_edca_ac_be_aifsn=8
he_mu_edca_ac_be_aci=0
he_mu_edca_ac_be_ecwmin=9
he_mu_edca_ac_be_ecwmax=10
he_mu_edca_ac_be_timer=255
he_mu_edca_ac_bk_aifsn=15
he_mu_edca_ac_bk_aci=1
he_mu_edca_ac_bk_ecwmin=9
he_mu_edca_ac_bk_ecwmax=10
he_mu_edca_ac_bk_timer=255
he_mu_edca_ac_vi_ecwmin=5
he_mu_edca_ac_vi_ecwmax=7
he_mu_edca_ac_vi_aifsn=5
he_mu_edca_ac_vi_aci=2
he_mu_edca_ac_vi_timer=255
he_mu_edca_ac_vo_aifsn=5
he_mu_edca_ac_vo_aci=3
he_mu_edca_ac_vo_ecwmin=5
he_mu_edca_ac_vo_ecwmax=7
he_mu_edca_ac_vo_timer=255

radio_config_id=1715fdb63709205a1336c2bc003ea25a
interface=wlan1-1
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_group_rekey=0
wpa_ptk_rekey=0
wpa_gmk_rekey=0
wpa_passphrase=88888888
wpa_psk_file=/var/run/hostapd-wlan1-1.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=X-WRT_68FE
bridge=br-lan
wds_bridge=
snoop_iface=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wlan1-1.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
config_id=7e1ac22aa493b0fd45023321f72ff266
bssid=e6:67:1e:ab:69:00

```

/var/run/wpa_supplicant-wlan1.conf 
```

country=CN
network={
	scan_ssid=1
	ssid="X-WRT_5A71_5G"
	key_mgmt=WPA-PSK
	psk="88888888"
	proto=RSN
	beacon_int=100
}
```

ptpt52, 2021-11-26:
oh
it looks good now:
```
root@X-WRT:~# iwinfo wlan1 assoclist
00:0F:43:26:60:00  -21 dBm / unknown (SNR -21)  4760 ms ago
	RX: 780.0 MBit/s, VHT-MCS 9, 80MHz, VHT-NSS 2      2371 Pkts.
	TX: 866.7 MBit/s, VHT-MCS 9, 80MHz, VHT-NSS 2        55 Pkts.
	expected throughput: unknown

```

it take some time to get connected and switch to 866Mbps