openwrt/mt76 #628: MT7628 OpenWrt 21.02 - MCU message 08 (seq 1) timed out

Linaro1985, 2021-11-26:
OpenWrt 21.02 commit https://github.com/openwrt/openwrt/commit/0e0192098a52568d46bdde48a5749f13d3093fc5

Wifi stops working after 3-4 minutes loading of ubuntu iso via torrent. There is no any message in the log before 'wifi down' command:
mt76_wmac 10300000.wmac: MCU message 08 (seq 1) timed out

The same problem on 21.02.1. All works fine on 19.07.8 version

sanitariu, 2021-11-26:
I have the same problem with mediatek 7628 - TP-Link 840 version 6.0/6.1/6.2/14
Seems like the same bug on heavy traffic is reintroduced again.
Heavy torrent traffic breaks the router. You loose connection and after 1-2 minutes you are back again.
Maybe nbd should look inside this...

Linaro1985, 2021-11-26:
Is there thermal support in MT7628 driver? I think it's because of high temperature on chip.

Linaro1985, 2021-11-26:
I think it's because of high chip temperature under heavy load.

cat /sys/kernel/debug/ieee80211/phy0/mt76/reset
             TX hang: 1
   TX DMA busy stuck: 0
   RX DMA busy stuck: 0
        Beacon stuck: 2
   RX PSE busy stuck: 67
            MCU hang: 0
    PSE reset failed: 0

It's work more longer if limit tx power to 5 (option txpower '5')

sanitariu, 2021-11-26:
Maybe this is possible but txpower 5 means you loose the router in the next room. Old versions and original somehow manage this.

Linaro1985, 2021-11-26:
>> txpower 5 means you loose the router in the next room
But my phone show this level after two walls in my home (6 meters approx distance)
[image]

r0ck3r, 2021-12-04:
Same bug for me on Mi Router 3 Pro with 21.02

nbd168, 2021-12-04:
Did you test the latest 21.02 branch?

r0ck3r, 2021-12-04:
Didn't yet

myapple, 2021-12-06:
I have test latest 21.02 and 19.07 , have similar situation.

I used  two same  mt7628 samples,  test using  iperf3 for one dut at one time
pc1 --- dut A ---  pc2 
pc1 --- dut B ---  pc2 

and I find if run uplink and downlink traffic simultaneously， the dut hang In a minute or two
if run traffic just in one direction，both dut can work ok for a long time， 30 minutes or more

and I add same logs in  mt7603/muc.c ,  found the main issue is  MCU HANG ,  and reset not work ,  just reboot device can make wifi work again 

also I have checked /sys/kernel/debug/ieee80211/phy0/mt76/reset
             TX hang: 4
   TX DMA busy stuck: 0
   RX DMA busy stuck: 0
        Beacon stuck: 19
   RX PSE busy stuck: 45
            MCU hang: 78
    PSE reset failed: 0

---

and I have test several other mt7628 samples ,  A small number of samples can work fine without any hangs.

Linaro1985, 2021-12-08:
@nbd168 I'm testing during 2 days latest 21.02 snapshot and no problems now. I will continue to watch.

Linaro1985, 2021-12-11:
Hi @nbd168 . The hang problem is still exists on latest 21.02 snapshot:
root@kndrt31r3:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/reset
             TX hang: 1
   TX DMA busy stuck: 0
   RX DMA busy stuck: 0
        Beacon stuck: 14
   RX PSE busy stuck: 15
            MCU hang: 0
    PSE reset failed: 0

Linaro1985, 2021-12-26:
Hi @nbd168 . I made some investigation, more information here: https://forum.openwrt.org/t/mt7628-dsa-driver/114772/9?u=linaro

Areopagitics, 2022-01-25:
will there be an update soon? I'm using https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt76x8/openwrt-21.02.1-ramips-mt76x8-tplink_tl-wr841n-v13-squashfs-tftp-recovery.bin and it still disconnects unless I uncheck WMM mode, but the wifi speed is now under 20mb/s

biboc, 2024-04-19:
@nbd168 
Any update on this? 
I also get the messages
I run OpenWrt 23.05.2 on MT7628 Hilink board

I have 20 routers in a wifi mesh network, it worked great on Openwrt 19

I get lost of transmission fail OpenWrt 23.05.2 and other routers can't connect with this one anymore (ping is 4 to 10 sec!)
```
# cat /sys/kernel/debug/ieee80211/phy0/mt76/reset
             TX hang: 33
   TX DMA busy stuck: 0
   RX DMA busy stuck: 0
        Beacon stuck: 924
   RX PSE busy stuck: 0
            MCU hang: 427
    PSE reset failed: 0
    ```