openwrt/mt76 #630: MT7915E Totolink X5000R same MAC Address on 5 GHz different Devices

Wlanfr3ak, 2021-12-01:
I setup 2 Totolink X5000R with 802.11s mesh on 2,4 GHz and on both Devices its generate/use a different MAC. When i try it on the 5 GHz Interface its use the SAME MAC on DIFFERENT Devices!

Tested on:
OpenWRT Snapshot from 2021-11-30 (OpenWRT-SNAPSHOT r18244-101300b842

Example:

2,4 GHz MAC Device1: 5C:92:5E:A3:13:30
2,4 GHz MAC Device2: 5C:92:5E:A3:0E:A0
5 GHz MAC Device1: 00:0C:43:4A:21:4E
5 GHz MAC Device2: 00:0C:43:4A21:4E

Workaround: On one Router i setup an Client Interface and disable it, now the new Mesh Interfaces gets this MAC:  02:0C:43:4A:21:4E

dangowrt, 2021-12-01:
> 5 GHz MAC Device1: 00:0C:43:4A:21:4E
> 5 GHz MAC Device2: 00:0C:43:4A21:4E

@hauke @981213 This looks like the MAC address in EEPROM data is a dummy and shouldn't be used. We may need to use nvram mac-address or other means of overriding it on this device.

ptpt52, 2021-12-01:
jus as what I did here to resolve such case of issue
https://github.com/openwrt/mt76/pull/476

HS1542, 2022-11-28:
I see this exact behavior on 2 identical X5000R running 22.03.2, the 5GHz MAC addresses are static set to 00:0C:43:4A:21:4E on both devices.
And I am unable to configure another MAC address on one of the 5GHz interfaces.

981213, 2023-03-26:
Fixed in https://github.com/openwrt/openwrt/commit/212632540043cc9b911c2efb86156479f2710836

closed 2023-03-26 by 981213