openwrt/mt76 #634: Archer A6 V3: 5GHz stopped working, failing with EIO

proski, 2021-12-05:
I flashed Archer A6 V3 with OpenWRT 21.02.1 originally. 5GHz wasn't working. I got it to work by setting the channel to 36. Then I started tweaking other settings in Luci (like WPA3, power etc), and the 5GHz radio stopped working. I was unable to restore the settings to make 5GHz work again, but I haven't tried a reset to defaults.
Then I upgraded the router to the December 3 snapshot, r18270-fc4398fe71. 5 GHz is still not working. The network device wifi1 is down. If I try to bring it up, I get EIO.
```
# ip link set up wlan1
ip: SIOCSIFFLAGS: I/O error
```
I discovered that the `kmod-mt7615-firmware` package wasn't installed, so I installed it with opkg, but it made no difference, even after reboot. There are no error messages in the kernel log when the driver if reloaded and an attempt is made to bring the interface up.
```
root@Archer6:~# rmmod mt7615e
root@Archer6:~# modprobe mt7615e
root@Archer6:~# ip link set up wlan1
ip: SIOCSIFFLAGS: I/O error
root@Archer6:~# dmesg | tail -n 5
[   21.205584] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   23.075785] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[   23.083029] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[ 1105.096121] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=23
[ 1105.112859] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
```
The only message containing mt7615 in the kernel log is:
```
root@Archer6:~# dmesg | grep -i mt7615
[   11.712050] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
```
Country is set to US on both WiFi interfaces.

Full kernel log and /etc/config/wireless are attached.

[dmesg.txt](https://github.com/openwrt/mt76/files/7656451/dmesg.txt)
[wireless.txt](https://github.com/openwrt/mt76/files/7656452/wireless.txt)

nbd168, 2021-12-05:
Please show me the MAC address of wlan1

proski, 2021-12-05:
e4:c3:2a:a3:31:c5

jtommasi72, 2021-12-06:
Same problem here. I tried reset to defaults, but still not able to make 5 Ghz WLAN to come up.

root@OpenWrt:~# cat /sys/class/ieee80211/phy1/macaddress
00:5f:67:9a:12:88
root@OpenWrt:~#

jtommasi72, 2021-12-06:
Same problem here. I tried reset to defaults, but still not able to make 5 Ghz WLAN to come up.

`root@OpenWrt:~# cat /sys/class/ieee80211/phy1/macaddress`
`00:5f:67:9a:12:88`
`root@OpenWrt:~#`

jtommasi72, 2021-12-06:
Same problem here. I tried reset to defaults, but still not able to make 5 Ghz WLAN to come up.

`root@OpenWrt:# cat /sys/class/ieee80211/phy1/macaddress
00:5f:67:9a:12:88
root@OpenWrt:#`

jtommasi72, 2021-12-06:
Same problem here. I tried reset to defaults, but still not able to make 5 Ghz WLAN to come up.

```
root@OpenWrt:# cat /sys/class/ieee80211/phy1/macaddress
00:5f:67:9a:12:88
root@OpenWrt:#
```

d-me3, 2021-12-07:
I faced this issue once with Archer C6 v3.2 (same hardware as the A6 v3). A soft reset did not work. I was able to restore the 5Ghz radio only by doing a hard factory reset:

https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset#hard_factory_reset

proski, 2021-12-07:
@d-me3: thank you! It looks like the hard reset just resets the settings, not the hardware. So it should be possible to find that bad setting by comparing all files before and after.

I'm going to try it. The problem is that I'm using that router currently as it has an excellent 2.4GHz coverage, so I'll need to swap it out first. I'm actually puzzled that the driver is silent about the error. One would think that a failure to bring up the interface would warrant a kernel log message. Also loading the firmware (or perhaps a failure to load it) should be logged. I have no idea what the driver is doing, `modinfo mt7615e` doesn't show any parameters to enable debugging.

System log shows that hostapd tries to bring up the interface and fails, which reassures me that bringing up the interface is where things go wrong.

```
Tue Dec  7 11:25:10 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy2.conf (phy wlan2) --> new PHY
Tue Dec  7 11:25:10 2021 daemon.err hostapd: Could not set interface wlan2 flags (UP): I/O error
Tue Dec  7 11:25:10 2021 daemon.err hostapd: nl80211: Could not set interface 'wlan2' UP
Tue Dec  7 11:25:10 2021 daemon.notice hostapd: nl80211: deinit ifname=wlan2 disabled_11b_rates=0
Tue Dec  7 11:25:10 2021 daemon.err hostapd: nl80211 driver initialization failed.
Tue Dec  7 11:25:10 2021 daemon.notice hostapd: wlan2: CTRL-EVENT-TERMINATING
Tue Dec  7 11:25:10 2021 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan2 wasn't started
Tue Dec  7 11:25:10 2021 daemon.notice netifd: radio1 (13518): Command failed: Invalid argument
Tue Dec  7 11:25:10 2021 daemon.notice netifd: radio1 (13518): Device setup failed: HOSTAPD_START_FAILED
Tue Dec  7 11:25:10 2021 daemon.notice netifd: Wireless device 'radio1' set retry=0
Tue Dec  7 11:25:10 2021 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Tue Dec  7 11:25:10 2021 daemon.notice netifd: Wireless device 'radio1' is now down
```

`wlan1` became `wlan2` after reloading the module.

d-me3, 2021-12-07:
@proski, since Archer C6 v3 has a reset button, I did the "hard reset" by long pressing it for a couple of seconds. It might have reset something else other than the settings. It is also possible that OpenWRT might do some hardware initialization only when recreating the jffs2 partition, but this is only a guess.

proski, 2021-12-07:
Sure, there are so many things I want to try, and that will be one of them.
By the way, the MAC address keeps incrementing by 2 every time I reload the `mt7615e` module:
e4:c3:2a:a3:31:c5, e4:c3:2a:a3:31:c7 etc, now it's e4:c3:2a:a3:31:cd.
The sticker on the router says: "MAC: E4-C3-2A-A3-31-C1". That's the MAC address of eth0.
I also tried reloading `mt7615_common` with `prefer_offload_fw=false`. It made no difference.

SiNONiMiTY, 2021-12-18:
Confirming this issue on the Archer A6 V3. I am on the latest snapshot as of this posting. I/O error appears on the syslog. Haven't tried a full factory reset yet.

proski, 2021-12-27:
Upgrading to the December 25 snapshot (r18404-236c3ea730) with Linux kernel 5.10.87 made no difference.

SiNONiMiTY, 2021-12-27:
Hi @proski 

Have you tried the hard factory reset method? I haven't tried it yet since I need connectivity.

proski, 2022-01-01:
I tried reset using Luci (System -> Backup / Flash Firmware -> Perform Reset). Once I set up the 5 GHz WiFi again, it failed with the same symptoms. But then I power-cycled the router using the power button on the back of the router, and the 5 GHz WiFi started working! I've tried everything I was trying initially, but the interface keeps working.

We have a workaround now. Of course, it would be better if the driver could initialize the device from that confused state. Even a better kernel log message would have been helpful.

SiNONiMiTY, 2022-01-01:
I see, so we need a hard factory reset, not just the soft one. Will do this at a non-critical time and confirm here if this workaround works. Thank you @proski.

proski, 2022-01-01:
It's possible that power cycling the router without a reset would work. I missed an opportunity to test it.

SiNONiMiTY, 2022-01-07:
I tried the hard factory reset method, it worked. 5Ghz no longer stalls. Running on the latest openwrt snapshot.

vinaobraga, 2022-02-11:
Hello. I'm running my router in the last version available, and the problem still occurs. I tried to reset to factory but still same issue. But in my case it is the 2.4GHz not working

Hostname | OpenWrt
Model | Netgear R6220
Architecture | MediaTek MT7621 ver:1 eco:3
Firmware Version | OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-22.025.79177-4ec18ec
Kernel Version | 5.4.154
Local Time | 2022-02-11 17:00:39

SiNONiMiTY, 2022-02-11:
@vinaobraga the fixes are on the SNAPSHOT versions, not on the stable one. If you are not familiar in installing SNAPSHOT versions, search for guides. If you are not confident enough, you have to wait for the next stable release.

vinaobraga, 2022-02-11:
tks for the information @SiNONiMiTY

HausnerR, 2022-03-12:
Hello, I have the same problem on another router: Totolink A7000R.
Latest openwrt, checked it after hard factory reset (I just enabled wifi) and problem exists.

When router is started by putting plug in socket, everything is fine and wifi starts like it should.
When I do reboot by software, radio0 cant get up. Nothing help.

@SiNONiMiTY, can you link to SNAPSHOTs you find bug is fixed?

Best regards,
Kuba

SiNONiMiTY, 2022-03-13:
@HausnerR I am on the latest snapshot and this is working good. Make sure that you perform a HARD factory reset and do the 30-30-30 reset method.

HausnerR, 2022-03-13:
Ok, hard reset not helped.

I found solution in other thread here: https://github.com/openwrt/mt76/issues/316#issuecomment-727223566

Hope it helps someone else :)

Borromini, 2022-04-30:
@proski Your device has a MT7613BE radio, it doesn't use the mt7615 driver but the mt7663 one.