openwrt/mt76 #636: crash log on mt7603+mt7612

ptpt52, 2021-12-10:
log1:
```
<6>[164308.266546] ralink_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[167462.695917] ralink_soc_eth 1e100000.ethernet eth0: port 2 link up
<6>[167462.976208] ralink_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[167465.169850] ralink_soc_eth 1e100000.ethernet eth0: port 2 link up
<6>[167491.136058] ralink_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[167493.801837] ralink_soc_eth 1e100000.ethernet eth0: port 2 link up
<6>[168728.952502] ralink_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[200972.511374] ralink_soc_eth 1e100000.ethernet eth0: port 3 link up
<1>[203114.071260] CPU 3 Unable to handle kernel paging request at virtual address 000007b8, epc == 8d039188, ra == 8d039840
<4>[203114.081960] Oops[#1]:
<4>[203114.084316] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 5.4.159 #0
<4>[203114.090378] $ 0   : 00000000 00000001 8fc13dc8 8b699450
<4>[203114.095672] $ 4   : 00000480 8fc13dc4 00000000 8f0c8d80
<4>[203114.100961] $ 8   : 00000001 00000008 000000ff 8eaa2480
<4>[203114.106250] $12   : 000000ff 8eaa2e50 00000000 00009a94
<4>[203114.111538] $16   : 8f0c8d80 00000480 8d0e8c40 8eaa2cc0
<4>[203114.116827] $20   : 8eaa2bc4 00000003 8b71cab0 00000002
<4>[203114.122116] $24   : 00000001 805d2664                  
<4>[203114.127406] $28   : 8fc66000 8fc13d78 00000000 8d039840
<4>[203114.132697] Hi    : 0000000a
<4>[203114.135646] Lo    : 66666669
<4>[203114.138749] epc   : 8d039188 ieee80211_check_fast_xmit+0x534/0x9c0 [mac80211]
<4>[203114.145985] ra    : 8d039840 ieee80211_get_buffered_bc+0xc0/0x1ac [mac80211]
<4>[203114.153084] Status: 11007c03	KERNEL EXL IE 
<4>[203114.157338] Cause : 40800008 (ExcCode 02)
<4>[203114.161410] BadVA : 000007b8
<4>[203114.164358] PrId  : 0001992f (MIPS 1004Kc)
<4>[203114.168515] Modules linked in: ksmbd qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ts_fsm ts_bm tcp_hybla tcp_bbr slhc sch_cake r8152 poly1305_mips ntfs3 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log
<4>[203114.168701]  nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libblake2s_generic ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat chacha_mips cdc_wdm br_netfilter asn1_decoder natflow natcap fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set
<4>[203114.255368]  ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos bonding ip6_gre ip_gre gre ifb nat46 ip6_udp_tunnel udp_tunnel sit ip6_tunnel tunnel6 tunnel4 ip_tunnel rpcsec_gss_krb5 auth_rpcgss oid_registry tun vfat fat lockd sunrpc grace autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes cts ctr cmac ccm cbc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform
<4>[203114.341885]  ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic
<4>[203114.445517] Process swapper/3 (pid: 0, threadinfo=610fb311, task=c0d4275d, tls=00000000)
<4>[203114.453657] Stack : 8fc13e20 8f53b044 8d07a078 00000000 8f0c8d80 00000480 8d0e8c40 8eaa2cc0
<4>[203114.462075]         8eaa2bc4 00000003 8b71cab0 00000002 00000000 8d039840 8d0ea03c 8eaa2e50
<4>[203114.470488]         8f0c8d80 000000ff 00000000 00000000 8f0c80c0 8f0c80c0 00000001 00000000
<4>[203114.478896]         8d0e8c40 8eaa2480 00000000 8f702c00 00000100 00000004 8fc13ecc 8eaa2e48
<4>[203114.487305]         00000006 00000004 00000003 8fc13ecc 00000001 8d0f8110 00000001 00000000
<4>[203114.495713]         ...
<4>[203114.498245] Call Trace:
<4>[203114.500905] [<8d039188>] ieee80211_check_fast_xmit+0x534/0x9c0 [mac80211]
<4>[203114.507913] Code: afb60028  afb50024  afb40020 <8c940338> aca20004  aca20008  aca0000c  aca00010  aca0001c 
<4>[203114.517714] 
<4>[203114.519694] ---[ end trace 068e079e0b0a7874 ]---
root@X-WRT:~# opkg list | grep mt76
kmod-mt76-core - 5.4.159+2021-11-26-20211126-4
kmod-mt7603 - 5.4.159+2021-11-26-20211126-4
kmod-mt76x02-common - 5.4.159+2021-11-26-20211126-4
kmod-mt76x2 - 5.4.159+2021-11-26-20211126-4
kmod-mt76x2-common - 5.4.159+2021-11-26-20211126-4
```

siwind, 2021-12-15:
ksmbd  not stable?

Neustradamus, 2022-08-01:
@ptpt52: It has been solved?