openwrt/mt76 #637: mesh never works

ptpt52, 2021-12-11:
config
```
config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'mesh'
	option encryption 'none'
	option mesh_id 'x-wrt'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option ifname 'mesh0'
	option network 'lan'
```

log:
```
Mon Dec  6 12:01:21 2021 kern.info kernel: [  295.764627] br-lan: port 4(mesh0) entered blocking state
Mon Dec  6 12:01:21 2021 kern.info kernel: [  295.764672] br-lan: port 4(mesh0) entered disabled state
Mon Dec  6 12:01:21 2021 kern.info kernel: [  295.769450] device mesh0 entered promiscuous mode
Mon Dec  6 12:01:22 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:23 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:24 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:25 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:26 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:27 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:28 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:29 2021 daemon.notice wpa_supplicant[2998]: mesh0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Dec  6 12:01:32 2021 daemon.err wpa_supplicant[2998]: mesh0: Could not join mesh
```

each node use the same config
and I notice that mesh0 and wlan1-1 share the same mac address
```
root@X-WRT:~# iwinfo mesh0 info
mesh0     ESSID: "x-wrt"
          Access Point: XX:XX:XX:B4:53:C0
          Mode: Mesh Point  Channel: unknown (unknown)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -88 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 168C:003C 168C:4019 [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
root@X-WRT:~# iwinfo wlan1-1 info
wlan1-1   ESSID: "X-WRT_53C1"
          Access Point: XX:XX:XX:B4:53:C0
          Mode: Master  Channel: 11 (2.462 GHz)
          Center Channel 1: 11 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -88 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 168C:003C 168C:4019 [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

I know this may no issue of mt76 but issue on mac80211 or somewhat.

ptpt52, 2021-12-11:
same test on mt7915/ mt7603 / mt7612, same issue.

ryderlee1110, 2021-12-13:
never? I remember I can run mesh couple months ago.

ptpt52, 2021-12-13:
@ryderlee1110  sure now mesh cannot work.
tested on mt76 or ath10k
could you try it and make it confirm?

ptpt52, 2021-12-28:
looks like that mt7603 do not support 80211s?

myapple, 2021-12-28:
mesh on 7628/7603/7610/7612 works fine
but ，tx stucks for 7613 when run in mesh mode
@ptpt52

LorenzoBianconi, 2022-01-11:
@ptpt52 just tested mesh on 7915 link and it works fine for me.

ptpt52, 2022-01-12:
thanks
7915, 7612 works, I have to fix the channel

Djfe, 2024-02-09:
@ptpt52 Shall this be closed then?