openwrt/mt76 #638: MT7613 tx hangs when running  mesh mode

myapple, 2021-12-17:
MT7613 tx hangs when running  mesh mode, but it work ok in AP mode
```
cat /sys/kernel/debug/ieee80211/phy1/mt76/xmit-queues
     queue | hw-queued |      head |      tail |
         0 |         0 |       138 |       138 |
         1 |       111 |       178 |        67 |
         2 |       480 |       401 |       433 |
         3 |         0 |         0 |         0 |
         4 |         0 |         2 |         2 |
```

mpowell3bb, 2021-12-20:
I see the same thing. Have been following #518 because it seems similar.

In my scenario, it is queue 0 that gets stuck.

nbd168, 2022-06-25:
Please try the latest version

an-kaly, 2022-07-12:
Mesh function is stable now on 22.03 snapshot on Tplink archer C6U