openwrt/mt76 #640: 80211s mesh speedtest

ptpt52, 2021-12-28:
node1:
```
mt7915 11AC
```

node2:
```
mt7612e 11AC
```

```
top:
client----(node1)----mesh----(node2)---internet
                                                         |
iperf3 test:
350Mbps: client<-----(node1)
15Mbps: client<-----(node1)<----(node2)
110Mbps: (node1)<-----(node2)
```

123testing1237, 2022-01-06:
@nbd168 He is autistic and dumb.

These drivers are garbage and at the DD-WRT level, every day adding a commit that "supposedly" fixes something, but 10 things break.

@nbd168 🤢🤮 
mt76 🤢🤮

123testing1237, 2022-01-06:
@nbd168 is autistic and dumb.

These drivers are garbage and at the DD-WRT level, every day adding a commit that "supposedly" fixes something, but 10 things break.

@nbd168 🤢🤮 
mt76 🤢🤮

ptpt52, 2022-01-14:
hi, I notice that it is the issue related to channel
fixed-channel need to be set for high speed test.

nscnd, 2022-04-14:
Hi,
Seem I have the same problem whith a similar setup (mt7915e AX mesh point talking to an AC only mesh point) 
what do you mean by "fixed-channel" ?

phil2sat, 2022-04-21:
means both mesh devices need to be set on a wifi channel, not one eg. on 36 and the second to auto.
both needs to be on channel 36