openwrt/mt76 #642: MT7615 slow RX performance with pre-calib enabled

tarori, 2022-01-06:
I am using WRC-2533GST2 with latest 21.02 snapshot (MT7621A + MT7615N*2). Thanks for developing, both 2.4GHz and 5GHz connection is stable.
The problem is 2.4GHz RX is slow. TX is about 110Mbps, which is almost same as stock firmware. But RX is about 30 ~ 40Mbps, when stock firmware gets 110Mbps.
$ iw wlan0 station dump
shows rx bitlate is only about 52Mbps though tx bitlate is 144Mbps.
[iw_stationdump.txt](https://github.com/openwrt/mt76/files/7821006/iw_stationdump.txt)

I enabled online-calibration by the patch attached, then RX speed becomes 110Mbps.
What is the cause? Is there anyone having same problem ?

[203-disable-pre-calib.patch.txt](https://github.com/openwrt/mt76/files/7820984/203-disable-pre-calib.patch.txt)

tarori, 2022-01-06:
my eeprom and otp file
[eeprom.zip](https://github.com/openwrt/mt76/files/7821447/eeprom.zip)
.

tarori, 2022-01-06:
my eeprom and otp file
MAC address and power limit are read correctly.
[eeprom.zip](https://github.com/openwrt/mt76/files/7821452/eeprom.zip)

ptpt52, 2022-01-11:
to follow this thread

nbdSorry, 2022-01-13:
Sorry @nbd168 for insulting you.

[image]

tarori, 2022-01-15:
This is because some types of pre-cal are not supported.
MT7615 ships with pre-cal such as TXLPF and RXFD, but the current mt76 only applies rx-dcoc and tx-dpd. This is insufficient for some boards.
I added some code to apply them, and now it works properly.

rodsmar, 2022-01-16:
Hey, @tarori 

Can you share this code (TXLPG, RXFD, and others) with us?

tarori, 2022-01-17:
@rodsmar 
You can check the calibration with the patch attached to the first comment. In most cases, online calibration is enough.
I'm also having trouble with a 5GHz VHT80 with a slow TX speed of about 200-300Mbps. Probably because the MT7621A is weak. I'm looking for something that uses a lot of CPU.

castiel652, 2022-01-18:
MT7615 currently doesn't support hw tx aggregation and encap offloading.
That's probably why TX path is slower.

tarori, 2022-02-06:
In my case, the slow tx throughput was due to the client device. I used iPhone13 as a client and was able to get 620Mbps mt7615->client and 700Mbps client->mt7615 with VHT80.
I will close this issue after sending the pre-cal patch.

csharper2005, 2024-02-10:
The same problem TP-Link EC330-g5u (MT7621A + MT7615N*2). 23.05.2.

```
        inactive time:  10 ms
        rx bytes:       2504194
        rx packets:     14335
        tx bytes:       14293152
        tx packets:     15372
        tx retries:     366
        tx failed:      0
        rx drop misc:   0
        signal:         -28 [-36, -32, -41, -31] dBm
        signal avg:     -28 [-35, -31, -41, -31] dBm
        tx bitrate:     144.4 MBit/s MCS 15 short GI
        tx duration:    3646952 us
        rx bitrate:     65.0 MBit/s MCS 6 short GI
        rx duration:    2019894 us
        airtime weight: 256
        expected throughput:    112.243Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            yes
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 2357 seconds
        associated at [boottime]:       832.827s
        associated at:  1707571128520 ms
        current time:   1707573485234 ms
```

@tarori have you already sent the patch?

tarori, 2024-03-05:
@csharper2005 Sorry for the late response, I was locked out from my github account...
No, I don't have a plan to send the patch because:
- I'm not using this device anymore. I've bought a new device with MT7622 and MT7915, and as far as I know, it doesn't have pre-cal eeprom. It is working fine without any patch.
- The patch I wrote for the test purpose was copy and paste from the proprietary mtk software found on github. I don't have offical right to access the proprietary software or any of the hardware information, so I totally don't know if it is working correctly or not.
Anyway, you can check the simple patch on the first comment.

This is an issue, not PR. So, I hope anyone who has right access and knowledge can fix that...

csharper2005, 2024-04-07:
@tarori thanks. I had low RX performance in both bands (TP-Link EC330-g5u, 2x MediaTek MT7615E) and I patched eeprom 0x52 (MT_EE_CALDATA_FLASH) value instead of the driver patch. Default value was 0x06, the new one is 0x00. This workaround is also working:
[image]

@nbd168 do you have any idea about solution or better workaround (e.g. 0x52 patching at runtime or via dts)?