openwrt/mt76 #644: Driver Timeout at boot, dual MT7915E, UniElec U7621-06 (16M flash) Openwrt

klojcnik, 2022-01-28:
Boot from poweroff works perfect, when I reboot the system the driver seems to have a problem. If i only use one of the two modules, everything is alright.

rmmod mt7915e
modprobe 7915e
--> same error

If I unplugg 12V and put it back in, the two radios work perfect again. if i click reboot it happens again.
--> driver only works once after poweroff reset --> softreset leads to that error

Same thing on Openwrt 20.02.1 and current snap (28.01.22)

Thu Jan 27 16:47:10 2022 user.info kernel: [   13.779669] kmodloader: loading kernel modules from /etc/modules.d/*
Thu Jan 27 16:47:10 2022 kern.info kernel: [   14.089957] Loading modules backported from Linux version v5.15.8-0-g43e577d7a2cb
Thu Jan 27 16:47:10 2022 kern.info kernel: [   14.097482] Backport generated by backports.git v5.15.8-1-0-g83f664bb
Thu Jan 27 16:47:10 2022 kern.info kernel: [   14.815303] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=24
Thu Jan 27 16:47:10 2022 kern.info kernel: [   14.820778] pci 0000:00:01.0: enabling device (0000 -> 0003)
Thu Jan 27 16:47:10 2022 kern.info kernel: [   14.826504] **mt7915e 0000:02:00.0: enabling device (0000 -> 0002)**
Thu Jan 27 16:47:10 2022 kern.err kernel: [   35.183619] mt7915e 0000:02:00.0: Message 00000010 (seq 1) timeout
Thu Jan 27 16:47:10 2022 kern.err kernel: [   35.189802] mt7915e 0000:02:00.0: Failed to get patch semaphore
Thu Jan 27 16:47:10 2022 kern.err kernel: [   55.663612] mt7915e 0000:02:00.0: Message 00000010 (seq 2) timeout
Thu Jan 27 16:47:10 2022 kern.err kernel: [   55.669786] mt7915e 0000:02:00.0: Failed to get patch semaphore
Thu Jan 27 16:47:10 2022 kern.warn kernel: [   55.675942] mt7915e: probe of 0000:02:00.0 failed with error -11
Thu Jan 27 16:47:10 2022 kern.info kernel: [   55.682031] mt7621-pci 1e140000.pcie: bus=3 slot=2 irq=25
Thu Jan 27 16:47:10 2022 kern.info kernel: [   55.687480] pci 0000:00:02.0: enabling device (0000 -> 0003)
Thu Jan 27 16:47:10 2022 kern.info kernel: [   55.693125] mt7915e 0000:03:00.0: enabling device (0000 -> 0002)
Thu Jan 27 16:47:10 2022 kern.err kernel: [   76.143606] mt7915e 0000:03:00.0: Message 00000010 (seq 1) timeout
Thu Jan 27 16:47:10 2022 kern.err kernel: [   76.149787] mt7915e 0000:03:00.0: Failed to get patch semaphore
Thu Jan 27 16:47:10 2022 kern.err kernel: [   96.623585] mt7915e 0000:03:00.0: Message 00000010 (seq 2) timeout
Thu Jan 27 16:47:10 2022 kern.err kernel: [   96.629743] mt7915e 0000:03:00.0: Failed to get patch semaphore
Thu Jan 27 16:47:10 2022 kern.warn kernel: [   96.635884] mt7915e: probe of 0000:03:00.0 failed with error -11
Thu Jan 27 16:47:10 2022 kern.info kernel: [   96.699024] PPP generic driver version 2.4.2
Thu Jan 27 16:47:10 2022 kern.info kernel: [   96.705681] NET: Registered protocol family 24
Thu Jan 27 16:47:10 2022 user.info kernel: [   96.718951] kmodloader: done loading kernel modules from /etc/modules.d/*

brunosalezze, 2022-07-02:
I have the same issue, if you have openwrt in a vm is even worst, since it cant load the MT7915 sucessfully at all.

klojcnik, 2022-07-02:
The strange Thing is, the 7915 dual 2x2 2.4 and 5ghz works fine.
But little different setup, unielec 7523, and custom build with kernel
5.15.43

Greetings

Klojsch

brunosalezze, 2022-07-02:
I have the  7915 dual 2x2 2.4 and 5ghz, and I was not able yet to make it work in a VM or after a rmmod and modprobe.

klojcnik, 2022-07-02:
In VM? You can forget that, you need PCIE Forward from the root system.
As far as i know its only availible for AMD GPUs. I already wasted 2 days
of my life to find that out 6 months ago ;-)

brunosalezze, 2022-07-02:
Ok.. I'm using PCI PT with IOMMU on a Intel System, maybe its  better to run barebones afterall. Stragely, this seems to work with with older wifi cards

klojcnik, 2022-07-02:
Yeah seems to be a driver prob, forward worked for me to. Could see the
card with lspci. But as far as i understand, you need a special driver on
the root system for communication. If the card would work with MHI it would
be doable in theory. Plus the forward is often only PCIE GEN 1 ( why have
no idea, but thats the case for vm Ware ).

klojcnik, 2022-07-02:
I have a dedecated System now for develope, makes life easy:-)
Intel celeron on a 1151 mainboard matx with 4gb ram and 4 pcie ports ( one
3.0 x16, 3 x1 ) for about 120 Euro:-)

Juli 2022, 23:18:

brunosalezze, 2022-07-03:
I was able to make it work with a openwrt inside a proxmox by using IOMMU HW PT with a boot on startup option. I had to unplug power and power it again and the radios worked in OpenWRT and survived a VM reboot.

klojcnik, 2022-07-03:
https://unix.stackexchange.com/questions/73908/how-to-reset-cycle-power-to-a-pcie-device

Never tried it.

klojcnik, 2022-10-11:
Awsome! You could try reset power on the PCIE Port. I dont know if openwrt
can do that from inside vm. Depends on hw access.

cjdelisle, 2023-04-12:
I believe this issue is the same one I am having which I have reported here: https://github.com/openwrt/openwrt/issues/12359

The hint is `pci 0000:00:01.0: enabling device (0000 -> 0003)` which means the PCI bridge is coming up *without* bus mastering enabled.

klojcnik, 2023-04-12:
> here: openwrt/openwrt#12359

I also have a VM setup with openwrt as guest under Ubuntu 22.04 with 6.1
Kernel. Machine is a HP Z240 SFF, I7 6700, 40gb ddr4 2400.
KVM/QEMU as Hypervisor. PCIE passthrough to MT7916 and the intel onboard
nic.

Also had some trouble after a few days that devices could no longer connect
or could connect but where kicked after some seconds. That is with snapshot
and testfw.

I build myself a 22.03 image on 5.10 Basis. Now everything works perfect
but at that state no 6e is availible.

Sadly nothing in the logs that would indicate why that happens.

2023, 10:52: