openwrt/mt76 #645: MT7615E fails to authenticate with certain APs

kristrev, 2022-02-01:
Hello,

I am currently trying to use MT7615E as a client and hit a kernel oops when connecting to certain APs.  My board is a based on mt7622 and is running a recent OpenWRT snapshot (5d110c016184). I have also manually updated mt76 to commit 9b2ac62d6f31.

Right after I reload the wifi configuration or click on "Submit" in Luci, I see the kernel oops below. The error only occurs when I connect to certain APs, and I have tried different channels, bandwidths, etc. without any luck. My current reliable reproducer is to use another  mt7622/mt7615e board as an AP (running/using the same version of OpenWRT/mt76).

If I connect the board I use as a client to other APs, everything works fine (using the same channel, bandwidth, ...) - I am able to connect, send traffic, etc. I don't have that many other APs available, but I am able to connect with mt7615e to APs where the 5Ghz readio is mt7612 or ath10k.

Doe anyone know what might be wrong or where to start looking? I tried to look at the function at the end of the oops, but I wasnt able to make such sense of it. Thanks in advance for any help!

## Kernel oops

```
[  744.877821] wlan0: authenticate with 00:0c:43:26:60:00
[  745.377179] ------------[ cut here ]------------
[  745.381841] WARNING: CPU: 1 PID: 1645 at sta_info_insert_rcu+0x58/0x670 [mac80211]
[  745.389399] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat seqiv leds_gpio xhci_plat_hcd gpio_button_hotplug
[  745.444274] CPU: 1 PID: 1645 Comm: wpa_supplicant Tainted: G S      W         5.10.92 #0
[  745.452353] Hardware name: Celerway M5.3-E5L2W2 (DT)
[  745.457310] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[  745.463333] pc : sta_info_insert_rcu+0x58/0x670 [mac80211]
[  745.468821] lr : sta_info_insert_rcu+0x28/0x670 [mac80211]
[  745.474296] sp : ffffffc0116035f0
[  745.477600] x29: ffffffc0116035f0 x28: 0000000000000000 
[  745.482905] x27: ffffff8002262270 x26: 0000000000000000 
[  745.488209] x25: 0000000000000004 x24: ffffff8002260fe0 
[  745.493513] x23: ffffffc011603730 x22: ffffff800213e000 
[  745.498817] x21: ffffff8002260fe0 x20: ffffff80022616e0 
[  745.504121] x19: ffffff800213e000 x18: 0000000000000000 
[  745.509425] x17: 000000000000000e x16: ffffffc0107b83e0 
[  745.514729] x15: 000000000000002f x14: 0000000000000000 
[  745.520034] x13: 00000000000000fb x12: 00000000bd08a39e 
[  745.525337] x11: 0000000000000000 x10: 00000000000002f1 
[  745.530641] x9 : 0000000000000001 x8 : 0000000000000000 
[  745.535945] x7 : 0000000000000001 x6 : 000000000000ffff 
[  745.541248] x5 : ffffffc00892d484 x4 : 0000000026430c00 
[  745.546552] x3 : 0000000000000060 x2 : 0000000000000000 
[  745.551856] x1 : ffffff8002b9e8c0 x0 : 0000000000000000 
[  745.557161] Call trace:
[  745.559611]  sta_info_insert_rcu+0x58/0x670 [mac80211]
[  745.564751]  sta_info_insert+0x10/0x30 [mac80211]
[  745.569456]  ieee80211_chswitch_done+0x1774/0x2330 [mac80211]
[  745.575202]  ieee80211_mgd_auth+0x1ec/0x344 [mac80211]
[  745.580342]  ieee80211_color_change_finish+0xb74/0x2020 [mac80211]
[  745.586523]  cfg80211_mlme_auth+0xd4/0x130 [cfg80211]
[  745.591571]  cfg80211_vendor_cmd_reply+0x34f8/0x4190 [cfg80211]
[  745.597485]  genl_family_rcv_msg_doit+0xb4/0x110
[  745.602094]  genl_rcv_msg+0xd0/0x1c0
[  745.605661]  netlink_rcv_skb+0x58/0x120
[  745.609488]  genl_rcv+0x34/0x50
[  745.612621]  netlink_unicast+0x1e0/0x2cc
[  745.616535]  netlink_sendmsg+0x1a0/0x3b4
[  745.620451]  ____sys_sendmsg+0x1f0/0x260
[  745.624364]  ___sys_sendmsg+0x78/0xc0
[  745.628017]  __sys_sendmsg+0x54/0xa0
[  745.631582]  __arm64_sys_sendmsg+0x20/0x30
[  745.635672]  el0_svc_common.constprop.0+0x60/0x110
[  745.640453]  do_el0_svc+0x1c/0x24
[  745.643763]  el0_svc+0x10/0x1c
[  745.646808]  el0_sync_handler+0x9c/0x120
[  745.650723]  el0_sync+0x148/0x180
[  745.654028] ---[ end trace 8ca048d7e1217acc ]---
[  745.658681] wlan0: failed to insert STA entry for the AP (error -22)
```

## uci (AP) configuration

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '36'
	option band '5g'
	option htmode 'VHT80'
	option cell_density '0'

config wifi-iface 'wifinet1'
	option device 'radio0'
	option mode 'sta'
	option network 'wwan'
	option ssid 'WifiTest'
	option encryption 'psk2'
	option key 'abcdefgh'
```

## wifi info
```
[    9.037007] mt7615e 0000:01:00.0: assign IRQ: got 138
[    9.042117] pci 0000:00:00.0: enabling bus mastering
[    9.047152] mt7615e 0000:01:00.0: enabling device (0000 -> 0002)
[    9.053231] mt7615e 0000:01:00.0: enabling bus mastering
[    9.417446] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[    9.997867] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   10.167564] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
[   10.184739] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
[   10.653837] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   11.692565] mt7622-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20190801210006a
[   11.692565] 
[   11.728054] PPP generic driver version 2.4.2
[   11.733713] NET: Registered protocol family 24
[   11.743490] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.798873] mt7622-wmac 18000000.wmac: N9 Firmware Version: 2.0, Build Time: 20200131180931
```

nbd168, 2022-02-01:
In your reproducer setup, please check the MAC addresses of the AP and the station. Could it be that they're the same on both devices?

kristrev, 2022-02-01:
@nbd168 Thanks a lot for your quick reply and I believe that is the issue. Will test and report back.

kristrev, 2022-02-01:
I am happy to report that issue was that the two boards had the same eeprom-content, including mac address. After modifying the driver to set a random MAC, wifi client worked just fine. Lesson learned, next time I will check that first :) Thanks again for the help.