openwrt/mt76 #646: kern.warn

Markomax78, 2022-02-02:
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.332928] ------------[ cut here ]------------
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.342200] WARNING: CPU: 1 PID: 1096 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-12-03-678071ef/mt7603/mac.c:208 0x860546a4 [mt7603e@8ad96439+0x96c0]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.371356] Modules linked in: ksmbd qcserial pppoe ppp_async option iptable_nat cdc_mbim xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pppox ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG usbserial usbnet usblp ts_kmp ts_fsm ts_bm slhc poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libblake2s_generic iptable_raw iptable_mangle iptable_filter ip_tables
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.371637]  crc_ccitt compat chacha_mips cdc_wdm cdc_acm asn1_decoder fuse ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel tun vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 seqiv jitterentropy_rng drbg md5 md4 kpp hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 mbcache jbd2 usbcore nls_base usb_common mii crc32c_generic crc32_generic
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.712619] CPU: 1 PID: 1096 Comm: napi/phy0-7 Not tainted 5.4.171 #0
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.725424] Stack : 00000000 80850000 00021c08 8007e1c0 00000000 00000000 00000000 00000000
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.742053]         00000000 00000000 00000000 00000000 00000000 00000001 86a13c00 ca8c99e8
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.758680]         86a13c98 00000000 00000000 00000000 00000038 805ecf24 65746e69 2e352064
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.775307]         00000000 00000229 00000000 000adfab 00000000 86a13be0 00000000 860546a4
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.791935]         00000009 00003a98 80000000 00000001 00000000 8035d420 00000004 80820004
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.808562]         ...
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.813418] Call Trace:
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.813432] [<8007e1c0>] 0x8007e1c0
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.825275] [<805ecf24>] 0x805ecf24
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.832235] [<860546a4>] 0x860546a4 [mt7603e@8ad96439+0x96c0]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.843662] [<8035d420>] 0x8035d420
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.850618] [<8000b05c>] 0x8000b05c
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.857551] [<8000b064>] 0x8000b064
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.864483] [<805d2864>] 0x805d2864
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.871414] [<8007ac68>] 0x8007ac68
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.878346] [<8002c410>] 0x8002c410
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.885293] [<860546a4>] 0x860546a4 [mt7603e@8ad96439+0x96c0]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.896723] [<8002c4b8>] 0x8002c4b8
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.903694] [<860546a4>] 0x860546a4 [mt7603e@8ad96439+0x96c0]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.915158] [<86107d5c>] 0x86107d5c [mac80211@a5bb298e+0x81d40]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.927000] [<86054840>] 0x86054840 [mt7603e@8ad96439+0x96c0]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.938449] [<86b961cc>] 0x86b961cc [mt76@16ef5283+0xac40]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.949382] [<86b93078>] 0x86b93078 [mt76@16ef5283+0xac40]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.960359] [<86051bc0>] 0x86051bc0 [mt7603e@8ad96439+0x96c0]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.971856] [<86b94f54>] 0x86b94f54 [mt76@16ef5283+0xac40]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.982890] [<86b916a8>] 0x86b916a8 [mt76@16ef5283+0xac40]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11134.993821] [<80069444>] 0x80069444
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.000829] [<8043d61c>] 0x8043d61c
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.007762] [<8043d504>] 0x8043d504
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.014709] [<805ee4a8>] 0x805ee4a8
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.021655] [<8043d6c0>] 0x8043d6c0
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.028588] [<8043d6dc>] 0x8043d6dc
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.035534] [<8004b724>] 0x8004b724
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.042528] [<8004bc34>] 0x8004bc34
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.049460] [<8004baf4>] 0x8004baf4
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.056418] [<8004baf4>] 0x8004baf4
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.063365] [<8004baf4>] 0x8004baf4
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.070297] [<80006718>] 0x80006718
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.077238]
Tue Feb  1 19:02:25 2022 kern.warn kernel: [11135.080765] ---[ end trace 262e0c61c61ce2e0 ]---

bananos, 2022-04-06:
I'm running `OpenWRT 21.02.2` on `TP-LINK Archer C20 v5` and recently started to experience problems with 2.4Ghz wifi. 
Here's a list of logs from `dmesg`:

```
[13513.552328] ------------[ cut here ]------------
[13513.557077] WARNING: CPU: 0 PID: 664 at target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2021-12-03-678071ef/mt7603/mac.c:208 0x836d4604 [mt7603e@9b0f47cb+0x9580]
[13513.571816] Modules linked in: pppoe ppp_async mt76x0e mt76x0_common iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio gpio_button_hotplug
[13513.626646] CPU: 0 PID: 664 Comm: napi/phy0-3 Not tainted 5.4.179 #0
[13513.633099] Stack : 00021c08 8005aeec 805d0000 805679e8 00000000 00000000 00000000 00000000
[13513.641594]         00000000 00000000 00000000 00000000 00000000 00000001 836ddc30 80939635
[13513.650090]         836ddcc8 00000000 00000000 00000000 00000038 804f2ec4 6465746e 342e3520
[13513.658582]         00000000 000002a0 00000000 00098fd6 00000000 836ddc10 00000000 836d4604
[13513.667074]         00000009 00003a98 80000000 00000004 00000003 802b9f0c 00000000 80700000
[13513.675565]         ...
[13513.678056] Call Trace:
[13513.678066] [<8005aeec>] 0x8005aeec
[13513.684098] [<805d0000>] 0x805d0000
[13513.687652] [<804f2ec4>] 0x804f2ec4
[13513.691210] [<836d4604>] 0x836d4604 [mt7603e@9b0f47cb+0x9580]
[13513.697051] [<802b9f0c>] 0x802b9f0c
[13513.700606] [<80009790>] 0x80009790
[13513.704161] [<80009798>] 0x80009798
[13513.707699] [<80023900>] 0x80023900
[13513.711275] [<836d4604>] 0x836d4604 [mt7603e@9b0f47cb+0x9580]
[13513.717125] [<800239a8>] 0x800239a8
[13513.720687] [<836d4604>] 0x836d4604 [mt7603e@9b0f47cb+0x9580]
[13513.726527] [<80358410>] 0x80358410
[13513.730093] [<836d479c>] 0x836d479c [mt7603e@9b0f47cb+0x9580]
[13513.735938] [<834f604c>] 0x834f604c [mt76@f9c4bba7+0xa9e0]
[13513.741526] [<834f3928>] 0x834f3928 [mt76@f9c4bba7+0xa9e0]
[13513.747108] [<834f4a64>] 0x834f4a64 [mt76@f9c4bba7+0xa9e0]
[13513.752696] [<836d1b64>] 0x836d1b64 [mt7603e@9b0f47cb+0x9580]
[13513.758547] [<834f4dd4>] 0x834f4dd4 [mt76@f9c4bba7+0xa9e0]
[13513.764155] [<834f1538>] 0x834f1538 [mt76@f9c4bba7+0xa9e0]
[13513.769733] [<8004c9a0>] 0x8004c9a0
[13513.773281] [<8004c1cc>] 0x8004c1cc
[13513.776848] [<80359b8c>] 0x80359b8c
[13513.780398] [<80359a74>] 0x80359a74
[13513.783946] [<804f42d8>] 0x804f42d8
[13513.787483] [<80359c28>] 0x80359c28
[13513.791030] [<80359c44>] 0x80359c44
[13513.794592] [<8003ffbc>] 0x8003ffbc
[13513.798141] [<8003fe84>] 0x8003fe84
[13513.801695] [<8003fe84>] 0x8003fe84
[13513.805252] [<800050b8>] 0x800050b8
[13513.808805] 
[13513.810324] ---[ end trace 23ddfe4dfddcdc5a ]---

```

```
cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.2'
DISTRIB_REVISION='r16495-bf0c965af0'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.2 r16495-bf0c965af0'
DISTRIB_TAINTS=''
```

kiryam, 2022-05-09:
I had same issue right now. [Powered by LuCI openwrt-21.02 branch (git-22.119.37126-a993714)](https://github.com/openwrt/luci) / OpenWrt 21.02.3 r16554-1d4dea6d4f

`[137858.777833] ------------[ cut here ]------------
[137858.782666] WARNING: CPU: 0 PID: 649 at target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2021-12-03-678071ef/mt7603/mac.c:208 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.797487] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun leds_gpio gpio_button_hotplug
[137858.849750] CPU: 0 PID: 649 Comm: napi/phy0-3 Not tainted 5.4.188 #0
[137858.856287] Stack : 00021c08 8005b064 805d0000 80567a08 00000000 00000000 00000000 00000000
[137858.864865]         00000000 00000000 00000000 00000000 00000000 00000001 8348dc30 3f57a075
[137858.873440]         8348dcc8 00000000 00000000 00000000 00000038 804f3584 65746e69 2e352064
[137858.882016]         00000000 00001aca 00000000 616e203a 00000000 8348dc10 00000000 836d4604
[137858.890592]         00000009 00003a98 80000000 00000009 00000000 802ba414 00000000 80700000
[137858.899168]         ...
[137858.901732] Call Trace:
[137858.901742] [<8005b064>] 0x8005b064
[137858.907948] [<805d0000>] 0x805d0000
[137858.911604] [<804f3584>] 0x804f3584
[137858.915259] [<836d4604>] 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.921185] [<802ba414>] 0x802ba414
[137858.924828] [<80009790>] 0x80009790
[137858.928467] [<80009798>] 0x80009798
[137858.932103] [<80023900>] 0x80023900
[137858.935747] [<836d4604>] 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.941674] [<800239a8>] 0x800239a8
[137858.945325] [<836d4604>] 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.951269] [<836d479c>] 0x836d479c [mt7603e@75b3653b+0x9580]
[137858.957201] [<834c604c>] 0x834c604c [mt76@71f30320+0xa9e0]
[137858.962869] [<834c3928>] 0x834c3928 [mt76@71f30320+0xa9e0]
[137858.968544] [<836d1b64>] 0x836d1b64 [mt7603e@75b3653b+0x9580]
[137858.974483] [<834c4dd4>] 0x834c4dd4 [mt76@71f30320+0xa9e0]
[137858.980182] [<834c1538>] 0x834c1538 [mt76@71f30320+0xa9e0]
[137858.985844] [<8004cac4>] 0x8004cac4
[137858.989476] [<8004c344>] 0x8004c344
[137858.993130] [<8035a02c>] 0x8035a02c
[137858.996768] [<80359f14>] 0x80359f14
[137859.000408] [<804f4998>] 0x804f4998
[137859.004050] [<8035a0c8>] 0x8035a0c8
[137859.007702] [<8035a0e4>] 0x8035a0e4
[137859.011361] [<80040144>] 0x80040144
[137859.014998] [<8004000c>] 0x8004000c
[137859.018630] [<8004000c>] 0x8004000c
[137859.022273] [<800050b8>] 0x800050b8
[137859.025914] 
[137859.027520] ---[ end trace 4884d8d414b47bd3 ]---`

alexz-kh, 2022-08-04:
same,
`

Model | Xiaomi Mi Router 3G
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621
Firmware Version | OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.213.35964-87836ca
Kernel Version | 5.4.188

`

lukasz1992, 2022-08-04:
mt7628 devices do not work correctly, because of broken switch driver - more described here: https://github.com/openwrt/mt76/issues/628