openwrt/mt76 #648: MT7612U injection performance

xqms, 2022-02-09:
Hi,

I'm trying to use packet injection on an MT7612U (Alfa AWUS036ACM). It works nicely, but is limited to around 1000 packets per second  with 500byte payloads, which translates to around 4 MBit/s). Is there any reason inside the driver / firmware that would make it that slow?

I'd be glad for any hints where to look :)

imsanju251200, 2022-02-11:
> Hi,
>
> I'm trying to use packet injection on an MT7612U (Alfa AWUS036ACM). It works nicely, but is limited to around 1000 packets per second with 500byte payloads, which translates to around 4 [...]

Hii, 
can you help me out how to connect that wifi adapter to latest kali linux with requried drivers for that

morrownr, 2022-04-20:
@imsanju251200 

> can you help me out how to connect that wifi adapter to latest kali linux with requried drivers for that

The driver is in the kernel. Plug the ALFA ACM in a port and it will work.