openwrt/mt76 #649: MCU message 1f (seq 13) timed out

wangyq, 2022-02-25:
My router (mt7621 with mt7603/mt7612)  suffers a random WIFI down and up, after a heavy traffic.
The logs is as follows:
```
[17188.279986] mt76x2e 0000:02:00.0: MCU message 1f (seq 13) timed out
[17188.341429] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[17188.346943] mt76x2e 0000:02:00.0: Build: 1
[17188.351365] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[17188.380024] mt76x2e 0000:02:00.0: Firmware running!
[17188.390352] ieee80211 phy1: Hardware restart was requested
[17269.283882] device wlan1 left promiscuous mode
[17269.288570] br-lan: port 6(wlan1) entered disabled state
[17271.310295] br-lan: port 6(wlan1) entered blocking state
[17271.315616] br-lan: port 6(wlan1) entered disabled state
[17271.321573] device wlan1 entered promiscuous mode
[17272.357804] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[17272.364501] br-lan: port 6(wlan1) entered blocking state
[17272.369885] br-lan: port 6(wlan1) entered forwarding state
[19006.009070] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[19006.014558] mt76x2e 0000:02:00.0: Build: 1
[19006.018948] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[19006.047937] mt76x2e 0000:02:00.0: Firmware running!
[19006.058226] ieee80211 phy1: Hardware restart was requested
```

========================
root@OpenWrt:~# uname -a
Linux OpenWrt 5.10.100 #0 SMP Mon Feb 21 15:51:47 2022 mips GNU/Linux
root@OpenWrt:~# opkg list | grep mt76
kmod-mt76-core - 5.10.100+2022-02-15-c67df0d3-4
kmod-mt7603 - 5.10.100+2022-02-15-c67df0d3-4
kmod-mt76x02-common - 5.10.100+2022-02-15-c67df0d3-4
kmod-mt76x2 - 5.10.100+2022-02-15-c67df0d3-4
kmod-mt76x2-common - 5.10.100+2022-02-15-c67df0d3-4
root@OpenWrt:~# 
root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10
root@OpenWrt:~#

DragonBluep, 2022-06-04:
What is the model of your router? I have a ZTE E8820 V2, this problem occurs after upgrading the firmware and restarting, but it can be solved by re-plugging the power supply. Someone say it seems to be a hardware issue. 
https://www.right.com.cn/forum/thread-5925493-1-1.html

ptpt52, 2023-10-10:
same issue reported on latest code.