openwrt/mt76 #650: MT7613 Channel 153(5765MHz) not work

intdev32, 2022-02-28:
I updated openwrt to latest snapshot build (r18953-b9251e3b40) and found 5GHz channel 153 is not working.

When I set channel to '153', it goes to 149.

But 149, 157, 161 working well.

Only 153 doesn't work.

OpenWrt SNAPSHOT r18953 > Drivers and firmware are:

kmod-mt76-connac	5.10.100+2022-02-15-c67df0d3-4	-	-	
kmod-mt76-core	5.10.100+2022-02-15-c67df0d3-4	-	-	
kmod-mt7603	5.10.100+2022-02-15-c67df0d3-4	-	-	
kmod-mt7615-common	5.10.100+2022-02-15-c67df0d3-4	-	-	
kmod-mt7615e	5.10.100+2022-02-15-c67df0d3-4	-	-	
kmod-mt7663-firmware-ap	5.10.100+2022-02-15-c67df0d3-4

intdev32, 2022-03-02:
I found this problem on Linksys EA4500 too. (OpenWrt 21.02.2 Release)
EA4500 uses Marvell 88W8366 wireless chip.

So this problem is not mt76 specific. Maybe not mt76 driver/firmware issue.

castillofrancodamian, 2024-03-06:
> ¿Se ve según las [especificaciones](https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/j/n/ac/ax)) ? El canal 153 es un canal exclusivo de 20 MHz. Cualquier cosa más ancha (HT40 o VHT80) comienza en el canal 149. Tenga en cuenta que 157 y 161 todavía están en ese mismo bloque de canales. Abra un escáner WiFi y no verá ninguna diferencia allí.
> 
> [image]

Your theory that channels 157 and 161 also force you to use channel 149 is not true. When I select channel 161, it broadcasts on 157. If I select channel 157, it broadcasts on 157. If I select channel 153, it broadcasts on 149. And if I select channel 149, it broadcasts on 149, logically. I use 80 MHz. MERCUSYS MR70X device.