openwrt/mt76 #653: MT7921k mesh mode

spiccinini, 2022-03-08:
Hi! Please can you clarify if the chips supports mesh mode and the driver will eventually support it?
Thanks!

jackloomen, 2022-03-29:
does this chip even have the ability to emit at 5ghz and 6ghz? Or is it locked down like intel (no-ir)?

SajjadPourali, 2022-04-15:
> does this chip even have the ability to emit at 5ghz and 6ghz? Or is it locked down like intel (no-ir)?
> 
> can someone post iw list if they have this chip?

As I checked, it can scan on 2.4ghz and 5ghz with `kmod-mt7921-common` package, but it has some issues to connect.

```Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * monitor
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: Station
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
				HE SU PPDU with 1x HE-LTF and 0.8us GI
				NDP with 4x HE-LTF and 3.2us GI
				STBC Tx <= 80MHz
				STBC Rx <= 80MHz
				Full Bandwidth UL MU-MIMO
				Partial Bandwidth UL MU-MIMO
				DCM Max Constellation: 2
				DCM Max Constellation Rx: 2
				SU Beamformee
				Beamformee STS <= 80Mhz: 3
				Ng = 16 SU Feedback
				Ng = 16 MU Feedback
				Codebook Size SU Feedback
				Codebook Size MU Feedback
				Triggered CQI Feedback
				Partial Bandwidth Extended Range
				PPE Threshold Present
				Power Boost Factor ar
				HE SU PPDU & HE PPDU 4x HE-LTF 0.8us GI
				20MHz in 40MHz HE PPDU 2.4GHz
				DCM Max BW: 1
[... 289 lines cut ...]
			* 7115 MHz [233] (disabled)
	valid interface combinations:
		 * #{ managed } <= 4,
		   total <= 4, #channels <= 1, STA/AP BI must match
	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: 65535
	max scan plan iterations: 0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support```

TaylorTWBrown, 2022-05-24:
I'm having trouble getting openwrt to start the device (MT7921K, PCIe).

Are you using a special kmod or firmware package to get the ap to work? I can detect it with Iw list, and see it in luci - but that's about it. 
[image]

dukzcry, 2022-06-16:
> does this chip even have the ability to emit at 5ghz and 6ghz? Or is it locked down like intel (no-ir)?

So its completely locked. Is it possible to bypass restrictions? Like its possible on ath9k and ath10k

TaylorTWBrown, 2022-06-16:
From what I can tell, it's locked. However, it's given me some trouble getting it working.

That said, I did get it working, although only 2.4 and 5Ghz. 6Ghz was not listed - I haven't investigate to see if it's something I'm missing.

nbd168, 2022-06-25:
MT7921 does not support mesh mode

closed 2022-06-25 by nbd168

trackoftime, 2024-04-29:
> MT7921 does not support mesh mode

Can you specify the reason? Is mesh related to hardware? Or is it that we don't support it now and will support it in the future? Thank you.

Dikkedibbes, 2024-11-22:
I was also having this issue and tried to do something with the driver until I stumbled upon this piece of code in the mt7921 driver. Seems that the hardware just doesnt support mesh/adhoc mode. 

```
/* The hardware does not support per-STA RX GTK, fallback
	 * to software mode for these.
	 */
	if ((vif->type == NL80211_IFTYPE_ADHOC ||
	     vif->type == NL80211_IFTYPE_MESH_POINT) &&
	    (key->cipher == WLAN_CIPHER_SUITE_TKIP ||
	     key->cipher == WLAN_CIPHER_SUITE_CCMP) &&
	    !(key->flags & IEEE80211_KEY_FLAG_PAIRWISE))
		return -EOPNOTSUPP;
```