openwrt/mt76 #654: MT7615E VHT160 doesn't work in mesh mode. Works in AP mode

amaumene, 2022-03-11:
Hi,

I have a mesh network between 2x Xiaomi Redmi Router AC2100.

What works:
* 802.11s VHT80 channel 36.
* VHT160 channel 36 AP mode only.
* 802.11s VHT160 channel 100 (but lot of packets loss, might be specific to my setup?)

What doesn't work:
* 802.11s VHT160 channel 36.

I'm running latest snapshot with:
```
kmod-mt76-connac - 5.10.103+2022-02-24-64c74dc9-4
kmod-mt76-core - 5.10.103+2022-02-24-64c74dc9-4
kmod-mt7603 - 5.10.103+2022-02-24-64c74dc9-4
kmod-mt7615-common - 5.10.103+2022-02-24-64c74dc9-4
kmod-mt7615-firmware - 5.10.103+2022-02-24-64c74dc9-4
kmod-mt7615e - 5.10.103+2022-02-24-64c74dc9-4
```

My config is as follow:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option band '2g'
	option htmode 'HT20'
	option channel 'auto'
	option country 'SG'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'ssid'
	option encryption 'sae-mixed'
	option key 'key!!'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option band '5g'
	option cell_density '0'
	option country 'SG'
	option htmode 'VHT80'
	option channel '36'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'sae-mixed'
	option ssid 'ssid'
	option key 'key!!'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'mesh'
	option encryption 'sae'
	option mesh_id 'mesh_ssid'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option key 'mesh_key!!'
	option network 'lan'
```

This is the log when I try to enable VHT160 (please ignore wlan0 this is my 2.4G network):
[... 268 lines cut ...]
Fri Mar 11 16:18:07 2022 daemon.notice hostapd: wlan1-1: interface state DFS->ENABLED
Fri Mar 11 16:18:07 2022 daemon.notice hostapd: wlan1-1: AP-ENABLED
Fri Mar 11 16:18:08 2022 daemon.notice wpa_supplicant[1450]: wlan1: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Mar 11 16:18:08 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5260 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5260 cf2=0
Fri Mar 11 16:18:08 2022 daemon.notice wpa_supplicant[1450]: wlan1: Unknown event 54
Fri Mar 11 16:18:08 2022 daemon.notice wpa_supplicant[1450]: wlan1: Unknown event 54
Fri Mar 11 16:18:08 2022 daemon.notice wpa_supplicant[1450]: wlan1: Unknown event 54
Fri Mar 11 16:18:08 2022 daemon.notice wpa_supplicant[1450]: wlan1: Unknown event 54
Fri Mar 11 16:18:08 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5280 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5280 cf2=0
Fri Mar 11 16:18:08 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5300 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5300 cf2=0
Fri Mar 11 16:18:08 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5320 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5320 cf2=0
Fri Mar 11 16:18:08 2022 daemon.notice netifd: Network device 'wlan1-1' link is down
Fri Mar 11 16:18:08 2022 kern.info kernel: [  554.685037] br-lan: port 5(wlan1-1) entered disabled state
Fri Mar 11 16:18:13 2022 daemon.err wpa_supplicant[1450]: could not get valid channel
Fri Mar 11 16:18:13 2022 daemon.notice wpa_supplicant[1450]: wlan1: interface state UNINITIALIZED->DFS
Fri Mar 11 16:18:14 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:14 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:36 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:36 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:36 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:36 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:51 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:51 2022 daemon.notice hostapd: handle_probe_req: send failed
Fri Mar 11 16:18:52 2022 daemon.notice hostapd: handle_probe_req: send failed
```

amaumene, 2022-03-11:
If I disable the mesh part of my config, I happily get this:
```
Interface wlan1
	ifindex 41
	wdev 0x100000016
	addr 28:d1:27:10:fb:3f
	type AP
	wiphy 1
	channel 36 (5180 MHz), width: 160 MHz, center1: 5250 MHz
	txpower 23.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
```

hugotai101, 2022-03-24:
I got problem too, 802.11s is running at 80MHz, while in access point is 160MHz in channel 100.

Catfriend1, 2023-03-21:
hostapd: Can't set DFS state for freq 5180 MHz