openwrt/mt76 #655: MT7615D STA+AP not working on 2.4Ghz

simontretter, 2022-03-15:
When adding STA to the 2.4Ghz radio on the GL-MT1300 (MT7615D) 2.4Ghz, the radio does not come up anymore.

> DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r19119-da2b97210c'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r19119-da2b97210c'

> Tue Mar 15 13:06:54 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0-1) --> new PHY
Tue Mar 15 13:06:54 2022 kern.info kernel: [  306.269406] br-lan: port 3(wlan0-1) entered blocking state
Tue Mar 15 13:06:54 2022 kern.info kernel: [  306.274981] br-lan: port 3(wlan0-1) entered disabled state
Tue Mar 15 13:06:54 2022 kern.info kernel: [  306.281491] device wlan0-1 entered promiscuous mode
Tue Mar 15 13:06:54 2022 daemon.notice hostapd: wlan0-1: interface state UNINITIALIZED->ENABLED
Tue Mar 15 13:06:54 2022 daemon.notice hostapd: wlan0-1: AP-ENABLED
Tue Mar 15 13:06:56 2022 daemon.notice netifd: radio0 (6384): sh: out of range
Tue Mar 15 13:06:56 2022 daemon.notice netifd: Wireless device 'radio0' is now up
Tue Mar 15 13:06:56 2022 daemon.notice netifd: Interface 'test' is enabled
Tue Mar 15 13:06:56 2022 daemon.notice hostapd: handle_probe_req: send failed
Tue Mar 15 13:07:02 2022 daemon.notice wpa_supplicant[3621]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Tue Mar 15 13:07:11 2022 daemon.notice hostapd: handle_probe_req: send failed
Tue Mar 15 13:07:11 2022 daemon.notice hostapd: handle_probe_req: send failed
Tue Mar 15 13:07:11 2022 daemon.notice hostapd: handle_probe_req: send failed

daneov, 2022-03-30:
Same here. Anything I can do to help? Some data I can provide maybe? I can reproduce it rather consistently so ask away!

phil2sat, 2022-04-20:
try option noscan '1' or in luci force 40mhz
also remove any beacon options or in luci delete field

and mybe post your /etc/config/wireless