openwrt/mt76 #656: T-MB5EU: Wifi completely broken

nwwt, 2022-03-16:
Since today's snapshot r19123/5.10.104 wifi on my [Tenbay T-MB5EU v01](https://openwrt.org/toh/hwdata/tenbay/tenbay_t-mb5eu_v01) is completely broken. (According to wiki: MediaTek MT7621AT / MT7905, MT7975 MT7915E)

Last week's snapshot (5.10.103) was fine, apart from horrible to non-existing reception on its 5GHz.

dmesg now just shows these:
`mt7615e 0000:02:00.0: Message 00004eed (seq 12) timeout`

luci fails as well. With the previous `/etc/config/wireless` removed, it'll load again, but once I try to reconfigure it's more of the same `timeout`s and scans just don't complete.

It now lists the same? MT7915E device both for radio0/radio1, both now supposedly dual-band, and with 80MHz/160MHz ac/ax available for 2.4GHz.

nwwt, 2022-03-18:
Bisected the breakage to 630384cb. Confirmed a 2nd time that this the breaking commit by using @ptpt52's mt76 branch (due to its important fixes) with it manually reverted.

Haven't been able to test the actual connection yet (I'm remote) but so far no error messages in dmesg and scanning & bringing up the APs works fine.

ptpt52, 2022-03-18:
@nwwt  thanks for your report. could you try this patch:
```
diff --git a/mt7915/eeprom.c b/mt7915/eeprom.c
index 5b133bcd..12fb5ef3 100644
--- a/mt7915/eeprom.c
+++ b/mt7915/eeprom.c
@@ -135,6 +135,9 @@ static void mt7915_eeprom_parse_band_config(struct mt7915_phy *phy)
 
        val = eeprom[MT_EE_WIFI_CONF + phy->band_idx];
        val = FIELD_GET(MT_EE_WIFI_CONF0_BAND_SEL, val);
+       if (val == MT_EE_BAND_SEL_DEFAULT &&
+           (!is_mt7915(&dev->mt76) || dev->dbdc_support))
+               val = phy->band_idx ? MT_EE_BAND_SEL_5GHZ : MT_EE_BAND_SEL_2GHZ;
 
        if (!is_mt7915(&dev->mt76)) {
                switch (val) {

```

ptpt52, 2022-03-19:
or try this:
```
diff --git a/mt7915/eeprom.c b/mt7915/eeprom.c
index 5b133bcd..cd242bfe 100644
--- a/mt7915/eeprom.c
+++ b/mt7915/eeprom.c
@@ -136,7 +136,7 @@ static void mt7915_eeprom_parse_band_config(struct mt7915_phy *phy)
        val = eeprom[MT_EE_WIFI_CONF + phy->band_idx];
        val = FIELD_GET(MT_EE_WIFI_CONF0_BAND_SEL, val);
 
-       if (!is_mt7915(&dev->mt76)) {
+       if (!is_mt7915(&dev->mt76) && !is_mt7916(&dev->mt76)) {
                switch (val) {
                case MT_EE_V2_BAND_SEL_5GHZ:
                        phy->mt76->cap.has_5ghz = true;
```

ptpt52, 2022-03-19:
ok, the 3rd try
```
diff --git a/mt7915/eeprom.c b/mt7915/eeprom.c
index 5b133bcd..b3b44d50 100644
--- a/mt7915/eeprom.c
+++ b/mt7915/eeprom.c
@@ -135,6 +135,9 @@ static void mt7915_eeprom_parse_band_config(struct mt7915_phy *phy)
 
        val = eeprom[MT_EE_WIFI_CONF + phy->band_idx];
        val = FIELD_GET(MT_EE_WIFI_CONF0_BAND_SEL, val);
+       if (val == MT_EE_BAND_SEL_DEFAULT &&
+           (is_mt7915(&dev->mt76) && dev->dbdc_support))
+               val = phy->band_idx ? MT_EE_BAND_SEL_5GHZ : MT_EE_BAND_SEL_2GHZ;
 
        if (!is_mt7915(&dev->mt76)) {
                switch (val) {
```

nbd168, 2022-03-19:
Please try the latest version from mt76 master

ptpt52, 2022-03-19:
build failed @nbd168 
```
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c: In function 'mt7915_mcu_beacon_mbss':
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1849:44: error: 'struct ieee80211_mutable_offsets' has no member named 'mbssid_off'
 1849 |                             &skb->data[offs->mbssid_off],
      |                                            ^~
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/linux/ieee80211.h:3896:47: note: in definition of macro 'for_each_element'
 3896 |         for (_elem = (const struct element *)(_data);                   \
      |                                               ^~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1848:9: note: in expansion of macro 'for_each_element_id'
 1848 |         for_each_element_id(elem, WLAN_EID_MULTIPLE_BSSID,
      |         ^~~~~~~~~~~~~~~~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1849:44: error: 'struct ieee80211_mutable_offsets' has no member named 'mbssid_off'
 1849 |                             &skb->data[offs->mbssid_off],
      |                                            ^~
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/linux/ieee80211.h:3897:27: note: in definition of macro 'for_each_element'
 3897 |              (const u8 *)(_data) + (_datalen) - (const u8 *)_elem >=    \
      |                           ^~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1848:9: note: in expansion of macro 'for_each_element_id'
 1848 |         for_each_element_id(elem, WLAN_EID_MULTIPLE_BSSID,
      |         ^~~~~~~~~~~~~~~~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1850:44: error: 'struct ieee80211_mutable_offsets' has no member named 'mbssid_off'
 1850 |                             skb->len - offs->mbssid_off) {
      |                                            ^~
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/linux/ieee80211.h:3897:37: note: in definition of macro 'for_each_element'
 3897 |              (const u8 *)(_data) + (_datalen) - (const u8 *)_elem >=    \
      |                                     ^~~~~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1848:9: note: in expansion of macro 'for_each_element_id'
 1848 |         for_each_element_id(elem, WLAN_EID_MULTIPLE_BSSID,
      |         ^~~~~~~~~~~~~~~~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1849:44: error: 'struct ieee80211_mutable_offsets' has no member named 'mbssid_off'
 1849 |                             &skb->data[offs->mbssid_off],
      |                                            ^~
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/linux/ieee80211.h:3899:27: note: in definition of macro 'for_each_element'
 3899 |              (const u8 *)(_data) + (_datalen) - (const u8 *)_elem >=    \
      |                           ^~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1848:9: note: in expansion of macro 'for_each_element_id'
 1848 |         for_each_element_id(elem, WLAN_EID_MULTIPLE_BSSID,
      |         ^~~~~~~~~~~~~~~~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1850:44: error: 'struct ieee80211_mutable_offsets' has no member named 'mbssid_off'
 1850 |                             skb->len - offs->mbssid_off) {
      |                                            ^~
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/linux/ieee80211.h:3899:37: note: in definition of macro 'for_each_element'
 3899 |              (const u8 *)(_data) + (_datalen) - (const u8 *)_elem >=    \
      |                                     ^~~~~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-20220320/mt7915/mcu.c:1848:9: note: in expansion of macro 'for_each_element_id'
 1848 |         for_each_element_id(elem, WLAN_EID_MULTIPLE_BSSID,
      |         ^~~~~~~~~~~~~~~~~~~

```

nbd168, 2022-03-19:
Please revert commit 65042bfd759ce0524969cf625eb7074107cd0949 for now

nwwt, 2022-03-19:
Tried with @ptpt52's branch again as it now includes 104dd5c, OpenWrt now @ 54aab4e (ie. incl. your MBSSID backport).

However, I did not include the revert of 65042bf (iow, checked out 4f4a64e) as it built fine for me. (Should I?)

Seems to work fine. That is: Scanning & enabling multi-bss (2x AX each for 2.4/5GHz @ 40/80MHz) succeeds without error. Still haven't checked the connection with an actual client, though.

nbd168, 2022-03-19:
No need to revert that commit when you're running latest OpenWrt. Please open up a new ticket if any connection issues are still present.

closed 2022-03-19 by nbd168

nwwt, 2022-03-23:
Finally checked today with clients: It's indeed working great now, exceeding previous performance by far. AX and 5GHz MBSSID work fine now, too. (Haven't yet checked addons like 11r that previously caused issues, though.)

[There really are a bunch of uncontroversial commits in ptpt52 that imho should land in master, like the 160Mhz removal which otherwise is wrongly offered for my MT7915 DBDC or the various firmware updates. There are even newer versions in linux-firmware though, like [for the MT7915E](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/mediatek?id=13dca280f76009ba2c5f25408543a1aaaa062c25). I'm using that atm.]