openwrt/mt76 #657: mt76x2, mt76x3, mt7615: tx power with mt76 driver is lower than with mediatek driver

csharper2005, 2022-03-27:
Hi everybody,

I have had many mt7621 router with different combinations of wireless chips:
- mt7603 + mt7615n
- mt7602 + mt7612
- mt7603 + mt7612
- mt7603 + mt7613
- mt7615dn (dbdc)

**Problem:**
- In all cases tx power with mt76 driver (OpenWrt) is 4-10 dBm lower than with mediatek driver (OEM firmwares, Padavan, AsusWrt). I roughly check that using [WiFi Analyzer ](https://play.google.com/store/apps/details?id=com.farproc.wifi.analyzer&hl=pl&gl=US)app on my phone

**Example**
2 routers mt7603 + mt7615. 
OpenWrt config - 20 dBm TX power
Stock config - according to documentation router has 100 mWt tx power (also 20 dBm).
Routers are in the same place. OpenWrt real TX power is lower:
[image]

**Possible root causes**
- TX power in mt76 is ok. The vendors (Xiaomi, Sercomm, Arcadyan etc) violates the maximum allowable power. 
- TX power with mt76 is not enough for different reasons (i.e. stock use file eeprom, not on Flash). We correctly use various hacks (changes country to Venezuela, change eeprom data, make own patches) to get the full power.

Do you have similar user experience? I found many similar tickets here and would like to take a closer look at this issue. Maybe someone has more accurate devices for estimating power than a phone. 

@nbd168 maybe it's possible to define +delta dBm via dts or some configs? It would be safer than another hacks.
@ptpt52 what do you think?

JFtico, 2022-04-01:
I have the same experience with a MT7621 SoC with a 7612 + 7603e Wifi chips (ZBT WE3526).

OEM firmware (OpenWRT derived) with the MediaTek drivers is 4 to 10dB stronger.

I also note the OEM 7603 related code loads the 'eeprom' from a file, not the actual flash. So maybe an override is needed in DTS?

This issue is also related to what is discussed in https://github.com/openwrt/mt76/issues/633

csharper2005, 2022-04-02:
> Or, use a unii-1 channel (like 44) on both routers, that way you can set the openwrt output to 23 legally. Unii-1 is 23dbi global and China.

Hmm... Channel 44 + driver default power in OpenWrt gives approximately the same EIRP as stock firmware.

dekomote, 2022-04-02:
> > Or, use a unii-1 channel (like 44) on both routers, that way you can set the openwrt output to 23 legally. Unii-1 is 23dbi global and China.
> 
> Hmm... Channel 44 + driver default power in OpenWrt gives approximately the same EIRP as stock firmware.

Just as I suspected. If you now set the country to China, set a Unii 2 channel and set the output to 26dbi, i think you'll see the same.

levshutov, 2022-04-03:
Done some measurements.

Two Xiaomi AX3200 routers.
1. OpenWrt SNAPSHOT r19309 with included mt76 driver.
2. OpenWrt SNAPSHOT r19309 with MTK driver (https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/336).

MTK wifi settings are default (maximum power).
mt76 power varys.

Signal measured through 3 brick walls. 
Channel width - 80 MHz.
Scan interval - 5 seconds.

<details>
  <summary>Channel 36, mt76 power 20 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 36, mt76 power 23 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 36, mt76 power 27 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 52, mt76 power 20 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 52, mt76 power 23 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 52, mt76 power 27 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 149, mt76 power 20 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 149, mt76 power 23 dBm</summary>
  [image]
</details>
<details>
  <summary>Channel 149, mt76 power 27 dBm</summary>
  [image]
</details>

mog422, 2022-04-12:
I am also experiencing this problem in mt7915. so I am using a workaround by modifying wireless-regdb.

MeIsReallyBa, 2022-04-15:
> Done some measurements.
> 
> Two Xiaomi AX3200 routers.
> 
> 1. OpenWrt SNAPSHOT r19309 with included mt76 driver.
> 2. OpenWrt SNAPSHOT r19309 with MTK driver (https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/336).
> 
> MTK wifi settings are default (maximum power). mt76 power varys.
> 
> Signal measured through 3 brick walls. Channel width - 80 MHz. Scan interval - 5 seconds.
> 
> Channel 36, mt76 power 20 dBm
> Channel 36, mt76 power 23 dBm
> Channel 36, mt76 power 27 dBm
> Channel 52, mt76 power 20 dBm
> Channel 52, mt76 power 23 dBm
> Channel 52, mt76 power 27 dBm
> Channel 149, mt76 power 20 dBm
> Channel 149, mt76 power 23 dBm
> Channel 149, mt76 power 27 dBm

can you try to turn off sky like this and make another test?

https://github.com/MeIsReallyBa/mt76/commit/e73be47c93b800e1c0a24ce5a044526754d31a4c

levshutov, 2022-04-15:
> can you try to turn off sku like this and make another test?
> 
> [MeIsReallyBa@e73be47](https://github.com/MeIsReallyBa/mt76/commit/e73be47c93b800e1c0a24ce5a044526754d31a4c)

No difference compared with previous results

phil2sat, 2022-04-21:
So then youre not capped by the regdb, its either the chipset does max 27dbm or some antenna gain like 3dbi
you could ty:
"iw phy "$phy" set antenna_gain 0" 
to get to 30dbi 
or 
it could be that some calibration data or even coext settings are different for your device, im no dev but i dived a little bit trough the eeprom.c and what i saw was that the coext settings wasnt implemented fully (as far as my knowledge can say)

in this case there only the devs can help

sinancetinkaya, 2022-04-25:
I have MT7602E and MT76 open source driver is giving me headaches. Low power output is one of them but I can live with it,
The main problem is that after some time (in 1 or 2 days), wireless becomes unstable, stuttering, slow.
Restarting wireless card solves the problem temporarily.
I can't see anything in the logs.

d-me3, 2022-05-17:
> > Done some measurements.
> > Two Xiaomi AX3200 routers.
> > 
> > 1. OpenWrt SNAPSHOT r19309 with included mt76 driver.
> > 2. OpenWrt SNAPSHOT r19309 with MTK driver (https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/336).
> > 
> > MTK wifi settings are default (maximum power). mt76 power varys.
> > Signal measured through 3 brick walls. Channel width - 80 MHz. Scan interval - 5 seconds.
> > Channel 36, mt76 power 20 dBm
> > Channel 36, mt76 power 23 dBm
> > Channel 36, mt76 power 27 dBm
> > Channel 52, mt76 power 20 dBm
> > Channel 52, mt76 power 23 dBm
> > Channel 52, mt76 power 27 dBm
> > Channel 149, mt76 power 20 dBm
> > Channel 149, mt76 power 23 dBm
> > Channel 149, mt76 power 27 dBm
> 
> can you try to turn off sku like this and make another test?
> 
> [MeIsReallyBa@e73be47](https://github.com/MeIsReallyBa/mt76/commit/e73be47c93b800e1c0a24ce5a044526754d31a4c)

Interesting results, thanks for sharing.

One question: did you use 802.11**ax** or 802.11**ac** in your tests?

I'm asking this because I also have two AX6S and one AX3200 (same device), and while testing 802.11**ax** performance through a brick wall I've noticed that the 802.11**ax** performance is **worse** than 802.11**ac** (it happens with multiple ax devices, but with an iPhone 13 the problem is even worse). See more details of my test [in the forum here](https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/14?u=dsouza)).

My hypothesis is that mt76 drivers may not be enabling beamforming in 802.11**ax** that could explain the instability of the signal strength in your tests *if* you used 802.11ax. If so, you may want to repeat your tests with 802.11ac to compare the results.

rodsmar, 2022-11-01:
Looking at MediaTek's GPL code, the EEPROM size on the MT7663 is 1536 (0x600). I made the change in the MT76 driver and noticed an average 5dbm increase in signal strength.

My config:

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel '149'
        option band '5g'
        option htmode 'VHT80'
        option disabled '0'
        option country 'BR'
        option noscan '1'

I've tested on my TP-Link Archer C6 v3. Not sure if work on other devices.

[patch.txt](https://github.com/openwrt/mt76/files/9912770/patch.txt)

jamesmcguirepro, 2022-11-06:
> My hypothesis is that mt76 drivers may not be enabling beamforming in 802.11**ax** that could explain the instability of the signal strength in your tests _if_ you used 802.11ax. If so, you may want to repeat your tests with 802.11ac to compare the results.

I have a MT7621-equipped router (DIR-2640) that does not support 802.11ax. It does not appear beamforming is enabled:
```
# grep vht_capab /var/run/hostapd*.conf
/var/run/hostapd-phy1.conf:vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-A-MPDU-LEN-EXP7]
```

Capturing beacon frames confirm it's not advertising support for beamforming.

Anteus, 2022-11-07:
> I have a MT7621-equipped router (DIR-2640) that does not support 802.11ax. It does not appear beamforming is enabled:
>
> ```
>
> # grep vht_capab /var/run/hostapd*.conf [...]

Seems to be enabled on MT7622, a RT3200

var/run/hostapd-wl1.conf:vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][VHT160-80PLUS80][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]

EugeneKey, 2022-11-07:
> Looking at MediaTek's GPL code, the EEPROM size on the MT7663 is 1536 (0x600). I made the change in the MT76 driver and noticed an average 5dbm increase in signal strength.
>
> My config:
>
> ` config` [...]

Sounds great, maybe you could create pull request with this diff?

ptpt52, 2022-11-07:
> Looking at MediaTek's GPL code, the EEPROM size on the MT7663 is 1536 (0x600). I made the change in the MT76 driver and noticed an average 5dbm increase in signal strength.
>
> My config:
>
> ` config` [...]

ping @nbd168 
what do you think about this

rodsmar, 2022-11-07:
> 

Yes. But I need testers to see if there are improvements in all equipment that uses the MT7663/MT7613 chip.

rodsmar, 2022-11-07:
> Sounds great, maybe you could create pull request with this diff?

Yes. But I need testers to see if there are improvements in all equipment that uses the MT7663/MT7613 chip.

EugeneKey, 2022-11-08:
> Yes. But I need testers to see if there are improvements in all equipment that uses the MT7663/MT7613 chip.

I use Xiaomi AX3200 (MT7622/MT7615) Could there be similar changes for this chip? How to find out the EEPROM size for MT7622?

thelinuxdude, 2022-11-10:
Is the EEPROM_SIZE for the mt7615 1024 or 1536 (same as mt7663)?

nbd168, 2022-11-11:
If you have a MT7613/MT7663 based device, please test latest OpenWrt master. I've increased the EEPROM size without affecting MT7615

csharper2005, 2022-11-11:
@nbd168 [MT7615 eeprom](https://mega.nz/file/d4IyADAb#63evCKQMoVzQA4vIDxnKJ7jr5x9kCzsGBSgyzgt1Fjk) can be even bigger (0x2600 bytes). [This](https://openwrt.org/inbox/toh/beeline/smartbox_turbo_plus?s[]=turbo&s[]=plus) device.

rodsmar, 2022-11-11:
> @nbd168 [MT7615 eeprom](https://mega.nz/file/d4IyADAb#63evCKQMoVzQA4vIDxnKJ7jr5x9kCzsGBSgyzgt1Fjk) can be even bigger (0x2600 bytes). [This](https://openwrt.org/inbox/toh/beeline/smartbox_turbo_plus?s%5B%5D=turbo&s%5B%5D=plus) device. Upd. And [this](https://openwrt.org/toh/beeline/smartbox_turbo) too.

Hey, you may look at GPL code for your device. It's out of my knowledge, but MT7615 EEPROM has iPAiLNA, iPAeLNA, ePAiLNA and ePAeLNA.

Maybe the stock rom assembled all EEPROM binaries in a single file.

rodsmar, 2022-11-15:
> If you have a MT7613/MT7663 based device, please test latest OpenWrt master. I've increased the EEPROM size without affecting MT7615

Apparently everything is fine!

sinancetinkaya, 2022-11-16:
Can #575 be related to this?

csharper2005, 2022-12-18:
I noticed that mt7663 tx power is really good. Very nice and very closer to stock. Thanks, @nbd168!

Also noticed that speed approximately the same with 40 and 80 MHz width.
40 MHz:
[image]

80 Mhz: 
[image]

So 40 MHz might be even better. Can anyone else confirm this observations for 7612, 7615?

csharper2005, 2022-12-18:
7603, 20 MHz:
[image]

7603, 40 MHz
[image]

sinancetinkaya, 2022-12-18:
> I noticed that mt7663 tx power is really good with the current SNAPSHOT. Very nice and very closer to stock. Thanks, @nbd168!
> 
> Also noticed that speed approximately the same with 40 and 80 MHz width. 40 MHz: [image]
> 
> 80 Mhz: [image]
> 
> So 40 MHz might be even better. Can anyone else confirm this observations for 7612, 7615?

I have mt7612e (movie card) does it count? Should I test it?

csharper2005, 2022-12-18:
> I have mt7612e (mpcie card) does it count? Should I test it?

I guess so.  Let's test them all.

sinancetinkaya, 2022-12-18:
@csharper2005  Sorry, I accidentally removed my post. Is it the latest snapshot?

csharper2005, 2022-12-19:
@sinancetinkaya, I've made a table:
Model | SNAPSHOT | Driver (mt76) | 7602 | 7603 | 7612 | 7613 | 7615 | 7615dbdc | 2g, 20 MHz | 2g, 40 MHz | 5g, 40 MHz | 5g, 80 MHz | Test conditions
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Beeline   SmartBox GIGA | r21523-4609228432 | 2022-12-09 |   | x |   | x |   |   | 88 / 53 | 60 / 87 | 144 / 140 | 110 / 168 | Walls, Intel(R)   Wireless-AC 9560
Etisalat S3 | r21549-963c77158c | 2022-12-09 |   | x |   |   | x |   | 63 / 7 | 49 / 8 | 182 / 228 | 303 / 326 | Walls, Intel(R)   Wireless-AC 9560

sinancetinkaya, 2022-12-19:
I have [UniElec u7621-06 16M](https://openwrt.org/toh/hwdata/unielec/unielec_u7621-06_256m_16m) with mini-PCIe cards: [MT7602E](https://tr.aliexpress.com/item/32457332512.html), [MT7612E](https://www.aliexpress.com/item/32816737559.html) 10 dbi antennas attached
These cards suppose to have excellent signal strength but my old cheap ISP provided modem's internal wifi has better signal

csharper2005, 2022-12-21:
Guys, I tested 7602, 7603, 7612, 7615 myself and didn't find any problems with TX power with current SNAPSHOT. So I decided to close to issue. 

The finding with strange speed on 7613 and 40MHz / 80 MHz channel width should be investigated separately.

closed 2022-12-21 by csharper2005