openwrt/mt76 #662: MT7612U USB AP Mode Support ?

yaotthaha, 2022-04-17:
I want to ask whether mt7612u (usb) supported AP mode. I try to use it, but it doesn't work.

morrownr, 2022-04-20:
It is supported.

Have you installed the driver package? kmod-mt76x2u

yaotthaha, 2022-04-20:
Yes, the driver is working properly. But when I connect, it disconnects within seconds. And the connection speed does not exceed 6Mbit/s

phil2sat, 2022-04-20:
Any logs? whats the reason for disconnects? which supplicant and encrryption do u use? does it work without encryption?
on console logread is very handy logread -f and then connect, you should see whats going on
or 
luci system/kernel log

A complete kernel log would also be nice.

also in /etc/config/wireless under the radio0 or 1 option log_level '0' gives much more debug 

on snapshots you could also play with "coverage cell density=1 to 3 (normal to very hight)" to maybe get over the 6/0mbit cant handshake problem.

Depending on Router type i can also help to rebuild image with -O3 since some cpu's are simply to slow for sae handshake.
Other thing is try to reduce distance or set a higher txpower on router or connecting device, had similar issues while router 1 does 10dbi and router 2 does 17dbi got MESH-SAE-AUTH-FAILURE in system log

closed 2022-04-21 by yaotthaha