openwrt/mt76 #664: mt76x0e 0000:01:00.0: MAC error detected / MESH-SAE-AUTH-FAILURE

phil2sat, 2022-04-19:
Model | TP-Link RE200 v2
Architecture | MediaTek MT7628AN ver:1 eco:2
Target Platform | ramips/mt76x8
Firmware Version | OpenWrt SNAPSHOT r19472-de7535a6ac / LuCI Master git-22.089.43958-7110635
Kernel Version | 5.10.111

Cant join my 5ghz mesh network
log ist spamming:
mt76x0e 0000:01:00.0: MAC error detected
MESH-SAE-AUTH-FAILURE
Also saw -5 error while testing

BTW 2.4ghz is working with the exact same config on both mesh devices:
`config wifi-iface 'wifinet0'                    
        option device 'radio0'          
        option network 'lan'            
        option wpa_group_rekey '86400'          
        option key 'rabime0301'                 
        option mode 'mesh'              
        option mesh_id 'phawk-mesh-node'
        option mesh_fwding '1'                  
        option mesh_rssi_threshold '0'  
        option disassoc_low_ack '0' 
        option skip_inactivity_poll '1'         
        option ifname 'wlan0-1'         
        option encryption 'sae' 

config wifi-iface 'wifinet1'                    
        option device 'radio1'          
        option network 'lan'            
        option wpa_group_rekey '86400'          
        option key 'rabime0301'                 
        option mode 'mesh'              
        option mesh_id 'phawk-mesh-node'
        option mesh_fwding '1'                  
        option mesh_rssi_threshold '0'  
        option disassoc_low_ack '0' 
        option skip_inactivity_poll '1'         
        option ifname 'wlan1-1'         
        option encryption 'sae' `

In Bootlog is no error reguarding firmware -2 or else seems fine loading 

Mac adresses are correct so far as i understand this, as on my device is a sticker with B0:4E:26:EB:01:E6 

so my macs are:
B0:4E:26:EB:01:E6  #lan port
B0:4E:26:EB:01:E7 #2.4ghz mesh
B2:4E:26:EB:01:E7 #2.4ghz AP
B0:4E:26:EB:01:E8 #5hgz mesh
B2:4E:26:EB:01:E8 #5ghz AP

tried all different combinations B0/B2 E6-E8 for the 5ghz device same error

Here is my actual Radio-Partition
[RE200.mtd6.bin.gz](https://github.com/openwrt/mt76/files/8510284/RE200.mtd6.bin.gz)

Boot log:
`Sun Apr 17 23:17:20 2022 kern.notice kernel: [    0.000000] Linux version 5.10.111 (phil2sat@lenovo) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19171-f811c33b19) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Sat Apr 16 12:02:13 2022
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] Board has DDR2
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] Analog PMU set to hw control
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] Digital PMU set to hw control
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7628AN ver:1 eco:2
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] MIPS: machine is TP-Link RE200 v2
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Apr 17 23:17:20 2022 kern.info kernel: [    0.000000] Zone ranges:
[... 294 lines cut ...]
Sun Apr 17 23:17:51 2022 daemon.notice wpa_supplicant[1240]: wlan1-1: new peer notification for 2c:3a:fd:84:b9:69
Sun Apr 17 23:17:51 2022 daemon.notice wpa_supplicant[1240]: wlan1-1: new peer notification for 2c:3a:fd:84:b9:69
Sun Apr 17 23:17:51 2022 daemon.notice wpa_supplicant[1240]: wlan1-1: mesh plink with 2c:3a:fd:84:b9:69 established
Sun Apr 17 23:17:51 2022 daemon.notice wpa_supplicant[1240]: wlan1-1: MESH-PEER-CONNECTED 2c:3a:fd:84:b9:69
Sun Apr 17 23:17:51 2022 kern.err kernel: [   49.946349] mt76x0e 0000:01:00.0: MAC error detected
Sun Apr 17 23:17:56 2022 kern.err kernel: [   54.736368] mt76x0e 0000:01:00.0: MAC error detected
Sun Apr 17 23:17:57 2022 kern.err kernel: [   55.876354] mt76x0e 0000:01:00.0: MAC error detected
Sun Apr 17 23:18:00 2022 daemon.notice wpa_supplicant[1240]: wlan0-1: MESH-SAE-AUTH-FAILURE addr=2c:3a:fd:84:b9:6a
Sun Apr 17 23:18:00 2022 kern.err kernel: [   59.256391] mt76x0e 0000:01:00.0: MAC error detected
Sun Apr 17 23:18:02 2022 kern.err kernel: [   61.266457] mt76x0e 0000:01:00.0: MAC error detected
Sun Apr 17 23:18:04 2022 kern.err kernel: [   63.066349] mt76x0e 0000:01:00.0: MAC error detected
Sun Apr 17 23:18:04 2022 user.info : luci: accepted login on /admin/system/opkg for root from 192.168.1.151
Sun Apr 17 23:18:05 2022 kern.err kernel: [   64.186718] mt76x0e 0000:01:00.0: MAC error detected
Sun Apr 17 23:18:07 2022 kern.err kernel: [   65.486393] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:10:57 2022 kern.err kernel: [   68.506379] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:10:59 2022 kern.err kernel: [   69.636705] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:01 2022 kern.err kernel: [   72.446409] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:03 2022 kern.err kernel: [   73.576348] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:03 2022 daemon.notice wpa_supplicant[1240]: wlan0-1: MESH-SAE-AUTH-FAILURE addr=2c:3a:fd:84:b9:6a
Tue Apr 19 08:11:06 2022 kern.err kernel: [   76.616778] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:09 2022 kern.err kernel: [   80.126338] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:11 2022 kern.err kernel: [   82.346717] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:13 2022 kern.err kernel: [   83.946401] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:14 2022 kern.err kernel: [   85.516726] mt76x0e 0000:01:00.0: MAC error detected
Tue Apr 19 08:11:16 2022 kern.err kernel: [   86.756499] mt76x0e 0000:01:00.0: MAC error detected`

phil2sat, 2022-04-20:
Can be closed, BUG in QCA4019-CT firmware

closed 2022-04-21 by phil2sat

DragonBluep, 2022-04-23:
The master branch I compiled today has the same problem, maybe we still need this issue. 5.2G band works fine, but 5.8G band shows error:
`mt76x0e 0000:01:00.0: MAC error detected`