openwrt/mt76 #666: mt76x0e 0000:01:00.0: MAC error detected

DragonBluep, 2022-04-25:
After setting the channel to 149-161, I get an error:

```
[ 1001.703487] device wlan0 left promiscuous mode
[ 1001.712709] br-lan: port 3(wlan0) entered disabled state
[ 1006.241216] br-lan: port 3(wlan0) entered blocking state
[ 1006.251895] br-lan: port 3(wlan0) entered disabled state
[ 1006.263030] device wlan0 entered promiscuous mode
[ 1006.272751] br-lan: port 3(wlan0) entered blocking state
[ 1006.283398] br-lan: port 3(wlan0) entered forwarding state
[ 1006.296176] br-lan: port 3(wlan0) entered disabled state
[ 1006.748576] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1006.761507] br-lan: port 3(wlan0) entered blocking state
[ 1006.772157] br-lan: port 3(wlan0) entered forwarding state
[ 1013.620856] mt76x0e 0000:01:00.0: MAC error detected
[ 1014.067483] mt76x0e 0000:01:00.0: MAC error detected
[ 1016.825002] mt76x0e 0000:01:00.0: MAC error detected
[ 1018.850382] mt76x0e 0000:01:00.0: MAC error detected
[ 1020.427553] mt76x0e 0000:01:00.0: MAC error detected
[ 1022.630532] mt76x0e 0000:01:00.0: MAC error detected
[ 1023.502530] mt76x0e 0000:01:00.0: MAC error detected
[ 1027.817828] mt76x0e 0000:01:00.0: MAC error detected
[ 1036.921109] mt76x0e 0000:01:00.0: MAC error detected
```
Associated with #664

phil2sat, 2022-04-27:
im getting this also, but on all 100+ channels

i guess it has something to do with missing eeprom data

mtkeepmgr shows nothing
rrom@phoenixhawk:~/eeprom/mtkeepmgr# ./mtkeepmgr -F RE200.mtd6.bin
[EEPROM identification]
  ChipID        : 7628h
  Version       : 2.0
  Chip          : MT7628

[Device identification]
  MacAddr       : 00:0c:43:e1:76:90

DragonBluep, 2022-04-27:
> i guess it has something to do with missing eeprom data

This is due to mtkeepmgr not supporting mt7628. Contrary to yours, my device factory firmware only supports 149-161 channels, but now only 36-64 channels work.

DragonBluep, 2022-04-28:
> i feared using other eeprom since i cant change the mac in file fo fit mine.

@phil2sat I just checked the source code, the MAC of the RE200 is not stored in the eeprom partition, so you don't have to worry about it. By the way, I guess your MT7628 is also not working properly. Sometimes the signal disappears suddenly. This is a problem that has existed for a long time.

phil2sat, 2022-04-28:
but if its not stored in eeprom, why mtkeepmgr shows mac from eeprom? also the mac error on non supported channels, seems like some parts in eeprom missing, as at least you can use 100+ channels, for me its only the 36-58 range.

and since my hack, the mesh network runs stable, without any reboot.

root@RE200:~# uptime
 11:09:44 up 8 days, 19:09,  load average: 0.05, 0.04, 0.00

DragonBluep, 2022-04-28:
Most devices store the mac in the eeprom, usually located at eeprom@0x04. The mac you see from mtkeepmgr is not the real value, for the RE200 it is in the "config" partition, 0x2008.
[	nvmem-cells = <&macaddr_config_2008>;](https://github.com/openwrt/openwrt/blob/573ce80ca612b0f642b7cbd5d9d33d89b7e96670/target/linux/ramips/dts/mt7628an_tplink_re200.dtsi#L142)

The "MAC" in the log "MAC error detected" is not mean "MAC address", they are two different things.

DragonBluep, 2022-05-06:
Before this commit 20d1fed838b958701ed5d030b663ba9227477ab1, there is a chance that channel 100+ will work properly. This commit seems to have added more stringent checks, which caused MT7610 to fail to start. I have looked around at the GPL source code and there seems to be no check for 0x10f4 register in MT7610. I am not sure if this register is only available in MT7612.
https://github.com/openwrt/mt76/blob/747c70fc6c890c1f0e4f33d4eff20bb6498dc662/mt76x02_mac.c#L1052

LearZhou, 2022-05-09:
[70516.605224] mt76x2e 0000:01:00.0: MAC error detected

Sun May  8 06:04:23 2022 kern.err kernel: [70516.605224] mt76x2e 0000:01:00.0: MAC error detected

Using OpenWRT snapshot build with the latest mt76 driver, D-Link DIR-860L B1.

chinghanyu, 2022-05-20:
@phil2sat  MT7628 is the 802.11bgn SoC (CPU + WLAN) so it does not support 5 GHz. MT7610E is the 5 GHz WLAN card. The onboard EEPROM has 64Kbytes, the first 32K (0-32767) are for MT7628, the second 32K are for MT7610E.

Use the following command to strip the mtd6.bin:
```
dd if=mtd6.bin of=mt7628eep.bin bs=512 count=1
dd if=mtd6.bin of=mt7610eep.bin bs=512 skip=1
```
Then use `mtkeemgr` to check `mt7610eep.bin` to print out the EEPROM content. The GPL tarball released by TP-Link seems to have an SKUdata.dat file so I highly suspect that all TP-Link's MediaTek-based products uses external files to determine TX power and do not rely on EEPROM content. So if I guess right, you can to either manually write target power values to the EEPROM, or use my module parameter hack approach in #633.

DragonBluep, 2022-06-20:
This problem seems to be caused by noise. The modified antenna produced a lot of noise. On the same model without any modification, except the CH165, all channels can be used.
[image]

nbd168, 2022-06-25:
Please try the latest version

DragonBluep, 2022-06-25:
"MAC error detected" problem was fixed. Although channel 149-161 still doesn't work, it might be a hardware bug, so maybe we can close this issue.

closed 2022-06-25 by nbd168

LearZhou, 2023-01-13:
There're still similar error logs on my D-Link DIR-860L B1 running openwrt snapshot.
```
[257938.974472] mt76x2e 0000:01:00.0: MAC error detected
[258658.955640] mt76x2e 0000:01:00.0: MAC error detected
```

vers-2071, 2023-01-14:
On my ASUS RT-N56U B1 with the release  OpenWrt 22.03.3 also has these errors

[42192.663667] mt76x2e 0000:01:00.0: MAC error detected
[42272.221710] mt76x2e 0000:01:00.0: MAC error detected
[42298.641011] mt76x2e 0000:01:00.0: MAC error detected
[43173.059364] mt76x2e 0000:01:00.0: MAC error detected
[43191.118853] mt76x2e 0000:01:00.0: MAC error detected
[43203.128558] mt76x2e 0000:01:00.0: MAC error detected
[43372.434374] mt76x2e 0000:01:00.0: MAC error detected
[43388.733976] mt76x2e 0000:01:00.0: MAC error detected

niansa, 2023-02-28:
Same issue on the `Asus RT-AC51U`

khanjui, 2023-03-18:
Sat Mar 18 13:09:23 2023 kern.err kernel: [825180.433285] mt76x2e 0000:01:00.0: MAC error detected
Sat Mar 18 13:11:09 2023 kern.err kernel: [825285.483278] mt76x2e 0000:01:00.0: MAC error detected
Sat Mar 18 13:11:09 2023 kern.err kernel: [825285.703219] mt76x2e 0000:01:00.0: MAC stop failed
Sat Mar 18 13:11:09 2023 kern.info kernel: [825285.754461] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Sat Mar 18 13:11:09 2023 kern.info kernel: [825285.760052] mt76x2e 0000:01:00.0: Build: 1
Sat Mar 18 13:11:09 2023 kern.info kernel: [825285.764280] mt76x2e 0000:01:00.0: Build Time: 201607111443____
Sat Mar 18 13:11:09 2023 kern.info kernel: [825285.793272] mt76x2e 0000:01:00.0: Firmware running!
Sat Mar 18 13:11:09 2023 kern.info kernel: [825285.803381] ieee80211 phy1: Hardware restart was requested
Sat Mar 18 13:16:48 2023 kern.err kernel: [825624.443228] mt76x2e 0000:01:00.0: MAC error detected
Sat Mar 18 13:17:43 2023 kern.err kernel: [825679.793243] mt76x2e 0000:01:00.0: MAC error detected
Sat Mar 18 13:23:21 2023 kern.err kernel: [826018.233192] mt76x2e 0000:01:00.0: MAC error detected

Today this happened while using a MacBook Air 2020 (Intel).

sinancetinkaya, 2023-07-12:
same with OpenWrt 22.03.3 r20028-43d71ad93e 
`Wed Jul 12 16:08:30 2023 kern.err kernel: [ 2156.969937] mt76x2e 0000:03:00.0: MAC error detected`

serj-47, 2023-10-25:
The problem is still relevant on mi router 4 giga. Today, I upgraded to 23.05 from 22.03.5 in the hope that the problem has already been solved. But it didn't help.

[ 2529.359747] mt76x2e 0000:01:00.0: MAC error detected
[ 2529.585130] mt76x2e 0000:01:00.0: MAC stop failed
[ 2529.750815] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 2529.756377] mt76x2e 0000:01:00.0: Build: 1
[ 2529.760582] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[ 2529.789662] mt76x2e 0000:01:00.0: Firmware running!
[ 2529.795889] ieee80211 phy1: Hardware restart was requested

serj-47, 2023-10-25:
The problem is still relevant on mi router 4 giga. Today, I upgraded to 23.05 from 22.03.5 in the hope that the problem has already been solved. But it didn't help.

```
[ 2529.359747] mt76x2e 0000:01:00.0: MAC error detected
[ 2529.585130] mt76x2e 0000:01:00.0: MAC stop failed
[ 2529.750815] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 2529.756377] mt76x2e 0000:01:00.0: Build: 1
[ 2529.760582] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[ 2529.789662] mt76x2e 0000:01:00.0: Firmware running!
[ 2529.795889] ieee80211 phy1: Hardware restart was requested
```

aaron13223, 2024-07-14:
Can anyone confirm if whoever is having this issue has RF shield or not on the wifi chips on their router? 

I tried every fix mentioned and nothing worked. I tried the snapshot release, 23.X, 22.X. I tried using a higher quality power adapter, made sure no usb peripheral were connected, installed heatsinks and even a fan to rule out overheating issues but only after I made the RF Shields using copper and soldered them on the problem went away. The router has been sooo stable ever since I did that mod. Here's how it turned out:

[image]

Ig if anyone is still facing this problem, it's worth a shot at least. (I recently came to know that Mi cheaped out on the later revisions of my model [Mi Router 4a] and removed the shielding)

Anyway, copper is so easy to solder on that it was was a breeze, the only time consuming part was bending the copper sheet into the correct size. I also installed thermal pads inside so that it also acts like a heatsink. I used a 0.3mm copper sheet btw.

10/10 would recommend :p