openwrt/mt76 #669: kernel oopses, mt76, OpenWrt 22.03.0-rc1

rewter2, 2022-05-09:
There are frequent kernel oopses with the latest 22.03.0-rc1 and other 22.03.xx versions on Xiaomi Redmi AC2100 (mt7621 et all):

Mon May  9 00:39:33 2022] ------------[ cut here ]------------
[Mon May  9 00:39:33 2022] WARNING: CPU: 3 PID: 1312 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-15-053668ac/mt7603/mac.c:208 0x82554814 [mt7603e@a7af8836+0x9660]
[Mon May  9 00:39:33 2022] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[Mon May  9 00:39:33 2022] CPU: 3 PID: 1312 Comm: hostapd Not tainted 5.10.111 #0
[Mon May  9 00:39:33 2022] Stack : 00000000 8071d058 809a0000 80000000 00000000 00000000 00000000 00000000
[Mon May  9 00:39:33 2022]         00000000 00000000 00000000 00000000 00000000 00000001 82bcf928 ce9a1b4b
[Mon May  9 00:39:33 2022]         82bcf9c0 00000000 00000000 82bcf7d0 00000038 80398944 ffffffea 00000000
[Mon May  9 00:39:33 2022]         82bcf7dc 0000018d 807c0838 ffffffff 82bcf908 806ec06c 80000000 806f0000
[Mon May  9 00:39:33 2022]         00000009 00000005 00003a98 80000000 00000018 8040235c 0000000c 8098000c
[Mon May  9 00:39:33 2022]         ...
[Mon May  9 00:39:33 2022] Call Trace:
[Mon May  9 00:39:33 2022] [<80398944>] 0x80398944
[Mon May  9 00:39:33 2022] [<8040235c>] 0x8040235c
[Mon May  9 00:39:33 2022] [<80007b08>] 0x80007b08
[Mon May  9 00:39:33 2022] [<80007b10>] 0x80007b10
[Mon May  9 00:39:33 2022] [<8037cd00>] 0x8037cd00
[Mon May  9 00:39:33 2022] [<82554814>] 0x82554814 [mt7603e@a7af8836+0x9660]
[Mon May  9 00:39:33 2022] [<8007fb54>] 0x8007fb54
[Mon May  9 00:39:33 2022] [<82554814>] 0x82554814 [mt7603e@a7af8836+0x9660]
[Mon May  9 00:39:33 2022] [<8002ff28>] 0x8002ff28
[Mon May  9 00:39:33 2022] [<82554814>] 0x82554814 [mt7603e@a7af8836+0x9660]
[Mon May  9 00:39:33 2022] [<8002fff0>] 0x8002fff0
[Mon May  9 00:39:33 2022] [<82554814>] 0x82554814 [mt7603e@a7af8836+0x9660]
[Mon May  9 00:39:33 2022] [<8274a1d4>] 0x8274a1d4 [mac80211@a3c90e61+0x81e80]
[Mon May  9 00:39:33 2022] [<82551a70>] 0x82551a70 [mt7603e@a7af8836+0x9660]
[Mon May  9 00:39:33 2022] [<82542f54>] 0x82542f54 [mt76@e524f0bc+0xafa0]
[Mon May  9 00:39:33 2022] [<8274aa58>] 0x8274aa58 [mac80211@a3c90e61+0x81e80]
[Mon May  9 00:39:33 2022] [<8018de98>] 0x8018de98
[Mon May  9 00:39:33 2022] [<825430c4>] 0x825430c4 [mt76@e524f0bc+0xafa0]
[Mon May  9 00:39:33 2022] [<82707ef8>] 0x82707ef8 [mac80211@a3c90e61+0x81e80]
[Mon May  9 00:39:33 2022] [<827098bc>] 0x827098bc [mac80211@a3c90e61+0x81e80]
[Mon May  9 00:39:33 2022] [<80099bf4>] 0x80099bf4
[Mon May  9 00:39:33 2022] [<82709c5c>] 0x82709c5c [mac80211@a3c90e61+0x81e80]
[Mon May  9 00:39:33 2022] [<82613c5c>] 0x82613c5c [cfg80211@39734fb7+0x45e20]
[Mon May  9 00:39:33 2022] [<80570fc4>] 0x80570fc4
[Mon May  9 00:39:33 2022] [<8056fa6c>] 0x8056fa6c
[Mon May  9 00:39:33 2022] [<804dac78>] 0x804dac78
[Mon May  9 00:39:33 2022] [<8056f0dc>] 0x8056f0dc
[Mon May  9 00:39:33 2022] [<82613b9c>] 0x82613b9c [cfg80211@39734fb7+0x45e20]
[Mon May  9 00:39:33 2022] [<80100010>] 0x80100010
[Mon May  9 00:39:33 2022] [<80570e70>] 0x80570e70
[Mon May  9 00:39:33 2022] [<8056fdb0>] 0x8056fdb0
[Mon May  9 00:39:33 2022] [<80570718>] 0x80570718
[Mon May  9 00:39:33 2022] [<8056f37c>] 0x8056f37c
[Mon May  9 00:39:33 2022] [<801a0dd8>] 0x801a0dd8
[Mon May  9 00:39:33 2022] [<80202d84>] 0x80202d84
[Mon May  9 00:39:33 2022] [<8056f814>] 0x8056f814
[Mon May  9 00:39:33 2022] [<80075768>] 0x80075768
[Mon May  9 00:39:33 2022] [<804dd800>] 0x804dd800
[Mon May  9 00:39:33 2022] [<8056f4b0>] 0x8056f4b0
[Mon May  9 00:39:33 2022] [<804d1040>] 0x804d1040
[Mon May  9 00:39:33 2022] [<8032297c>] 0x8032297c
[Mon May  9 00:39:33 2022] [<804d2ed0>] 0x804d2ed0
[Mon May  9 00:39:33 2022] [<8062bc28>] 0x8062bc28
[Mon May  9 00:39:33 2022] [<804d2384>] 0x804d2384
[Mon May  9 00:39:33 2022] [<804d30e0>] 0x804d30e0
[Mon May  9 00:39:33 2022] [<804cfd40>] 0x804cfd40
[Mon May  9 00:39:33 2022] [<80075a60>] 0x80075a60
[Mon May  9 00:39:33 2022] [<80014578>] 0x80014578

[Mon May  9 00:39:33 2022] ---[ end trace 98d878ad3cbe4026 ]---
[Mon May  9 10:31:03 2022] br-lan: port 2(lan2) received tcn bpdu
[Mon May  9 10:31:03 2022] br-lan: topology change detected, propagating

Any idea why might that be?

JFtico, 2022-05-09:
Also seeing the same with an MT7621 / 7603 platform, looks like the 2.4 (7603) driver is involved.

`Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 16:3f:3d:0e:49:53
Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 16:3f:3d:0e:49:53
Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:01:54 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 16:3f:3d:0e:49:53
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.214851] ------------[ cut here ]------------
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.219643] WARNING: CPU: 1 PID: 1538 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-15-053668ac/mt7603/mac.c:208 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.234312] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.235513]  act_gact ledtrig_usbport ifb sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.344478] CPU: 1 PID: 1538 Comm: hostapd Not tainted 5.10.111 #0
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.350727] Stack : 00000000 8071d058 809a0000 80000000 00000000 00000000 00000000 00000000
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.359167]         00000000 00000000 00000000 00000000 00000000 00000001 82689928 a578131e
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.367605]         826899c0 00000000 00000000 826897d0 00000038 80398944 ffffffea 00000000
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.376041]         826897dc 0000022c 807c0838 ffffffff 82689908 806ec06c 80000000 806f0000
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.384476]         00000009 00000002 00003a98 80000000 00000000 8040235c 00000004 80980004
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.392912]         ...
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.395443] Call Trace:
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.395521] [<80398944>] 0x80398944
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.401609] [<8040235c>] 0x8040235c
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.405174] [<80007b08>] 0x80007b08
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.408738] [<80007b10>] 0x80007b10
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.412297] [<8037cd00>] 0x8037cd00
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.415879] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.421684] [<8007fb54>] 0x8007fb54
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.425250] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.431067] [<8002ff28>] 0x8002ff28
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.434636] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.440450] [<8002fff0>] 0x8002fff0
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.444011] [<806d20c8>] 0x806d20c8
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.447594] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.453425] [<8284a1d4>] 0x8284a1d4 [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.459437] [<82721a70>] 0x82721a70 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.465283] [<82892f54>] 0x82892f54 [mt76@6b7daca7+0xafa0]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.470837] [<8284aa58>] 0x8284aa58 [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.476860] [<828930c4>] 0x828930c4 [mt76@6b7daca7+0xafa0]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.482419] [<82807ef8>] 0x82807ef8 [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.488436] [<828098bc>] 0x828098bc [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.494420] [<80099bf4>] 0x80099bf4
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.498000] [<82809c5c>] 0x82809c5c [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.503996] [<82793c5c>] 0x82793c5c [cfg80211@83ecf167+0x45e20]
[... cut ...]
Tue May  3 18:07:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:07:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 16:3f:3d:0e:49:53
Tue May  3 18:07:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:07:28 2022 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:07:28 2022 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
`

kiryam, 2022-05-09:
I had same issue right now. OpenWrt 21.02.3 r16554-1d4dea6d4f on Xiaomi Mi Router 4C

[137858.782666] WARNING: CPU: 0 PID: 649 at target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2021-12-03-678071ef/mt7603/mac.c:208 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.797487] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun leds_gpio gpio_button_hotplug
[137858.849750] CPU: 0 PID: 649 Comm: napi/phy0-3 Not tainted 5.4.188 #0
[137858.856287] Stack : 00021c08 8005b064 805d0000 80567a08 00000000 00000000 00000000 00000000
[137858.864865]         00000000 00000000 00000000 00000000 00000000 00000001 8348dc30 3f57a075
[137858.873440]         8348dcc8 00000000 00000000 00000000 00000038 804f3584 65746e69 2e352064
[137858.882016]         00000000 00001aca 00000000 616e203a 00000000 8348dc10 00000000 836d4604
[137858.890592]         00000009 00003a98 80000000 00000009 00000000 802ba414 00000000 80700000
[137858.899168]         ...
[137858.901732] Call Trace:
[137858.901742] [<8005b064>] 0x8005b064
[137858.907948] [<805d0000>] 0x805d0000
[137858.911604] [<804f3584>] 0x804f3584
[137858.915259] [<836d4604>] 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.921185] [<802ba414>] 0x802ba414
[137858.924828] [<80009790>] 0x80009790
[137858.928467] [<80009798>] 0x80009798
[137858.932103] [<80023900>] 0x80023900
[137858.935747] [<836d4604>] 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.941674] [<800239a8>] 0x800239a8
[137858.945325] [<836d4604>] 0x836d4604 [mt7603e@75b3653b+0x9580]
[137858.951269] [<836d479c>] 0x836d479c [mt7603e@75b3653b+0x9580]
[137858.957201] [<834c604c>] 0x834c604c [mt76@71f30320+0xa9e0]
[137858.962869] [<834c3928>] 0x834c3928 [mt76@71f30320+0xa9e0]
[137858.968544] [<836d1b64>] 0x836d1b64 [mt7603e@75b3653b+0x9580]
[137858.974483] [<834c4dd4>] 0x834c4dd4 [mt76@71f30320+0xa9e0]
[137858.980182] [<834c1538>] 0x834c1538 [mt76@71f30320+0xa9e0]
[137858.985844] [<8004cac4>] 0x8004cac4
[137858.989476] [<8004c344>] 0x8004c344
[137858.993130] [<8035a02c>] 0x8035a02c
[137858.996768] [<80359f14>] 0x80359f14
[137859.000408] [<804f4998>] 0x804f4998
[137859.004050] [<8035a0c8>] 0x8035a0c8
[137859.007702] [<8035a0e4>] 0x8035a0e4
[137859.011361] [<80040144>] 0x80040144
[137859.014998] [<8004000c>] 0x8004000c
[137859.018630] [<8004000c>] 0x8004000c
[137859.022273] [<800050b8>] 0x800050b8
[137859.025914] 
[137859.027520] ---[ end trace 4884d8d414b47bd3 ]---```

tcohar, 2022-07-14:
Xiaomi Mi Router 3G
OpenWrt 22.03.0-rc5 r19523-bfd070e7fa

[51713.845733] WARNING: CPU: 3 PID: 798 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-07-03-93e3fce9/mt7603/mac.c:208 0x82b64874 [mt7603e@3e6ee6ae+0x96e0]
[51713.860229] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat ledtrig_usbport sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[51713.930912] CPU: 3 PID: 798 Comm: napi/phy0-7 Not tainted 5.10.127 #0
[51713.937331] Stack : 00000000 8071f078 809a0000 80000000 00000000 00000000 00000000 00000000
[51713.945704]         00000000 00000000 00000000 00000000 00000000 00000001 829b3c08 2f8a065f
[51713.954076]         829b3ca0 00000000 00000000 829b3ab0 00000038 80399644 ffffffea 00000000
[51713.962449]         829b3abc 00000167 807c2860 ffffffff 829b3be8 806ee090 00000000 806f0000
[51713.970824]         00000009 00000007 00003a98 80000000 00000018 80402f84 0000000c 8098000c
[51713.979193]         ...
[51713.981640] Call Trace:
[51713.981693] [<80399644>] 0x80399644
[51713.987607] [<80402f84>] 0x80402f84
[51713.991094] [<80007b08>] 0x80007b08
[51713.994573] [<80007b10>] 0x80007b10
[51713.998049] [<8037e738>] 0x8037e738
[51714.001547] [<82b64874>] 0x82b64874 [mt7603e@3e6ee6ae+0x96e0]
[51714.007271] [<8007faec>] 0x8007faec
[51714.010760] [<82b64874>] 0x82b64874 [mt7603e@3e6ee6ae+0x96e0]
[51714.016484] [<8002ff28>] 0x8002ff28
[51714.019974] [<82b64874>] 0x82b64874 [mt7603e@3e6ee6ae+0x96e0]
[51714.025700] [<8002fff0>] 0x8002fff0
[51714.029194] [<82b64874>] 0x82b64874 [mt7603e@3e6ee6ae+0x96e0]
[51714.034934] [<82b50268>] 0x82b50268 [mt76@d862cbaa+0xaf00]
[51714.040434] [<82b64a08>] 0x82b64a08 [mt7603e@3e6ee6ae+0x96e0]
[51714.046171] [<82b563fc>] 0x82b563fc [mt76@d862cbaa+0xaf00]
[51714.051650] [<82b53ba8>] 0x82b53ba8 [mt76@d862cbaa+0xaf00]
[51714.057145] [<82b61b68>] 0x82b61b68 [mt7603e@3e6ee6ae+0x96e0]
[51714.062894] [<82b55074>] 0x82b55074 [mt76@d862cbaa+0xaf00]
[51714.068400] [<82b51aec>] 0x82b51aec [mt76@d862cbaa+0xaf00]
[51714.073898] [<80503a38>] 0x80503a38
[51714.077381] [<8003499c>] 0x8003499c
[51714.080865] [<80503bfc>] 0x80503bfc
[51714.084344] [<80503c18>] 0x80503c18
[51714.087826] [<800508e4>] 0x800508e4
[51714.091322] [<80503b58>] 0x80503b58
[51714.094800] [<80050b4c>] 0x80050b4c
[51714.098275] [<80050a10>] 0x80050a10
[51714.101760] [<80050a10>] 0x80050a10
[51714.105243] [<80050a10>] 0x80050a10
[51714.108718] [<80002f38>] 0x80002f38
[51714.112209] 
[51714.113939] ---[ end trace d4974020faa7d2b2 ]---

bittorf, 2022-08-15:
This it *not* an oops, but a WARNING.

araujorm, 2022-09-05:
Issue persists, wireless interface stops working and only a reboot restores functionality.
OpenWrt 22.03.0-rc6 r19590-042d558536 / LuCI openwrt-22.03 branch git-22.204.42822-9a18337
```
[1391454.092802] WARNING: CPU: 0 PID: 576 at target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2022-07-03-93e3fce9/mt7603/mac.c:208 0x81a54804 [mt7603e@c6c9b09c+0x9620]
[1391454.121912] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[1391454.247505] CPU: 0 PID: 576 Comm: napi/phy0-3 Not tainted 5.10.134 #0
[1391454.260609] Stack : 00000002 80060764 806c0000 805c2bac 00000000 00000000 00000000 00000000
[1391454.277534]         00000000 00000000 00000000 00000000 00000000 00000001 81a5dc40 65a9b9ec
[1391454.294460]         81a5dcd8 00000000 00000000 81a5dae8 00000038 802c68c4 00000000 ffffffea
[1391454.311385]         00000000 81a5daf4 000000e0 80646fc8 81a5dc20 8059b6e4 00000000 81a54804
[1391454.328306]         00000009 00000002 00003a98 80000000 00000001 80323ecc 00000000 80800000
[1391454.345226]         ...
[1391454.350406] Call Trace:
[1391454.350419] [<80060764>] 0x80060764
[1391454.362865] [<802c68c4>] 0x802c68c4
[1391454.370147] [<81a54804>] 0x81a54804 [mt7603e@c6c9b09c+0x9620]
[1391454.381881] [<80323ecc>] 0x80323ecc
[1391454.389149] [<80006d48>] 0x80006d48
[1391454.396414] [<80006d50>] 0x80006d50
[1391454.403654] [<800266ac>] 0x800266ac
[1391454.410923] [<81a54804>] 0x81a54804 [mt7603e@c6c9b09c+0x9620]
[1391454.422657] [<80026774>] 0x80026774
[1391454.429927] [<81a54804>] 0x81a54804 [mt7603e@c6c9b09c+0x9620]
[1391454.441676] [<81a40218>] 0x81a40218 [mt76@6de359a5+0xaca0]
[1391454.452916] [<81a54994>] 0x81a54994 [mt7603e@c6c9b09c+0x9620]
[1391454.464662] [<81a4624c>] 0x81a4624c [mt76@6de359a5+0xaca0]
[1391454.475899] [<81a42ed4>] 0x81a42ed4 [mt76@6de359a5+0xaca0]
[1391454.487121] [<8006dd04>] 0x8006dd04
[1391454.494388] [<81a51b18>] 0x81a51b18 [mt7603e@c6c9b09c+0x9620]
[1391454.506139] [<81a44ee4>] 0x81a44ee4 [mt76@6de359a5+0xaca0]
[1391454.517391] [<81a419f0>] 0x81a419f0 [mt76@6de359a5+0xaca0]
[1391454.528612] [<80051a84>] 0x80051a84
[1391454.535891] [<803d0b58>] 0x803d0b58
[1391454.543136] [<8058366c>] 0x8058366c
[1391454.550398] [<803d0d0c>] 0x803d0d0c
[1391454.557656] [<803d0d28>] 0x803d0d28
[1391454.564935] [<803d0c78>] 0x803d0c78
[1391454.572180] [<800440a4>] 0x800440a4
[1391454.579443] [<80043f68>] 0x80043f68
[1391454.586705] [<80043f68>] 0x80043f68
[1391454.593959] [<800025f8>] 0x800025f8
```

bosim, 2022-10-23:
Seen this on openwrt 22.03.2 / netgear r6220