openwrt/mt76 #672: [MT7615E/MT7622] Some devices lose connectivity but remain connected to the WiFi network

howl, 2022-05-17:
I commented about this in #518 but it seems to be kind different.

I have an ASUS RT-AC85P with MT7615E and over the 2.4 GHz band I have connectivity lost with some devices, Xiaomi Mi A2 Lite (stock), Xiaomi Redmi 7A (stock) and Motorola Moto E 2nd Gen (LineageOS 18). They remain connected to the WiFi network but they can't reach Internet.

Now I have tested a router with MT7622 for the 2.4 GHz radio and the result is the same for the Xiaomi Mi A2 Lite but the other two devices seems to work right, at least for almost a complete day. Will update if they fail.

This issue seems to be a kind of incompatibility in the way the router and the devices talk each other.

howl, 2022-05-17:
Forget to mention that the the three devices works without problems when I use another router as a bridge for 2.4 GHz (an ath9k one) even with a lot less power signal.

Just to mention, that ath9k have issues with Intel 8265 / 8275 where is hard to appear the WiFi in the network list but when it appears it works without problems, I mention this because I think is a kind of similar issue in the way some specifically devices talk each other.

howl, 2022-05-19:
Forget to mention that when this happens ping the device doesn't work, no ping from router to and no ping form another device to. Reconnecting the devices restores connectivity and makes ping and internet in the device work again.

howl, 2022-05-20:
Have tried changing cell density even Disabled, different Channel numbers and 20 Mhz instead 40 Mhz, none solves the issue.

viniribeirossa, 2022-05-21:
Hey. I also had this problem on my D-Link 878 A1 using the stable build, but I solved it using SNAPSHOTS
https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/

Download and install.
Once installed, use the commands to install LuCi
opkg update
opkg install luci

howl, 2022-05-21:
I have tried also SNAPSHOT but in my case the problem still there. With
MT7622 I have not tried SNAPSHOT but 22.03-SNAPSHOTS, where the same
version of mt76 is.

howl, 2022-05-22:
@viniribeirossa the last snapshot I used I think was with 5.10 and now it's with 5.15 so, I should not discard your suggestion so easily. I have installed mt7622 snapshot to test wed, so will check at the same time your suggestion.

howl, 2022-05-23:
@viniribeirossa it appears that for me is also working the last snapshot when I replied. But I did a bunch of changes, upgraded to snapshot, enable wed, install bridger and irqbalance.

You just installed snapshot or applied something more like me?

I will test just disabling wed and also bridger and irqbalance init scripts and check if it stays connected these days.

howl, 2022-05-23:
First day all just snapshot, no wed, no bridger, no irqbalance, yes thermal_zone as per default degrees and the device has not scream no connectivity.

viniribeirossa, 2022-05-24:
My D-Link DIR-878 A1 use kernel 5.10.115 runing 3 days, no problems!

I'm using snapshots--

<br class="Apple-interchange-newline">

viniribeirossa, 2022-05-27:
In the next stable version the problem will surely be fixed

howl, 2022-05-27:
@viniribeirossa I didn't see what exact commit solved thus, but, it's solved. I'm going to stick with snapshot despite 22.03 because in the ASUS there is an issue with nftables and now I have that router only to check if it gets solved eventually, and the other router to test WED I need to stay with 5.15 so.

Thanks for your advice to check snapshots.

closed 2022-05-27 by howl

viniribeirossa, 2022-05-29:
@howl How SNAPSHOSTS builds are built every day. Install this package. With it you will be able to DOWNLOAD AND UPDATE using SNAPSHOTS without losing your packages. You still have the option to download the image compiled with LuCI and its packages

opkg update
opkg install luci-app-attendedsysupgrade
--

<br class="Apple-interchange-newline">

luci-app-attendedsysupgrade
--

<br class="Apple-interchange-newline">

luci-app-attendedsysupgrade
--

<br class="Apple-interchange-newline">

schuettecarsten, 2022-09-06:
I have exactly this issue with 5.15 kernel and 22.03-snapshots on a Belkin RT-3200 with MT7622 and MT7915E wifi, on 5 GHz and 2,4 GHz. Reboot sometimes help, issue comes back after a few hours. Router itself is reachable from LAN, but clients connected to the Wifi are "dead".