openwrt/mt76 #673: MT7628AN htmode limited at HT40

chinghanyu, 2022-05-20:
The chip reports that is support both HT20 and HT40
```
root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# iwinfo wlan0 htmodelist
HT20 HT40
```
but only runs HT20; HT40 option is not effective.
```
root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# iw dev wlan0 info
Interface wlan0
        ifindex 14
        wdev 0x5
        addr 68:ff:7b:4d:f9:47
        ssid OpenWrt-2G
        type AP
        wiphy 0
        channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
        txpower 22.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       151     0       0       0       0       40848           162
```

chinghanyu, 2022-05-20:
Tried to override the source code to force using HT40 but still got HT20, implying the bandwidth setting logic is broken.

DragonBluep, 2022-05-20:
Have you tried checking "Force 40Mhz" in LuCI or add `option noscan   '1'` to /etc/config/wireless?

chinghanyu, 2022-05-20:
See detail instructions for .11n settings at [https://openwrt.org/docs/guide-user/network/wifi/basic#htmodewi-fi_channel_width](https://openwrt.org/docs/guide-user/network/wifi/basic#htmodewi-fi_channel_width)

chinghanyu, 2022-05-21:
I think the secondary channel offset routine is completely missing in OpenWRT driver, which can be found in the GPL driver at [https://github.com/openwrt/mtk-wifi-gpl/blob/5e8dae211cb1b9f30dd3e7281cb8fab434a5483f/mgmt/mgmt_ht.c#L294](https://github.com/openwrt/mtk-wifi-gpl/blob/5e8dae211cb1b9f30dd3e7281cb8fab434a5483f/mgmt/mgmt_ht.c#L294)

chinghanyu, 2022-05-26:
After a deep look. OpenWRT's mt76 driver dose nothing wrong but it is difficult for some device to run at HT40 in 2.4 GHz band. Sorry for the false alarm.

closed 2022-05-26 by chinghanyu