openwrt/mt76 #678: Switch settings wrong on WR902AC

micw, 2022-06-13:
Hello,

I'm analyzing an issue with a "freifunk" firmware based on OpenWRT 21.02. The problem is that a switch setting is not applied correctly on my WR902AC.

I can reproduce it with the following commands:

```
swconfig dev switch0 vlan 1062 set ports "4t 6t"
swconfig dev switch0 vlan 1 set ports "4 6t"
swconfig dev set apply
swconfig dev switch0 show
```

The result I expect is that VLAN 1062 is tagged on both ports and VLAN 1 is tagged in port 6 only and untagged on port 4. But the result is:

```
VLAN 1:
	ports: 4 6t 
VLAN 1062:
	ports: 4 6t 
```

When I swap the `swconfig dev switch0 vlan` commands, the result is

```
VLAN 1:
	ports: 4t 6t 
VLAN 1062:
	ports: 4t 6t 
```

So it seems that the port config of the last command is copied to all VLAN ids.

micw, 2022-06-13:
Just do be sure that it is not causes by the high VLAN id, I tried it with VLAN 9 instead of 1062 - results are the same.

nbd168, 2022-06-25:
This project is only for the mt76 wireless driver, not for anything related to switch issues

closed 2022-06-25 by nbd168