openwrt/mt76 #680: MT7615 disconnection inactivity but it does not reset

viniribeirossa, 2022-06-26:
i am using the latest mt76 update
22.03-SNAPSHOT

I don't touch the phone for a while, it goes into idle mode, but when I touch the device again, the connection doesn't reestablish itself again

```
daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED e0:b6:55:b4:53:a2
Sun Jun 26 00:49:35 2022 daemon.info hostapd: wlan1: STA e0:b6:55:b4:53:a2 IEEE 802.11: disassociated due to inactivity
Sun Jun 26 00:49:36 2022 daemon.info hostapd: wlan1: STA e0:b6:55:b4:53:a2 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jun 26 06:00:25 2022 daemon.info hostapd: wlan1: STA e0:b6:55:b4:53:a2 IEEE 802.11: authenticated
```

mrkiko, 2022-07-13:
Is this an iDevice?
If yes, is autojoin turned on? Probably yes, but just to make sure.
What are the steps you need to perform to re-establish the connection?

viniribeirossa, 2022-07-14:
Yes.
I turned off the device's wifi and turned it back on.

kocoman1, 2022-11-15:
its some bug disassoc_low_ack and changr country to 00/world but still not work https://www.reddit.com/r/openwrt/comments/lgsqsn/newifid2_deauthenticated_due_to_inactivity/?utm_source=share&utm_medium=ios_app&utm_name=iossmf

thelinuxdude, 2023-09-22:
@kocoman1 can you post your wireless config.  WE have seen similar issues with the mt7615d with DBDC mode where radio0 is a 2.4 AP and the radio1 is a 5 GHz station connection.  In our case the station is connected to a 5GHz DFS channel.

thelinuxdude, 2023-09-23:
In our situation we have a 5GHz STA connection to an AP that switches to DFS, when this happens the AP on the other radio in this case is a 2.4 AP will drop all clients and will not work until restart the radio using `wifi up radio0` which is the radio being used for the 2.4 AP.  We see similar issues after a site survey where the AP starts to act up and drop  packets.  Heard there are some TX race condition and stability that needs clean up and was not sure if this is realated to that issue.

wifi: mt76: fix race condition related to checking tx queue fill status