openwrt/mt76 #681: mt7921e driver crash in kernel 5.18

kaienkira, 2022-06-28:
machine:
Lenovo Thinkbook 14p Gen2 with 5800H

network card:
01:00.0 Network controller: MEDIATEK Corp. MT7921 802.11ax PCI Express Wireless Network Adapter

Jun 23 17:36:39 ChkTB14PDev kernel: mt7921e 0000:01:00.0: driver own failed
Jun 23 17:36:40 ChkTB14PDev kernel: mt7921e 0000:01:00.0: driver own failed
Jun 23 17:36:40 ChkTB14PDev kernel: mt7921e 0000:01:00.0: chip reset
Jun 23 17:36:41 ChkTB14PDev kernel: mt7921e 0000:01:00.0: driver own failed
Jun 23 17:36:43 ChkTB14PDev kernel: mt7921e 0000:01:00.0: Timeout for driver own
Jun 23 17:36:44 ChkTB14PDev kernel: mt7921e 0000:01:00.0: driver own failed
Jun 23 17:36:44 ChkTB14PDev kernel: ------------[ cut here ]------------
Jun 23 17:36:44 ChkTB14PDev kernel: WARNING: CPU: 11 PID: 3988 at kernel/kthread.c:659 kthread_park+0x7f/0x90
Jun 23 17:36:44 ChkTB14PDev kernel: Modules linked in: cmac ccm vfat fat snd_sof_amd_renoir snd_sof_amd_acp snd_sof_pci snd_sof intel_rapl_msr snd_sof_utils intel_rapl_common mt7921e snd_hda_codec_realtek snd_soc_core snd_hda_c>
Jun 23 17:36:44 ChkTB14PDev kernel:  tpm_tis_core sparse_keymap snd drm_dp_helper platform_profile ccp snd_pci_acp3x soundcore k10temp industrialio_triggered_buffer i2c_piix4 mac_hid roles wmi tpm rfkill video kfifo_buf i2c_hid>
Jun 23 17:36:44 ChkTB14PDev kernel: CPU: 11 PID: 3988 Comm: kworker/u32:1 Tainted: G           OE     5.18.5-zen1-1-zen #1 1b0f14670b06387fbfac0d7e749656f6285dc8ca
Jun 23 17:36:44 ChkTB14PDev kernel: Hardware name: LENOVO 20YN/LNVNB161216, BIOS GWCN33WW 06/28/2021
Jun 23 17:36:44 ChkTB14PDev kernel: Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
Jun 23 17:36:44 ChkTB14PDev kernel: RIP: 0010:kthread_park+0x7f/0x90
Jun 23 17:36:44 ChkTB14PDev kernel: Code: 89 df e8 94 c8 01 00 48 85 c0 74 27 31 c0 5b 5d c3 cc 0f 0b 48 8b ab d0 06 00 00 a8 04 74 af 0f 0b b8 da ff ff ff 5b 5d c3 cc <0f> 0b b8 f0 ff ff ff eb db 0f 0b eb d5 0f 1f 40 00 f3 0f >
Jun 23 17:36:44 ChkTB14PDev kernel: RSP: 0018:ffff988e8242bdf0 EFLAGS: 00010202
Jun 23 17:36:44 ChkTB14PDev kernel: RAX: 0000000000000004 RBX: ffff89518e0d4100 RCX: 0000000000000000
Jun 23 17:36:44 ChkTB14PDev kernel: RDX: 0000000000000001 RSI: 0000000000004187 RDI: ffff89518e0d4100
Jun 23 17:36:44 ChkTB14PDev kernel: RBP: ffff895186c59980 R08: ffff89518e0024a0 R09: ffff988e8242bda8
Jun 23 17:36:44 ChkTB14PDev kernel: R10: ffffffffb9e5aa20 R11: 0000000000000002 R12: ffff89518e0008e0
Jun 23 17:36:44 ChkTB14PDev kernel: R13: ffff89518e0020e0 R14: ffff89518e008610 R15: ffff89518e002430
Jun 23 17:36:44 ChkTB14PDev kernel: FS:  0000000000000000(0000) GS:ffff8954200c0000(0000) knlGS:0000000000000000
Jun 23 17:36:44 ChkTB14PDev kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 23 17:36:44 ChkTB14PDev kernel: CR2: 00007fe694dfd778 CR3: 0000000369c10000 CR4: 0000000000750ee0
Jun 23 17:36:44 ChkTB14PDev kernel: PKRU: 55555554
Jun 23 17:36:44 ChkTB14PDev kernel: Call Trace:
Jun 23 17:36:44 ChkTB14PDev kernel:  <TASK>
Jun 23 17:36:44 ChkTB14PDev kernel:  mt7921e_mac_reset+0xa2/0x3d0 [mt7921e 7da390c800be1324199a013f8c9cad8a2e8a352b]
Jun 23 17:36:44 ChkTB14PDev kernel:  mt7921_mac_reset_work+0xa3/0x14e [mt7921_common a50a633efd758d3da02ef2cbeb0c45b919cd8b3b]
Jun 23 17:36:44 ChkTB14PDev kernel:  process_one_work+0x255/0x410
Jun 23 17:36:44 ChkTB14PDev kernel:  worker_thread+0x55/0x4d0
Jun 23 17:36:44 ChkTB14PDev kernel:  ? process_one_work+0x410/0x410
Jun 23 17:36:44 ChkTB14PDev kernel:  kthread+0x13f/0x160
Jun 23 17:36:44 ChkTB14PDev kernel:  ? kthread_complete_and_exit+0x20/0x20
Jun 23 17:36:44 ChkTB14PDev kernel:  ret_from_fork+0x22/0x30
Jun 23 17:36:44 ChkTB14PDev kernel:  </TASK>
Jun 23 17:36:44 ChkTB14PDev kernel: ---[ end trace 0000000000000000 ]---

kaienkira, 2022-06-28:
works in kernel 5.15.50, only appears in 5.18 kernel

kaienkira, 2022-06-28:
before driver crash, some error log here

Jun 23 17:17:46 ChkTB14PDev kernel: mt7921e 0000:01:00.0: driver own failed
Jun 23 17:17:46 ChkTB14PDev kernel: mt7921e 0000:01:00.0: chip reset
Jun 23 17:17:46 ChkTB14PDev kernel: mt7921e 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0009 address=0xff1fa000 flags=0x0020]
Jun 23 17:17:46 ChkTB14PDev kernel: mt7921e 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0009 address=0xff876800 flags=0x0020]
Jun 23 17:17:49 ChkTB14PDev kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 13) timeout
Jun 23 17:17:49 ChkTB14PDev kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Jun 23 17:17:49 ChkTB14PDev kernel: mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220311230842a
Jun 23 17:17:49 ChkTB14PDev kernel: mt7921e 0000:01:00.0: WM Firmware Version: ____010000, Build Time: 20220311230931

LoreleiPenn, 2022-07-06:
Same issue as well with zen kernel 5.18.

Vokinloksar, 2022-07-07:
same issue with kernel 5.18.9, is this the right place to ask?

kunaltyagi, 2022-07-11:
Issue not present in 5.17.5 either, and not fixed in 5.18.10

deren, 2022-07-14:
Hi guys,

I cannot reproduce this issue in my environment. Can someone help to provide following information?
1. How to reproduce? The scenario to show this issue.
2. Full dmesg log
3. Check if the issue happened in mainline or v5.19? (if possible)

Thanks

LoreleiPenn, 2022-07-14:
Also it seems not fixed in 5.18.11.

deren, 2022-07-18:
Hi guys,

I cannot see this issue on Lenovo Thinkbook 14p with Zen kernel v5.18.5. Can someone help to provide following information?

1. How do you reproduce this problem?
2. Full demsg log in your test
3. More detail about your environment settings, such as OS type, kernel branch .... etc.

Thanks

Vokinloksar, 2022-07-18:
> Hi guys,
>
> I cannot see this issue on Lenovo Thinkbook 14p with Zen kernel v5.18.5. Can someone help to provide following information?
>
> 1. How do you reproduce this problem?
> 2. Full demsg log in your [...]

Thanks Deren, this issue does just happen ramdomly so it's kinda hard to reproduce(from my exp). 

This is the frequency that this issue happend to my laptop ( thinkpad t14 gen2 AMD with arch linux 5.18.9), it seems this issue only happens on workdays when I'm using office wifi. I don't recall having the same issue using my home wifi. 

`journalctl` | grep 'chip reset'` shows:

```
Jun 23 16:08:47 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 24 12:42:29 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 24 17:32:06 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 25 16:32:10 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 27 14:07:05 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 27 20:53:24 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 27 20:57:06 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 28 10:27:33 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 28 15:00:39 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 30 12:26:01 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 30 15:18:31 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jun 30 15:29:21 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 01 12:03:53 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 01 18:04:42 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 01 20:04:06 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 04 11:12:56 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 04 12:04:06 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 05 13:26:07 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 07 17:16:20 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 08 12:51:43 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 13 15:59:17 xxx kernel: mt7921e 0000:03:00.0: chip reset
Jul 13 20:47:21 xxx kernel: mt7921e 0000:03:00.0: chip reset
```

deren, 2022-07-18:
Hi @Vokinloksar 

Thanks for the update. Can you please give me the dmesg log? (if the file is too big, you may strip one day log only)

If possible, can you point out when the problem show up? Maybe the notebook is moving or try to surf specific network service...etc.

Thanks

Vokinloksar, 2022-07-18:
The dmseg log is here:  https://termbin.com/06pz2. Hope that can be of help.

I don't yet find any specific pattern for such behavior, but I do find that this issue only happens when I use office wifi, which is like a  wifi mesh with a lot of wifi devices everywhere so that you can get wifi whenever you are in the office. In my home with one wifi router, I didn't recall having that issue.

Another possible occasion I recall is perhaps when the devices were idled for some time and turned into hibernate mode. I remember several times when I woke up from a launch nap and try wake up my laptop, the wifi would be gone by then. 

But there is also one occasion that I remember clearly is the first time this happened, when I'm trying to solve a problem for my colleague and searching online, and the wifi was just suddenly gone.

So no apparent pattern here that I can think of other than the router difference.

nairaner, 2022-07-18:
For me it mostly happens when downloading faster than 20-30MB/s (peak speed was around 60MB/s). When I was downloading some larger games it crashed every 5-10GBs. 
Pretty sure these are events from that day:
```
Jul 13 05:35:33.588476 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 10:10:05.934273 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 10:10:15.100796 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 10:11:00.673279 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 10:11:28.789268 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 10:16:53.129506 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 10:19:02.141521 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 10:44:55.579534 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 14:24:45.096285 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 17:11:05.244567 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 17:11:52.516412 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 17:18:26.498277 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 17:20:32.469586 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 17:21:14.731282 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 17:55:34.605272 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 17:58:28.330343 Kuro kernel: mt7921e 0000:06:00.0: chip reset
Jul 13 18:27:59.688237 Kuro kernel: mt7921e 0000:06:00.0: chip reset
```

Right now I am unable find any 5Ghz WiFi (pretty sure its related to linux-firmware update, but downgrading didn't help), so it doesn't happen for me right now, because most I get is ~10MB/s

Log of some previous boot with a few crashes: https://termbin.com/griv

Network card:
14c3:0608
06:00.0 Network controller: MEDIATEK Corp. MT7921K (RZ608) Wi-Fi 6E 80MHz

kaienkira, 2022-07-20:
I'm using archlinux, after switch to default linux kernel instead of linux-zen kernel, this issue did not happened again

LoreleiPenn, 2022-07-22:
> I'm using archlinux, after switch to default `linux` kernel instead of `linux-zen` kernel, this issue did not appear again.

Sadly for me last night I had the issue once again with the linux kernel of Arch. I will stick with linux-lts for the moment. Thank you.

objelf, 2022-07-23:
> 

Based on the NG log from [https://termbin.com/06pz2](https://termbin.com/06pz2) and [https://bugzilla-attachments.redhat.com/attachment.cgi?id=1897636], I found the issue both come from the same firmware version:  _mt7921e 0000:03:00.0: WM Firmware Version: ____010000, Build Time: 20220311230931_ And, I have tried to compare the difference between  v5.17.15 (the good one) and v5.18.4 (the bad one), and found the major changes for MT7921 would be "mt76: mt7921: fix up the monitor mode" and a bunch of refactoring code to introduce mt7921u. so I guessed that might be those new changes would be relying on newer firmware.

Could you help update the latest firmware to see whether the issue is still there?
The latest version of Wifi firmware is 20220608210922. 

To install or update the firmware: (Steps copied from [1])
Go to the following site:
[https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek)

Click on BT_RAM_CODE_MT7961_1_2_hdr.bin
Click on plain
Save file

Click on WIFI_MT7961_patch_mcu_1_2_hdr.bin
Click on plain
Save file

Click on WIFI_RAM_CODE_MT7961_1.bin
Click on plain
Save file

Create the needed directory (if necessary):
$ sudo mkdir /lib/firmware/mediatek
Copy the files to the following locations:

$ sudo cp BT_RAM_CODE_MT7961_1_2_hdr.bin /lib/firmware/mediatek
$ sudo cp WIFI_MT7961_patch_mcu_1_2_hdr.bin /lib/firmware/mediatek
$ sudo cp WIFI_RAM_CODE_MT7961_1.bin /lib/firmware/mediatek

Reboot:
$ sudo reboot

[1] [https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md](https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md)

LoreleiPenn, 2022-07-23:
Hello @objelf.
Within my system I have those firmware files you mention about. I verified it with a sha512sum. First I downloaded the files:
```bash
ls -l *bin               
-rw-r--r-- 1 lorelei lorelei 512K Jul 22 20:52 BT_RAM_CODE_MT7961_1_2_hdr.bin
-rw-r--r-- 1 lorelei lorelei  91K Jul 22 20:52 WIFI_MT7961_patch_mcu_1_2_hdr.bin
-rw-r--r-- 1 lorelei lorelei 681K Jul 22 20:53 WIFI_RAM_CODE_MT7961_1.bin
```
And here is the sha512sum test I did with the files I already have in /lib/firmware/mediatek:
```bash
for f in BT_RAM_CODE_MT7961_1_2_hdr.bin WIFI_MT7961_patch_mcu_1_2_hdr.bin WIFI_RAM_CODE_MT7961_1.bin ; do
sha512sum ${f}
xzcat /lib/firmware/mediatek/${f}.xz |sha512sum
done
5bc77f5c3dd3478ff8c77097431e20a10c0d28d4d47a6b0f46c69fed0d48ad6d098c0b002e6240139b8875027b614b8a0393422250510d72fe66932007c364cf  BT_RAM_CODE_MT7961_1_2_hdr.bin
5bc77f5c3dd3478ff8c77097431e20a10c0d28d4d47a6b0f46c69fed0d48ad6d098c0b002e6240139b8875027b614b8a0393422250510d72fe66932007c364cf  -
05bdfe226151550c84ca318ef3d0eca1fcf9fc149def72eec2a261707e97751cffe2d96affbf8afc4504f243cffaaa704c20818abfe5f5ebd6da5d3bc9c6bff9  WIFI_MT7961_patch_mcu_1_2_hdr.bin
05bdfe226151550c84ca318ef3d0eca1fcf9fc149def72eec2a261707e97751cffe2d96affbf8afc4504f243cffaaa704c20818abfe5f5ebd6da5d3bc9c6bff9  -
42294298b3d8945aace5c0abcd758f77112f02b7349e670d45197b3d0b756c8b61b6a997e1a452e45a80bd8c1b67b9dfda1cdc6a2c76f97995d23f0c933bb265  WIFI_RAM_CODE_MT7961_1.bin
42294298b3d8945aace5c0abcd758f77112f02b7349e670d45197b3d0b756c8b61b6a997e1a452e45a80bd8c1b67b9dfda1cdc6a2c76f97995d23f0c933bb265  -
```
[image]

So I had the same problem with these firmware files and with Arch kernel `linux-5.18.12`. I have updated to 5.18.13 but currently am using the `linux-lts` to avoid the problem. Thank you.

a1b7c2d9, 2022-07-23:
Dear @objelf,

Thanks for the hint about the firmware.  Stable Fedora 35 has the previous version of the firmware, the new one is being tested right now.  Following your advice, I replaced the three bin.xz files by the latest ones, and rebooted with kernel 5.18.11.  This configuration looks stable at the first sight: no wifi crash after almost 2 hours of uptime, and dmesg is clean.  (In contrast, the crashy set-up had lots of driver failure messages in dmesg before the system collapsed.)

To be honest, I don't know how much I can trust the system's stability; after all, the 5.18.4 kernel with the old firmware crashed once a week or so on my laptop.  I'll report if I see strange things, but I try to be positive and optimistic for the time being.

Thank you, again.

objelf, 2022-07-25:
Hello @LoreleiPenn  @a1b7c2d9 

Thanks for all your help testing against the latest firmware from linux-firmware.git. I have been testing the kernel with v5.18.5 and the latest mt76 tree for two weeks, but I cannot reproduce the issue in my environment which is a laptop ROG-Zephyrus-G14-GA401H with CPU model AMD Ryzen7 running on.

It seemed to me it might be some platform difference that causes the issue might not easily happen in every machine. I wanna check if the PCI-E ASPM is all enabled in your machine when the issue happens.  If the mt7921 driver _accidentally_ accesses the PCIe bus in the L1s state (a low power state at the PCIe link), there would be an unexpected and possible driver error over there and trigger the 'chip reset' to self recover from time to time. So the first thing I want is to clarify if the issue is highly related to ASPM.

1. How to check if ASPM is supported by the kernel and platform, we see by the output of the following command:.
**journalctl -b | grep ASPM**
2. Disable ASPM and try again to see if the issue is still reproducible. We can use either way as the following to disable ASPM
- Force the kernel not to apply ASPM at boot time
  **Append the 'pcie_aspm=off'  to the kernel cmd line** by temporarily editing the GRUB menu
- Force the kernel not to apply ASPM powersave mode at runtime
  Using the command '**echo performance > /sys/module/pcie_aspm/parameters/policy'** in superuser to avoid PCIe possibly trapped to low power state.
3. Continue to test the MT7921 driver as usual.
4. Post the full dmesg here or attach in [https://termbin.com/](https://termbin.com/ ) for good if the issue is still there.

Thanks for all your help to make mt7921 better!

LoreleiPenn, 2022-07-26:
Hello @objelf 

In my case, the command `journalctl -b | grep ASPM` give me as output that ASPM is not supported in my system:
```
journalctl -b | grep ASPM
Jul 25 08:13:02 archlinux kernel: ACPI FADT declares the system doesn't support PCIe ASPM, so disable it
Jul 25 08:13:02 archlinux kernel: acpi PNP0A08:00: _OSC: OS supports [ExtendedConfig ASPM ClockPM Segments MSI EDR HPX-Type3]
Jul 25 08:13:02 archlinux kernel: acpi PNP0A08:00: FADT indicates ASPM is unsupported, using BIOS configuration
Jul 25 08:13:05 Galaxy kernel: r8169 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
```
On the other hand I have been testing the kernel `linux-5.18.14` for two consecutive days without the issue showing up. So if I get this issue, I will use https://termbin.com/ to share the full verbose dmesg log.

Thank you for your support.

a1b7c2d9, 2022-07-27:
Deal @objelf,

My case is similar to the case of @LoreleiPenn: no ASPM support.

So far, the combination of the newest firmware with kernel-5.18.11 has not reproduced the crash on my machine.  On one occasion, my system lost connection to the wifi router a few times and became unable to re-connect, but no error was reported in /var/log/messages or in dmesg.  (My tablet maintained a healthy internet connection via the same router.)  In contrast, kernel-5.8.11 with the out-of-date firmware keeps crashing, 5-90 minutes after booting.  The whole situation looks quite promising.

Thanks again for all your work.

a1b7c2d9, 2022-08-01:
Dear @objelf,

Unfortunately, the driver crash has popped up again for kernel-5.18.11 with the newest firmware.  You can find the complete dmesg file on https://bugzilla.redhat.com/show_bug.cgi?id=2107819.  If you need more information (e.g., /var/log/messages), please tell me how can I provide them.

Thank you.

objelf, 2022-08-03:
Dear @a1b7c2d9 

Applied with the patch in [1] can fix the kernel hung and make the system run even if something goes wrong in the driver
We are continuing to figure out why "mt7921e 0000:03:00.0: driver own failed" happens that is another issue we have to be clear about and find the root cause. Hopefully, we can update a new firmware for you to try to do some testing soon. If any update regarding the firmware, I will share it here. Really appreciate your effort and patience to make the mt7921 driver better.

[ 4980.998816]  mt7921e_mac_reset+0x9e/0x2d0 [mt7921e]
[ 4980.998827]  mt7921_mac_reset_work+0x8f/0x125 [mt7921_common]
[ 4980.998840]  process_one_work+0x1e5/0x3b0
[ 4980.998851]  worker_thread+0x50/0x3a0
[ 4980.998858]  ? rescuer_thread+0x370/0x370
[ 4980.998864]  kthread+0xe8/0x110
[ 4980.998869]  ? kthread_complete_and_exit+0x20/0x20
[ 4980.998875]  ret_from_fork+0x22/0x30

[1] [https://patchwork.kernel.org/project/linux-wireless/patch/727eb5ffd3c7c805245e512da150ecf0a7154020.1659452909.git.deren.wu@mediatek.com/](https://patchwork.kernel.org/project/linux-wireless/patch/727eb5ffd3c7c805245e512da150ecf0a7154020.1659452909.git.deren.wu@mediatek.com/)

a1b7c2d9, 2022-08-03:
Dear @objelf,

Thank you for the update.  I'll visit this site regularly, waiting for further developments.

Just one quick question.  If I understand the work-flow correctly, the patch you mentioned will appear in the official kernel at a certain point.  Is there an easy way to track whether the kernel has it?  (Pardon my ignorance, I don't feel comfortable to compile my own kernel.)

objelf, 2022-08-04:
Dear @a1b7c2d9 

yes, it would be eventually a part of the official kernel at some point, in general.  It would take probable two and more months and It cannot be part of a stable kernel until the patch has been accepted in the mainline kernel., so it would be a quite long journey for the end user can experience the patches.

LoreleiPenn, 2022-08-04:
Dear @objelf and @a1b7c2d9.

I can apply the kernel and try keeping a track of the behaviour of the driver. However I am usually kind of busy with work so have me patience please.

It has been a while since I manually patched and compiled a kernel (actually was a Gentoo user for quite a while 🙂).

I will try to have it running today. And no later than tomorrow night. 😊

LoreleiPenn, 2022-08-08:
Hello @objelf.

I have tried out the patch with linux-rc 5.18.13 version and reached again the point where the `driver own failed`. Here I share the segment in which the error happens:
```
Aug 08 16:58:59 Galaxy kernel: mt7921e 0000:02:00.0: driver own failed
Aug 08 16:58:59 Galaxy kernel: mt7921e 0000:02:00.0: chip reset
Aug 08 16:58:59 Galaxy kernel: mt7921e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220608210839a
Aug 08 16:58:59 Galaxy kernel: mt7921e 0000:02:00.0: WM Firmware Version: ____010000, Build Time: 20220608210922
Aug 08 16:59:00 Galaxy kernel: wlan0: Driver requested disconnection from AP e8:13:6e:76:55:98
Aug 08 16:59:03 Galaxy kernel: wlan0: authenticate with e8:13:6e:76:55:98
Aug 08 16:59:03 Galaxy kernel: wlan0: bad VHT capabilities, disabling VHT
Aug 08 16:59:03 Galaxy kernel: wlan0: Invalid HE elem, Disable HE
Aug 08 16:59:03 Galaxy kernel: wlan0: send auth to e8:13:6e:76:55:98 (try 1/3)
Aug 08 16:59:03 Galaxy kernel: wlan0: authenticated
Aug 08 16:59:03 Galaxy kernel: wlan0: associate with e8:13:6e:76:55:98 (try 1/3)
Aug 08 16:59:03 Galaxy kernel: wlan0: RX AssocResp from e8:13:6e:76:55:98 (capab=0x1431 status=0 aid=112)
Aug 08 16:59:03 Galaxy kernel: wlan0: associated
Aug 08 16:59:03 Galaxy kernel: wlan0: Limiting TX power to 21 (21 - 0) dBm as advertised by e8:13:6e:76:55:98
Aug 08 16:59:07 Galaxy kernel: audit: type=1130 audit(1659995947.521:158): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Aug 08 16:59:17 Galaxy kernel: audit: type=1131 audit(1659995957.549:159): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Aug 08 16:59:33 Galaxy kernel: mt7921e 0000:02:00.0: driver own failed
Aug 08 16:59:34 Galaxy kernel: mt7921e 0000:02:00.0: driver own failed
Aug 08 16:59:34 Galaxy kernel: mt7921e 0000:02:00.0: chip reset
Aug 08 16:59:35 Galaxy kernel: mt7921e 0000:02:00.0: driver own failed
Aug 08 16:59:36 Galaxy kernel: mt7921e 0000:02:00.0: Timeout for driver own
Aug 08 16:59:38 Galaxy kernel: mt7921e 0000:02:00.0: driver own failed
Aug 08 16:59:38 Galaxy kernel: ------------[ cut here ]------------
Aug 08 16:59:38 Galaxy kernel: WARNING: CPU: 11 PID: 17529 at kernel/kthread.c:659 kthread_park+0x7f/0x90
Aug 08 16:59:38 Galaxy kernel: Modules linked in: uhid snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm ccm cmac algif_hash algif_skcipher af_alg bnep btusb btrtl uvcvideo btbcm videobuf2_vmalloc btintel videobuf2_memops btmtk videobuf2_v4l2 videobuf2_common bluetooth videodev ecdh_generic mc 8021q garp mrp stp llc snd_sof_pci_intel_tgl snd_sof_intel_hda_common joydev mousedev soundwire_intel vfat fat soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_hdac_hda mei_pxp mei_hdcp snd_hda_ext_core spi_nor ee1004 iTCO_wdt snd_soc_acpi_intel_match mtd intel_pmc_bxt pmt_telemetry snd_soc_acpi intel_tcc_cooling iTCO_vendor_support intel_rapl_msr hid_multitouch pmt_class soundwire_bus x86_pkg_temp_thermal intel_powerclamp mt7921e asus_nb_wmi snd_soc_core coretemp mt7921_common asus_wmi platform_profile snd_compress mt76_connac_lib snd_hda_codec_realtek ac97_bus wmi_bmof snd_hda_codec_generic kvm_intel ledtrig_audio mt76
Aug 08 16:59:38 Galaxy kernel:  snd_hda_codec_hdmi snd_pcm_dmaengine snd_hda_intel kvm mac80211 snd_intel_dspcfg snd_intel_sdw_acpi processor_thermal_device_pci_legacy snd_hda_codec irqbypass processor_thermal_device snd_hda_core intel_cstate processor_thermal_rfim libarc4 intel_uncore r8169 processor_thermal_mbox snd_hwdep cfg80211 ucsi_acpi snd_pcm typec_ucsi processor_thermal_rapl pcspkr realtek spi_intel_pci i2c_i801 mei_me typec snd_timer intel_rapl_common spi_intel mdio_devres i2c_smbus snd libphy rfkill mei intel_vsec soundcore intel_soc_dts_iosf roles wmi i2c_hid_acpi i2c_hid int3403_thermal int340x_thermal_zone intel_hid mac_hid sparse_keymap asus_wireless int3400_thermal acpi_thermal_rel acpi_pad pkcs8_key_parser crypto_user fuse ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys trusted asn1_encoder tee dm_mod serio_raw crct10dif_pclmul atkbd nvme libps2 crc32_pclmul crc32c_intel nvme_core vivaldi_fmap ghash_clmulni_intel i915 tpm_crb aesni_intel crypto_simd
Aug 08 16:59:38 Galaxy kernel:  tpm_tis xhci_pci cryptd tpm_tis_core vmd xhci_pci_renesas i8042 tpm serio rng_core intel_gtt drm_buddy video drm_dp_helper ttm intel_lpss_pci intel_lpss idma64
Aug 08 16:59:38 Galaxy kernel: CPU: 11 PID: 17529 Comm: kworker/u24:0 Not tainted 5.18.13-rc1-1-rc #1 3090481eed4951e5bff7971d9447659813643225
Aug 08 16:59:38 Galaxy kernel: Hardware name: ASUSTeK COMPUTER INC. ASUS TUF Gaming F15 FX506HCB_FX506HCB/FX506HCB, BIOS FX506HCB.310 11/25/2021
Aug 08 16:59:38 Galaxy kernel: Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
Aug 08 16:59:38 Galaxy kernel: RIP: 0010:kthread_park+0x7f/0x90
Aug 08 16:59:38 Galaxy kernel: Code: 89 df e8 24 72 01 00 48 85 c0 74 27 31 c0 5b 5d c3 cc 0f 0b 48 8b ab d0 06 00 00 a8 04 74 af 0f 0b b8 da ff ff ff 5b 5d c3 cc <0f> 0b b8 f0 ff ff ff eb db 0f 0b eb d5 0f 1f 40 00 f3 0f 1e fa 0f
Aug 08 16:59:38 Galaxy kernel: RSP: 0018:ffffab50c678fe00 EFLAGS: 00010202
Aug 08 16:59:38 Galaxy kernel: RAX: 0000000000000004 RBX: ffff964b0bb3cd80 RCX: 0000000000000000
[... cut ...]
Aug 08 16:59:38 Galaxy kernel:  ? kthread_complete_and_exit+0x20/0x20
Aug 08 16:59:38 Galaxy kernel:  ret_from_fork+0x1f/0x30
Aug 08 16:59:38 Galaxy kernel:  </TASK>
Aug 08 16:59:38 Galaxy kernel: ---[ end trace 0000000000000000 ]---
```
and the full log can be check in here: https://termbin.com/ao9l

As with the other times I have experienced this error, my network turns useless and the system would only reboot with Sysrq REISUB. After restarting I have to shutdown the computer. Otherwise the WiFi card is not accessible for neither Linux nor Windows.

Lasoloz, 2022-09-14:
The issue happened to me as well for kernel `5.19` with the older 202202<something> version of the `mt7921` firmware.

I applied the patch as mentioned [before](https://github.com/openwrt/mt76/issues/681#issuecomment-1193030583), but I guess it's safe to assume that it will still probably occur with the newer `20220608210839a` build.

objelf, 2022-09-15:
@Lasoloz  @LoreleiPenn

I got help from the wireless maintainer and pushed the fixup patches to the Linus tree in [1]. I will backport the patch to k5.18 and k5.19 sooner to stop the kernel hang then the distribution will get the benefit from the newer stable kernel.

[1][https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/wireless/mediatek/mt76?id=fa3fbe64037839f448dc569212bafc5a495d8219]( https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/wireless/mediatek/mt76?id=fa3fbe64037839f448dc569212bafc5a495d8219)

dhewg, 2022-11-13:
It sounds like this is resolved and can be closed?

closed 2023-02-01 by kaienkira

xpt1x, 2023-05-22:
I need to revive this topic as I am facing kernel panic because of this
```zsh
journalctl -b -4 -g mt7921                                                                                                                                                    ✔ 
May 22 11:00:26 xpt1x-linux kernel: mt7921e 0000:03:00.0: enabling device (0000 -> 0002)
May 22 11:00:26 xpt1x-linux kernel: mt7921e 0000:03:00.0: ASIC revision: 79610010
May 22 11:00:26 xpt1x-linux kernel: mt7921e 0000:03:00.0: HW/SW Version: 0x8a108a10, Build Time: 20230302150916a
May 22 11:00:27 xpt1x-linux kernel: mt7921e 0000:03:00.0: WM Firmware Version: ____010000, Build Time: 20230302150956
May 22 11:00:27 xpt1x-linux NetworkManager[635]: <info>  [1684733427.8987] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:02.2/0000:03:00.0/ieee80211/phy0/rfkill>
May 22 11:00:27 xpt1x-linux kernel: mt7921e 0000:03:00.0 wlp3s0: renamed from wlan0
May 22 11:17:08 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:09 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:10 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:11 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:12 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:14 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:15 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:16 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:17 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:18 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:20 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:21 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:22 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:23 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:28 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:29 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:30 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
May 22 11:17:31 xpt1x-linux kernel: mt7921e 0000:03:00.0: Timeout for driver own
May 22 11:17:32 xpt1x-linux kernel: mt7921e 0000:03:00.0: driver own failed
```

Suddenly while surfing website stops responding (as wifi has dropped) and in seconds sometimes kernel panics (as I am seeing blinking caps and unresponsive system)
Then after rebooting wifi doesnt appear in network have to do 2-3 consecutive restarts/shutdowns to make it appear again

aritroCoder, 2023-06-22:
Same issue here. I have mt7921e wifi card and this issue happens in kernel `6.1.34`. 
Logs:
```bash
Jun 22 16:30:42 nixos ModemManager[4697]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.6/0000:04:00.0': not supported by any plugin
Jun 22 16:30:42 nixos ModemManager[4697]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.5/0000:03:00.0': not supported by any plugin
Jun 22 16:30:42 nixos ModemManager[4697]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3': not supported by any plugin
Jun 22 16:30:42 nixos kernel: mt7921e 0000:03:00.0: driver own failed
Jun 22 16:30:41 nixos kernel: mt7921e 0000:03:00.0: driver own failed
Jun 22 16:30:40 nixos dbus-daemon[874]: [system] Successfully activated service 'org.kde.powerdevil.backlighthelper'
Jun 22 16:30:40 nixos dbus-daemon[874]: [system] Activating service name='org.kde.powerdevil.backlighthelper' requested by ':1.35' (uid=1000 pid=1636 comm="/nix/store/fr4q5ssiq7n6hac2af8hv90hpjylxy43-powerd" label="kernel") (using serv>
Jun 22 16:30:40 nixos systemd[1]: Stopped target Post-Resume Actions.
Jun 22 16:30:40 nixos systemd[1]: Reached target Post-Resume Actions.
Jun 22 16:30:40 nixos systemd[1]: Finished Post-Resume Actions.
Jun 22 16:30:40 nixos systemd[1]: post-resume.service: Deactivated successfully.
Jun 22 16:30:40 nixos systemd[1]: Starting Post-Resume Actions...
Jun 22 16:30:40 nixos ModemManager[4697]: <info>  [sleep-monitor-systemd] system is resuming
Jun 22 16:30:40 nixos systemd-logind[945]: Operation 'sleep' finished.
Jun 22 16:30:40 nixos systemd[1]: Stopped target Sleep.
Jun 22 16:30:40 nixos systemd[1]: systemd-suspend.service: Consumed 1.514s CPU time, no IP traffic.
Jun 22 16:30:40 nixos systemd[1]: suspend.target: Job suspend.target/start failed with result 'dependency'.
Jun 22 16:30:40 nixos systemd[1]: Dependency failed for Suspend.
Jun 22 16:30:40 nixos systemd[1]: Failed to start System Suspend.
Jun 22 16:30:40 nixos plasmashell[1486]: Could not find the Plasmoid for Plasma::FrameSvgItem(0x1b8fb10) QQmlContext(0x1eb6c50) QUrl("file:///nix/store/n6zlcnm2mlxhdr4v8cv6q167q31j8zkx-plasma-workspace-5.27.5/share/plasma/plasmoids/org>
Jun 22 16:30:40 nixos plasmashell[1486]: Could not find the Plasmoid for Plasma::FrameSvgItem(0x1b8fb10) QQmlContext(0x1eb6c50) QUrl("file:///nix/store/n6zlcnm2mlxhdr4v8cv6q167q31j8zkx-plasma-workspace-5.27.5/share/plasma/plasmoids/org>
Jun 22 16:30:40 nixos systemd[1]: systemd-suspend.service: Failed with result 'exit-code'.
Jun 22 16:30:40 nixos systemd[1]: systemd-suspend.service: Main process exited, code=exited, status=1/FAILURE
Jun 22 16:30:40 nixos systemd-sleep[5119]: Failed to put system to sleep. System resumed again: Device or resource busy
Jun 22 16:30:40 nixos kernel: PM: suspend exit
Jun 22 16:30:40 nixos kernel: random: crng reseeded on system resumption
Jun 22 16:30:40 nixos kernel: Restarting tasks ... done.
```
I am using ASUS Tuf f15 laptop with nixos

sushantbasak, 2023-07-05:
@aritroCoder I was facing same issue in my ASUS TUF. It got fixed by changing the WIFI Hardware by ASUS Service.

zheltikov, 2023-07-26:
Just faced the same issue, WIFI randomly disconnected in the middle of my work :( 

dmesg shows the following logs:
```
driver own failed
Timeout for driver own
chip reset failed
Message 00020001 (seq 11) timeout
```

Im using Ubuntu 22.04.2 LTS, kernel 5.19.0, Lenovo Yoga 7

xpt1x, 2023-12-15:
> > @aritroCoder I was facing same issue in my ASUS TUF. It got fixed by changing the WIFI Hardware by ASUS Service.
> 
> Can you suggest me one I am thinking of throwing my mediatek chipset

Yeah, i was also thinking of replacing it, although just a workaround, dont knw if it will work for you or not, try opening up ur laptop and removing wires from wifi card and connecting it back.
First time when i read this thing i thought it doesnt make sense but still gave it a try, for me it worked for 2-3 months then i had the issue going on again for 2-3 days then i again did this thing and it went away again :laughing:

vikramnr, 2023-12-29:
I'm also facing the same issue with 6x version as well.

`6.6.6-76060606-generic`

[image]

ldl2030, 2024-04-25:
在双系统下切换到 Windows 下，使用快捷键关闭飞行模式，然后切回 linux 系统即可。
取下了 bios 电池重置 bios 固件，再次开机，wifi 恢复！

ldl2030, 2024-04-25:
> 在双系统下切换到 Windows 下，使用快捷键关闭飞行模式，然后切回 linux 系统即可。 取下了 bios 电池重置 bios 固件，再次开机，wifi 恢复！

最好是开启飞行模式后，立刻断电，强制关机