openwrt/mt76 #687: mt76x2u wireless broken with message 'vendor request req:07 off:10f4 failed:-110'

zhanglinqiang, 2022-07-25:
This fault occurs frequently. When it occurs, wifi device not working. reloading the mt76x2u driver  with modprobe cannot be recovered, and the system can only be restarted. Upgrading to the mainline kernel version (from 5.15.0.41.44 to 5.18.12) still has this problem. I also tried add 'options mt76_usb disable_usb_sg=1' to /etc/modprobe.d/mt76-usb.conf and Make sure the configuration was loaded correctly.Can any one help?

os version: Ubuntu 22.04.

kernel: Linux leenPC 5.18.12-051812-generic #202207150942 SMP PREEMPT_DYNAMIC Fri Jul 15 10:10:55 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux.

And 'systool -v -m mt76_usb' result information:
```
leen@leenPC:~$ systool -v -m mt76_usb
Module = "mt76_usb"

  Attributes:
    coresize            = "40960"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "2"
    srcversion          = "0FA481A5890E877A9C15D17"
    taint               = ""
    uevent              = <store method only>

  Parameters:
    disable_usb_sg      = "Y"

  Sections:

```
And 'lsusb' result information:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 413c:2113 Dell Computer Corp. KB216 Wired Keyboard
Bus 001 Device 003: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 002: ID 24ae:2013 Shenzhen Rapoo Technology Co., Ltd. Rapoo 2.4G Wireless Device
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

The dmesg and journalctl log information when occurs:

[dmesg.log](https://github.com/openwrt/mt76/files/9177928/dmesg.log)

[journalctl.log](https://github.com/openwrt/mt76/files/9177932/journalctl.log)

And after execute 'modprobe -r mt76x2u && modprobe mt76x2u', the dmesg and journalctl log: 

[dmesg-reload-mt76x2u.log](https://github.com/openwrt/mt76/files/9177941/dmesg-reload-mt76x2u.log)

[journalctl-reload-mt76x2u.log](https://github.com/openwrt/mt76/files/9177943/journalctl-reload-mt76x2u.log)

zhanglinqiang, 2022-07-25:
wifi crashed again after 3 hours. 
Here is the dmesg out put :
```
[    0.000000] Linux version 5.18.12-051812-generic (kernel@sita) (gcc (Ubuntu 11.3.0-4ubuntu1) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38.50.20220629) #202207150942 SMP PREEMPT_DYNAMIC Fri Jul 15 10:10:55 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.18.12-051812-generic root=/dev/mapper/vgubuntu-root ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: xstate_offset[9]:  960, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x21f, context size is 968 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 3632
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000092809fff] usable
[    0.000000] BIOS-e820: [mem 0x000000009280a000-0x000000009280afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009280b000-0x000000009282dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000009282e000-0x000000009282efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009282f000-0x0000000095c9afff] usable
[    0.000000] BIOS-e820: [mem 0x0000000095c9b000-0x0000000096c9afff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000096c9b000-0x0000000096dcafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000096dcb000-0x0000000096fe3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000096fe4000-0x0000000099ffefff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000099fff000-0x0000000099ffffff] usable
[    0.000000] BIOS-e820: [mem 0x000000009a000000-0x000000009e7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000085f7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x91a51018-0x91a6ea57] usable ==> usable
[    0.000000] e820: update [mem 0x91a51018-0x91a6ea57] usable ==> usable
[    0.000000] e820: update [mem 0x91a42018-0x91a50057] usable ==> usable
[    0.000000] e820: update [mem 0x91a42018-0x91a50057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000091a42017] usable
[    0.000000] reserve setup_data: [mem 0x0000000091a42018-0x0000000091a50057] usable
[    0.000000] reserve setup_data: [mem 0x0000000091a50058-0x0000000091a51017] usable
[    0.000000] reserve setup_data: [mem 0x0000000091a51018-0x0000000091a6ea57] usable
[    0.000000] reserve setup_data: [mem 0x0000000091a6ea58-0x0000000092809fff] usable
[    0.000000] reserve setup_data: [mem 0x000000009280a000-0x000000009280afff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000009280b000-0x000000009282dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000009282e000-0x000000009282efff] reserved
[    0.000000] reserve setup_data: [mem 0x000000009282f000-0x0000000095c9afff] usable
[    0.000000] reserve setup_data: [mem 0x0000000095c9b000-0x0000000096c9afff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000096c9b000-0x0000000096dcafff] ACPI data
[    0.000000] reserve setup_data: [mem 0x0000000096dcb000-0x0000000096fe3fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000096fe4000-0x0000000099ffefff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000099fff000-0x0000000099ffffff] usable
[    0.000000] reserve setup_data: [mem 0x000000009a000000-0x000000009e7fffff] reserved
[... 1186 lines cut ...]
[16488.826603] mt76x2u 1-4:1.0: vendor request req:07 off:1754 failed:-110
[16492.027123] mt76x2u 1-4:1.0: vendor request req:07 off:1758 failed:-110
[... cut ...]
[16540.283379] mt76x2u 1-4:1.0: vendor request req:07 off:1200 failed:-110
[16543.515383] mt76x2u 1-4:1.0: vendor request req:07 off:1200 failed:-110
[16546.747321] mt76x2u 1-4:1.0: vendor request req:07 off:1200 failed:-110
[16549.979216] mt76x2u 1-4:1.0: vendor request req:07 off:1004 failed:-110
[16553.179212] mt76x2u 1-4:1.0: vendor request req:06 off:1004 failed:-110
[16556.379082] mt76x2u 1-4:1.0: vendor request req:06 off:1004 failed:-110
[16559.578997] mt76x2u 1-4:1.0: vendor request req:06 off:1400 failed:-110
```

jaccc, 2022-08-04:
I have seen same error on MT7610U

cristian-ciobanu, 2022-09-24:
I'm experiencing the same issue. I'm using an EDUP AC-1605 wifi usb device and whenever I plug it in my desktop it start throwing a lot of errors
```
kernel: mt76x2u 4-4.2:1.0: vendor request req:07 off:1798 failed:-71
kernel: usb 4-4.2: Disable of device-initiated U1 failed.
kernel: usb 4-4.2: Disable of device-initiated U2 failed.
kernel: mt76x2u 4-4.2:1.0: MAC error detected

```
I also have used disable_usb_sg=1" >> /etc/modprobe.d/mt76_usb.conf and rebooted but even when connecting to an USB2 port it starts throwing errors like this in the log

```
kernel: mt76x2u 3-12:1.0: vendor request req:47 off:9018 failed:-110
kernel: mt76x2u 3-12:1.0: vendor request req:47 off:9018 failed:-110
```

patdohere, 2022-10-24:
> Duplicate of old unresolved bug #405 and also kernel bug [#202541](https://bugzilla.kernel.org/show_bug.cgi?id=202541). The disable_usb_sg only occasionality works for some people. The only solution is to plug it into a USB2 port instead of a USB3 port.

I'm currently running into similar issues but with a different [error code](https://github.com/openwrt/mt76/issues/704).  But I'm on a USB2 port so not sure if its entirely USB 3.0 related.

draxdeveloper, 2023-02-11:
Hello.

I am having the same issue, I tried to disable XHCI handoff and at the moment I did it it's seems that worked... But two days later I had the problem again so I can't say for sure if it's related.

Since it's an USB issue, maybe it's something related to the Mobo? So maybe if everyone that is having that issue post their specs we can reach a conclusion.

```
Operating System: Kubuntu 22.10
KDE Plasma Version: 5.25.5
KDE Frameworks Version: 5.98.0
Qt Version: 5.15.6
Kernel Version: 6.1.9-060109-generic (64-bit)
Graphics Platform: X11
Processors: 6 × AMD Ryzen 5 3500X 6-Core Processor
Memory: 15.6 GiB of RAM
Graphics Processor: AMD Radeon RX 560 Series
```

**This is mine.

And this is my motherboard:**

```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: PRIME B450M-GAMING/BR
        Version: Rev X.0x
        Serial Number: 200665183804602
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0
```

**lsusb results:**

```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 005: ID 04e8:6863 Samsung Electronics Co., Ltd Galaxy series, misc. (tethering mode)
Bus 003 Device 002: ID 30fa:0400  USB OPTICAL MOUSE 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
When I plug in a USB 2.0 it's stop detecting the 5.0ghz wi-fe

morrownr, 2023-02-11:
Interesting. I'll keep my eyes open for more info.

ulist, 2023-10-12:
```
c failed:-71
<3>[20160.743589] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20160.843573] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20160.950244] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20161.050238] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20161.150242] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20161.250233] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71                              <3>[20161.350228] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20161.450257] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20161.550240] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71                              <3>[20161.650341] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20161.750339] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<3>[20161.850338] mt7601u 1-1:1.0: Vendor request req:07 off:101c failed:-71
<6>[20161.862254] ufstw_clear_lu_flag_dynamic_tw:167 info: tw_flag LUN(0) [0x0E] 0
```

> 

>

schlosserleo, 2023-12-08:
Same issue here. I think its my Xbox wireless adapter that is recognized as wifi adapter(cause it also has 2,4GHZ). I use a intel wifi card so i know i dont have a mediatek chip. 
`$ dmesg | tail`
```
[  829.888463] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  833.089423] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  836.289488] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  839.489488] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  842.689521] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  845.889563] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  849.089134] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  852.288819] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  855.489085] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110
[  858.689601] mt76x2u 1-10:1.0: vendor request req:47 off:9018 failed:-110

```

dontholdmyhand, 2023-12-15:
Dec 15 18:52:42 kali /usr/libexec/gdm-x-session[8912]: thunar-volman: Unsupported USB device type "mt76x2u".
Dec 15 18:52:42 kali kernel: usb 1-1: reset high-speed USB device number 7 using ehci-pci
Dec 15 18:52:42 kali kernel: usb 1-1: unable to get BOS descriptor or descriptor too short
Dec 15 18:52:42 kali mtp-probe[8921]: checking bus 1, device 7: "/sys/devices/pci0000:00/0000:00:0b.0/usb1/1-1"
Dec 15 18:52:42 kali mtp-probe[8921]: bus: 1, device: 7 was not an MTP device
Dec 15 18:52:42 kali kernel: mt76x2u 1-1:1.0: vendor request req:07 off:0000 failed:-32
Dec 15 18:52:42 kali kernel: mt76x2u 1-1:1.0: ASIC revision: ffffffff
Dec 15 18:52:42 kali kernel: usb 1-1: USB disconnect, device number 7
Dec 15 18:52:42 kali dbus-daemon[526]: [system] Activating via systemd: service name='org.freedesktop.Avahi' unit='dbus-org.freedesktop.Avahi.service' requested by ':1.117' (uid=113 pid=8922 comm="/usr/libexec/colord-sane")
Dec 15 18:52:42 kali dbus-daemon[526]: [system] Activation via systemd failed for unit 'dbus-org.freedesktop.Avahi.service': Unit dbus-org.freedesktop.Avahi.service not found.
Dec 15 18:52:42 kali dbus-daemon[526]: [system] Activating via systemd: service name='org.freedesktop.Avahi' unit='dbus-org.freedesktop.Avahi.service' requested by ':1.118' (uid=113 pid=8922 comm="/usr/libexec/colord-sane")
Dec 15 18:52:42 kali dbus-daemon[526]: [system] Activation via systemd failed for unit 'dbus-org.freedesktop.Avahi.service': Unit dbus-org.freedesktop.Avahi.service not found.

something similar is happening to me, i thought it might be driver related? so i tried this 

[image]

but nothing worked, it seems to connect and get rejected by the system for some reason?

idk if these are correlated but this is where my google search led me

its an awus036acm btw

The-SamminAter, 2024-07-10:
> Same issue here. I think its my Xbox wireless adapter that is recognized as wifi adapter(cause it also has 2,4GHZ). I use a intel wifi card so i know i dont have a mediatek chip.

Thanks, that's what was causing the issue for me. The solution in my case was installing `xone-dongle-firmware`.

ghost9, 2024-07-12:
Facing a similar issue while connecting my usb mt7612u(Comfast CF-926AC V2) wifi dongle to virtualbox. After trying many solutions, I have found a [solution](https://github.com/openwrt/mt76/issues/405) that works for me may help others as well.

What works for me is disable scatter-gather.
Unplug the usb and run this command as root
`echo 1 > /sys/module/mt76_usb/parameters/disable_usb_sg`
plugin the usb again.

spiral009, 2024-07-12:
> Facing a similar issue while connecting my usb mt7612u(Comfast CF-926AC V2) wifi dongle to virtualbox. After trying many solutions, I have found a [...]

On my Android this path doesn't exist 
```
.../files/home # zcat /proc/config.gz |grep MT76
CONFIG_MT7601U=y
CONFIG_MT76_CORE=y
CONFIG_MT76_LEDS=y
CONFIG_MT76_USB=y
CONFIG_MT76x2_COMMON=y
CONFIG_MT76x0U=y
CONFIG_MT76x2E=y
CONFIG_MT76x2U=y
.../files/home # ls /sys/module/mt76_usb/
ls: cannot access '/sys/module/mt76_usb/': No such file or directory
.../files/home #
```

swoleportfolio, 2026-09-11:
> This fault occurs frequently. When it occurs, wifi device not working. reloading the mt76x2u driver with modprobe cannot be recovered, and the system can only be restarted. Upgrading to the mainline [...]

> On my Android this path doesn't exist
>
> ```
> .../files/home # zcat /proc/config.gz |grep MT76
> CONFIG_MT7601U=y
> CONFIG_MT76_CORE=y
> CONFIG_MT76_LEDS=y
> CONFIG_MT76_USB=y
> CONFIG_MT76x2_COMMON=y [...]

Can 

> Facing a similar issue while connecting my usb mt7612u(Comfast CF-926AC V2) wifi dongle to virtualbox. After trying many solutions, I have found a [...]

Can confirm in 2026 --> Root terminal: Unplug the usb and run this command as root
echo 1 > /sys/module/mt76_usb/parameters/disable_usb_sg
plugin the usb again.

Works.