openwrt/mt76 #689: mt7921e tx-power limited to 3 dbm

Gingernut1978, 2022-08-13:
Image built from latest master branch for a x86_64

[image]

`[   11.417551] mt7921e 0000:05:00.0: enabling device (0000 -> 0002)
[   11.445162] mt7921e 0000:05:00.0: ASIC revision: 79610010
[   11.548601] mt7921e 0000:05:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220311230842a
[   11.630784] mt7921e 0000:05:00.0: WM Firmware Version: ____010000, Build Time: 20220311230931`

Also tested with latested updated firmware from linux-firmware.git:
`[   12.093877] mt7921e 0000:05:00.0: enabling device (0000 -> 0002)
[   12.115789] mt7921e 0000:05:00.0: ASIC revision: 79610010
[   12.213882] mt7921e 0000:05:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220608210839a
[   12.305576] mt7921e 0000:05:00.0: WM Firmware Version: ____010000, Build Time: 20220608210922`

`BusyBox v1.35.0 (2022-08-11 17:38:10 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r20303+2-c0109537d1
 -----------------------------------------------------
root@MSI_CUBI:~# iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: managed
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
[... 315 lines cut ...]
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: 65535
	max scan plan iterations: 0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)`

Another user reported the same issue on the forum:
[https://forum.openwrt.org/t/tx-power-limited-to-3-dbm-on-an-amd-rz608-card/132917](url)

verycatdog, 2022-08-15:
experiencing on MT7921k PCIe as well

kerneltoast, 2022-08-21:
TX power isn't actually limited to 3 dBm, it's just that the driver can't change or report the TX power on mt7921/mt7922.

Gingernut1978, 2022-08-30:
Looks to be a visual bug and wireless TX power is actually not limited

closed 2022-08-30 by Gingernut1978

doublep21, 2024-12-06:
> Looks to be a visual bug and wireless TX power is actually not limited

how do you tested that? i think i can't actually set the tx power

sidiqsanjaya, 2025-01-31:
> Looks to be a visual bug and wireless TX power is actually not limited

> TX power isn't actually limited to 3 dBm, it's just that the driver can't change or report the TX power on mt7921/mt7922.

looks like not  only visual bug, but really limit tx power. because i just 3 meter away from router just got -74dBm with external antena.

```
root@LoliskWrt:~# iw phy phy1 set txpower fixed 2000
Interface phy1-ap0
        ifindex 12
        wdev 0x100000003
        addr xx:xx:xx:xx:xx
        ssid xxxxx
        type AP
        wiphy 1
        channel 149 (5745 MHz), width: 40 MHz, center1: 5755 MHz
        txpower 3.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       358     0       0       0       0       82306           401

root@LoliskWrt:~#  iwinfo phy1-ap0 txpower
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
*  3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
root@LoliskWrt:~# iw dev phy1-ap0 get power_save
Power save: off
root@LoliskWrt:~# ls /lib/firmware/mediatek/
WIFI_MT7922_patch_mcu_1_1_hdr.bin  WIFI_RAM_CODE_MT7922_1.bin
WIFI_MT7961_patch_mcu_1_2_hdr.bin  WIFI_RAM_CODE_MT7961_1.bin
root@LoliskWrt:~# lsmod | grep mt7921
cfg80211              724992  6 mt7921_common,ath10k_core,ath,mt76_connac_lib,mt76,mac80211
mac80211              933888  5 mt7921_common,ath10k_core,mt792x_lib,mt76_connac_lib,mt76
mt76                   81920  4 mt7921e,mt7921_common,mt792x_lib,mt76_connac_lib
mt76_connac_lib        49152  3 mt7921e,mt7921_common,mt792x_lib
mt7921_common          57344  1 mt7921e
mt7921e                16384  0
mt792x_lib             40960  2 mt7921e,mt7921_common

```

[image]

razvanphp, 2025-02-09:
I've tracked & fixed this bug, just submitted a wireless-kernel patch for it, should be backportable as well.

https://patchwork.kernel.org/project/linux-wireless/list/?series=931950

xmateox, 2025-02-24:
I don't think its just visual bug. I'm sitting with laptop with this card ~5meters from router and have over 50% retries for tx packets. If I go to other room, then ther is over 80% retries and everything going very slow, sometimes randomly disconnect. Meantime other laptop with intel wifi card tested in the same places has excellent connection - tx retries packets is below 0.1% and everything is blazing fast, even in another room. I believe that, the power level of mt7921 is truly locked on 3dBm.

[image]

niberius, 2025-04-11:
Hey @razvanphp ,

> I've tracked & fixed this bug, just submitted a wireless-kernel patch for it, should be backportable as well.
> https://patchwork.kernel.org/project/linux-wireless/list/?series=931950

Seems, the links is broken. Could you please provide a correct link or attach the patch somehow here?

sidiqsanjaya, 2025-04-12:
@niberius the patch already takeout because problem, i think, not tested on other kernel,
you can see on this
[https://github.com/openwrt/mt76/issues/783#issuecomment-2752387532](url)