openwrt/mt76 #690: mt7915e: Message timeout while waiting for mcu response

mriemensbergersw, 2022-08-15:
I'm seeing the following timeouts on a mt7621 board with mt7915e wifi:

```
[63464.928109] mt7915e 0000:02:00.0: Message 00005aed (seq 15) timeout
[63485.407585] mt7915e 0000:02:00.0: Message 0000aded (seq 1) timeout
[63505.887090] mt7915e 0000:02:00.0: Message 00005aed (seq 2) timeout
[63526.366557] mt7915e 0000:02:00.0: Message 000094ed (seq 3) timeout
[63546.846045] mt7915e 0000:02:00.0: Message 000094ed (seq 4) timeout
[63567.325544] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
[63587.805033] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
[63608.284525] mt7915e 0000:02:00.0: Message 00005aed (seq 7) timeout
[63628.764011] mt7915e 0000:02:00.0: Message 000025ed (seq 8) timeout
```

Once those messages show up, wifi is unusable. A reboot is necessary to fix it.

The board is running openwrt 21.02 with the mt76 from c67df0d3130a51d79b558f0329c2ca289c73b16e and with the latest mt76 master (minus multi-bss and WED, plus "mt76: lock mt76_queue before access").

The issue only reproduces with lots (40+) of connected active clients sometimes only after 40 hour and later after boot. Unfortunately, I can also not directly reproduce it locally myself. I only have reports from multiple deployed accesspoints.

It appears as if the mcu is locked up or at least communication with the mcu is no longer possible. SO it appears to be either a problem with the firmware or a problem with the communication between driver and firmware. Any ideas or suggestions further debug or fix the problem would be appreciated.

mriemensbergersw, 2022-08-15:
Another error log is:
```
[1313.808417] mt7915e 0000:02:00.0: Message 000007ed (seq 3) timeout
[1334.288216] mt7915e 0000:02:00.0: Message 000007ed (seq 4) timeout
```

bittorf, 2022-08-15:
I can confirm that. Archer C6U v1, OpenWRT trunk r20332

mriemensbergersw, 2022-08-17:
It seems that wifi scanning under certain conditions seems to trigger the issue. Is this related to #511 albeit for a different board? @nbd168 You apparently fixed #511 through a firmware update. Any idea whether mt7015e could exhibit a similar fw issue?

mriemensbergersw, 2022-08-17:
@ryderlee1110 @LorenzoBianconi @ptpt52 Since you were involved int #511, any ideas how to debug or fix this issue. Any help or ideas how to make progress in debugging or solving this issue would be appreciated.

everything411, 2022-08-23:
same problem on a Xiaomi Redmi AX6S running 22.03.0-rc6. 5Ghz wifi ping lantancy reaches about 1000ms with very slow connection speed which is about 200KB/s. rebooting the router does not help
```
[   85.248355] br-lan2: port 4(wlan1) entered blocking state
[   85.253750] br-lan2: port 4(wlan1) entered forwarding state
[ 2399.204837] mt7915e 0000:01:00.0: Message 00005aed (seq 8) timeout
[ 2419.684810] mt7915e 0000:01:00.0: Message 00005aed (seq 9) timeout
[ 2440.164800] mt7915e 0000:01:00.0: Message 00005aed (seq 10) timeout
[ 2460.644787] mt7915e 0000:01:00.0: Message 000026ed (seq 11) timeout
[ 2481.124774] mt7915e 0000:01:00.0: Message 00005aed (seq 12) timeout
[ 2501.604767] mt7915e 0000:01:00.0: Message 000026ed (seq 13) timeout
[ 2522.094915] mt7915e 0000:01:00.0: Message 00005aed (seq 14) timeout
[ 2542.564740] mt7915e 0000:01:00.0: Message 00005aed (seq 15) timeout
[ 2563.044735] mt7915e 0000:01:00.0: Message 00005aed (seq 1) timeout
[ 2583.524726] mt7915e 0000:01:00.0: Message 000026ed (seq 2) timeout
[ 2604.004707] mt7915e 0000:01:00.0: Message 00005aed (seq 3) timeout
[ 2624.484749] mt7915e 0000:01:00.0: Message 000026ed (seq 4) timeout
[ 2644.964696] mt7915e 0000:01:00.0: Message 00005aed (seq 5) timeout
```

RinCat, 2022-09-11:
Same issue on Linksys E8450 with OpenWRT 22.03 stable.
Once happened, the wifi is not usable(but can show SSID). It can be temporarily restored after a reboot.

```
kern.err kernel: [ 5905.757308] mt7915e 0000:01:00.0: Message 00005aed (seq 10) timeout
kern.err kernel: [ 5926.237086] mt7915e 0000:01:00.0: Message 00005aed (seq 11) timeout
kern.err kernel: [ 5946.716874] mt7915e 0000:01:00.0: Message 00005aed (seq 12) timeout
kern.err kernel: [ 5967.196667] mt7915e 0000:01:00.0: Message 000026ed (seq 13) timeout
kern.err kernel: [ 5987.676453] mt7915e 0000:01:00.0: Message 00005aed (seq 14) timeout
kern.err kernel: [ 6008.156298] mt7915e 0000:01:00.0: Message 00005aed (seq 15) timeout
kern.err kernel: [ 6028.636060] mt7915e 0000:01:00.0: Message 00005aed (seq 1) timeout
kern.err kernel: [ 6049.115844] mt7915e 0000:01:00.0: Message 000026ed (seq 2) timeout
daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
kern.err kernel: [ 6069.595634] mt7915e 0000:01:00.0: Message 00005aed (seq 3) timeout
kern.err kernel: [ 6090.075434] mt7915e 0000:01:00.0: Message 000025ed (seq 4) timeout
kern.err kernel: [ 6090.081639] wlan1: failed to remove key (0, xx:xx:xx:xx:xx:xx) from hardware (-110)
kern.err kernel: [ 6110.555218] mt7915e 0000:01:00.0: Message 00005aed (seq 5) timeout
kern.err kernel: [ 6131.035021] mt7915e 0000:01:00.0: Message 000025ed (seq 6) timeout
kern.info kernel: [ 6131.041226] wlan1: HW problem - can not stop rx aggregation for xx:xx:xx:xx:xx:xx tid 0
kern.err kernel: [ 6151.514715] mt7915e 0000:01:00.0: Message 00005aed (seq 7) timeout
kern.err kernel: [ 6171.994301] mt7915e 0000:01:00.0: Message 000025ed (seq 8) timeout
kern.debug kernel: [ 6172.000545] ------------[ cut here ]------------
kern.warn kernel: [ 6172.005174] WARNING: CPU: 1 PID: 1710 at 0xffffffc0088caa68 [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.013862] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat seqiv leds_gpio xhci_plat_hcd gpio_button_hotplug
kern.debug kernel: [ 6172.072741] CPU: 1 PID: 1710 Comm: hostapd Tainted: G S                5.10.138 #0
kern.debug kernel: [ 6172.080300] Hardware name: Linksys E8450 (UBI) (DT)
kern.debug kernel: [ 6172.085171] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
kern.debug kernel: [ 6172.091179] pc : 0xffffffc0088caa68 [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.097789] lr : 0xffffffc0088ca914 [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.104392] sp : ffffffc01158b800
kern.debug kernel: [ 6172.107697] x29: ffffffc01158b800 x28: ffffff80009d21c0
kern.debug kernel: [ 6172.113003] x27: ffffffc01158bdf0 x26: 0000000000000000
kern.debug kernel: [ 6172.118308] x25: ffffffc010acf000 x24: ffffff80021c90a0
kern.debug kernel: [ 6172.123612] x23: ffffff80021c60e8 x22: ffffffc0088c9144
kern.debug kernel: [ 6172.128916] x21: 0000000000000003 x20: ffffff8000d91a00
kern.debug kernel: [ 6172.134220] x19: ffffff800667f000 x18: 0000000000000161
kern.debug kernel: [ 6172.139525] x17: 00000000ffffffff x16: 0000000000000002
kern.debug kernel: [ 6172.144829] x15: ffffffc010a09998 x14: 0000000000000423
kern.debug kernel: [ 6172.150132] x13: 0000000000000161 x12: ffffffc01158b2f8
kern.debug kernel: [ 6172.155436] x11: ffffffc010a61998 x10: 00000000fffff000
kern.debug kernel: [ 6172.160740] x9 : ffffffc010a61998 x8 : 0000000000000000
kern.debug kernel: [ 6172.166044] x7 : ffffffc010a09998 x6 : 0000000000000001
kern.debug kernel: [ 6172.171348] x5 : 0000000000000000 x4 : 0000000000000000
kern.debug kernel: [ 6172.176652] x3 : 0000000000000000 x2 : 0000000000000001
kern.debug kernel: [ 6172.181956] x1 : 0000000000000002 x0 : 00000000ffffff92
kern.debug kernel: [ 6172.187261] Call trace:
kern.debug kernel: [ 6172.189705]  0xffffffc0088caa68 [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.195968]  0xffffffc0088c9144 [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.202227]  0xffffffc0088bf82c [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.208488]  0xffffffc0088c238c [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.214747]  0xffffffc0088d8a08 [mac80211@00000000d8798a7e+0x79000]
kern.debug kernel: [ 6172.221009]  0xffffffc0088865c4 [cfg80211@00000000e034fde9+0x44000]
kern.debug kernel: [ 6172.227265]  0xffffffc0106bc8d4
kern.debug kernel: [ 6172.230397]  0xffffffc0106bcc60
kern.debug kernel: [ 6172.233527]  0xffffffc0106bbab8
kern.debug kernel: [ 6172.236658]  0xffffffc0106bc234
kern.debug kernel: [ 6172.239790]  0xffffffc0106bb2e4
kern.debug kernel: [ 6172.242921]  0xffffffc0106bb570
kern.debug kernel: [ 6172.246052]  0xffffffc01062c500
kern.debug kernel: [ 6172.249184]  0xffffffc01062f858
kern.debug kernel: [ 6172.252315]  0xffffffc01062f924
[... cut ...]
kern.err kernel: [ 6438.229291] mt7915e 0000:01:00.0: Message 0000aded (seq 6) timeout
kern.err kernel: [ 6458.708933] mt7915e 0000:01:00.0: Message 00005aed (seq 7) timeout
kern.err kernel: [ 6479.188575] mt7915e 0000:01:00.0: Message 00005aed (seq 8) timeout
kern.err kernel: [ 6499.668244] mt7915e 0000:01:00.0: Message 00005aed (seq 9) timeout
kern.err kernel: [ 6520.147894] mt7915e 0000:01:00.0: Message 000026ed (seq 10) timeout
kern.err kernel: [ 6540.627544] mt7915e 0000:01:00.0: Message 00005aed (seq 11) timeout
```

ktgeek, 2022-09-11:
This is pure anecdotal evidence… but i was getting this daily with schedutil as the scheduler on my BRT3200. I switched back to ondemand a few days ago and it hasn’t happened since.Schedutil had the CPU running at a higher frequency on average, which could be more head… but that’s only speculation in what could be making the difference.Keith T.

kern.warn kernel: [ 6172.005174] WARNING: CPU: 1 PID: 1710 at 0xffffffc0088caa68 ***@***.***+0x79000]
kern.debug kernel: [ 6172.091179] pc : 0xffffffc0088caa68 ***@***.***+0x79000]
kern.debug kernel: [ 6172.097789] lr : 0xffffffc0088ca914 ***@***.***+0x79000]
kern.debug kernel: [ 6172.189705]  0xffffffc0088caa68 ***@***.***+0x79000]
kern.debug kernel: [ 6172.195968]  0xffffffc0088c9144 ***@***.***+0x79000]
kern.debug kernel: [ 6172.202227]  0xffffffc0088bf82c ***@***.***+0x79000]
kern.debug kernel: [ 6172.208488]  0xffffffc0088c238c ***@***.***+0x79000]
kern.debug kernel: [ 6172.214747]  0xffffffc0088d8a08 ***@***.***+0x79000]
kern.debug kernel: [ 6172.221009]  0xffffffc0088865c4 ***@***.***+0x44000]

nbd168, 2022-09-11:
You can write 1 to /sys/kernel/debug/ieee80211/phy*/mt76/fw_debug_wm and the firmware will print debug messages. Hopefully it will indicate where it is crashing

RinCat, 2022-09-12:
New log with debug info.
```
daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
kern.info kernel: [43330.952585] ieee80211 phy1: WA: Invalid baDirection
kern.info kernel: [43331.092026] ieee80211 phy1: WM: ( 362.767836:56:MQM-W)[WR] BSS usage overflows during removing entry
kern.info kernel: [43331.101243] ieee80211 phy1: WA: ucWlanIdx = 2, fgEnable = 0, ucAc = 15, u4MaxQLen = 2193
kern.info kernel: [43331.109342] ieee80211 phy1: WA: ==>rxmDeleteStaRec, prStaRecord->fgIsInUse = 1
daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
kern.info kernel: [43333.329291] ieee80211 phy1: WM: ( 365.004561:57:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
kern.info kernel: [43333.338648] ieee80211 phy1: WM: ( 365.013930:58:TXC-E)heACtrlInitStaRec(): prOperMode=401308, u2OperMode=51!
daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
daemon.notice hostapd: wlan1: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx RADIUS: starting accounting session A65A6B5BBA020564
daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
kern.info kernel: [43333.498698] ieee80211 phy1: WA: Invalid baDirection
daemon.notice netifd: Network device 'lan3' link is down
kern.info kernel: [44112.187084] mt7530 mdio-bus:00 lan3: Link is Down
kern.info kernel: [44112.192047] br-lan: port 3(lan3) entered disabled state
kern.info kernel: [44114.411110] mt7530 mdio-bus:00 lan3: Link is Up - 10Mbps/Full - flow control off
kern.info kernel: [44114.418538] br-lan: port 3(lan3) entered blocking state
kern.info kernel: [44114.423783] br-lan: port 3(lan3) entered forwarding state
daemon.notice netifd: Network device 'lan3' link is up
daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
kern.info kernel: [45118.277898] ieee80211 phy1: WA: Invalid baDirection
kern.info kernel: [45118.377341] ieee80211 phy1: WM: (2149.624953:59:MQM-W)[WR] BSS usage overflows during removing entry
kern.info kernel: [45118.386498] ieee80211 phy1: WA: ucWlanIdx = 2, fgEnable = 0, ucAc = 15, u4MaxQLen = 2193
kern.info kernel: [45118.394617] ieee80211 phy1: WA: ==>rxmDeleteStaRec, prStaRecord->fgIsInUse = 1
kern.info kernel: [45120.562681] ieee80211 phy1: WM: (2151.809768:60:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45120.621582] ieee80211 phy1: WM: (2151.868636:61:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45120.631182] ieee80211 phy1: WM: (2151.870955:62:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45120.820749] ieee80211 phy1: WM: (2152.067794:63:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45120.924803] ieee80211 phy1: WM: (2152.171828:64:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45121.052827] ieee80211 phy1: WM: (2152.299819:65:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45121.148159] ieee80211 phy1: WM: (2152.395125:66:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45121.224042] ieee80211 phy1: WM: (2152.470961:67:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45121.967933] ieee80211 phy1: WM: (2153.214705:68:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.004239] ieee80211 phy1: WM: (2153.250991:69:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.058598] ieee80211 phy1: WM: (2153.305343:70:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.158862] ieee80211 phy1: WM: (2153.405593:71:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.209652] ieee80211 phy1: WM: (2153.456343:72:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.249646] ieee80211 phy1: WM: (2153.496352:73:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.300190] ieee80211 phy1: WM: (2153.546859:74:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.433545] ieee80211 phy1: WM: (2153.680190:75:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.626761] ieee80211 phy1: WM: (2153.873366:76:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.664796] ieee80211 phy1: WM: (2153.911391:77:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45122.715881] ieee80211 phy1: WM: (2153.962478:78:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45123.606024] ieee80211 phy1: WM: (2154.852401:79:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45123.750816] ieee80211 phy1: WM: (2154.997145:80:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45123.803435] ieee80211 phy1: WM: (2155.049758:81:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45123.841175] ieee80211 phy1: WM: (2155.087478:82:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45123.915872] ieee80211 phy1: WM: (2155.162185:83:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[... cut ...]
kern.info kernel: [45200.435122] ieee80211 phy1: WM: (2231.663039:98:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45200.640734] ieee80211 phy1: WM: (2231.868636:99:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45200.945863] ieee80211 phy1: WM: (2232.173690:00:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.info kernel: [45201.151548] ieee80211 phy1: WM: (2232.379348:01:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
kern.err kernel: [45201.225819] mt7915e 0000:01:00.0: Message 000026ed (seq 15) timeout
```

bittorf, 2022-09-12:
```
# snapshot r20539
$ echo 1 >/sys/kernel/debug/ieee80211/phy1/mt76/fw_debug
$ wifi up
# next command hangs for ~16 sec:
$ echo 1 >/sys/kernel/debug/ieee80211/phy1/mt76/fw_debug
$ dmesg
[47859.318557] ieee80211 phy1: N9: (47895.829:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[47859.345363] ieee80211 phy1: N9: (47895.856:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[47860.589939] ieee80211 phy1: N9: Wifi ASSERT @ ../wifi/open_corebase/core/wificore/mlm/bss_cfg.c:225
[47880.848577] mt7615e 0000:02:00.0: Message 000026ed (seq 8) timeout
[47901.328563] mt7615e 0000:02:00.0: Message 000025ed (seq 9) timeout
[47921.808570] mt7615e 0000:02:00.0: Message 000049ed (seq 10) timeout
[47942.288564] mt7615e 0000:02:00.0: Message 000013ed (seq 11) timeout
[47962.768557] mt7615e 0000:02:00.0: Message 000025ed (seq 12) timeout
[47962.774844] wlan1: failed to move IBSS STA 64:09:80:7f:48:86 to state 3 (-145) - keeping it anyway
[47983.248555] mt7615e 0000:02:00.0: Message 000049ed (seq 13) timeout
[48006.288556] mt7615e 0000:02:00.0: Message 000049ed (seq 14) timeout
[48026.768630] mt7615e 0000:02:00.0: Message 000025ed (seq 15) timeout
```

everything411, 2022-09-13:
```
root@OpenWrt:~# dmesg
[75410.438944] ieee80211 phy1: WM: (2370.905700:20:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75420.441856] ieee80211 phy1: WM: (2380.906127:21:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75430.444861] ieee80211 phy1: WM: (2390.906646:22:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75440.448327] ieee80211 phy1: WM: (2400.907623:23:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75450.451538] ieee80211 phy1: WM: (2410.908355:24:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75460.454558] ieee80211 phy1: WM: (2420.908874:25:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75470.457900] ieee80211 phy1: WM: (2430.909728:26:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75480.461567] ieee80211 phy1: WM: (2440.910918:27:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75490.464830] ieee80211 phy1: WM: (2450.911681:28:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75500.468029] ieee80211 phy1: WM: (2460.912383:29:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75510.471025] ieee80211 phy1: WM: (2470.912902:30:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75520.474113] ieee80211 phy1: WM: (2480.913512:31:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75530.476946] ieee80211 phy1: WM: (2490.913848:32:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75540.479884] ieee80211 phy1: WM: (2500.914306:33:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75550.483306] ieee80211 phy1: WM: (2510.915221:34:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75560.487012] ieee80211 phy1: WM: (2520.916442:35:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75570.490028] ieee80211 phy1: WM: (2530.916961:36:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75580.492631] ieee80211 phy1: WM: (2540.917083:37:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75590.495247] ieee80211 phy1: WM: (2550.917205:38:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75600.498281] ieee80211 phy1: WM: (2560.917754:39:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75610.501859] ieee80211 phy1: WM: (2570.918853:40:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75620.505485] ieee80211 phy1: WM: (2580.919982:41:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75630.508532] ieee80211 phy1: WM: (2590.920531:42:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75640.511174] ieee80211 phy1: WM: (2600.920684:43:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75650.514125] ieee80211 phy1: WM: (2610.921142:44:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75660.517251] ieee80211 phy1: WM: (2620.921783:45:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75670.520292] ieee80211 phy1: WM: (2630.922332:46:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75680.523630] ieee80211 phy1: WM: (2640.923187:47:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75690.527158] ieee80211 phy1: WM: (2650.924224:48:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75700.530782] ieee80211 phy1: WM: (2660.925353:49:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75710.533431] ieee80211 phy1: WM: (2670.925536:50:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75720.536644] ieee80211 phy1: WM: (2680.926238:51:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75730.539363] ieee80211 phy1: WM: (2690.926482:52:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75740.542231] ieee80211 phy1: WM: (2700.926849:53:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75750.545770] ieee80211 phy1: WM: (2710.927917:54:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75760.548550] ieee80211 phy1: WM: (2720.928191:55:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75770.551911] ieee80211 phy1: WM: (2730.929076:56:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75780.555421] ieee80211 phy1: WM: (2740.930083:57:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75790.558802] ieee80211 phy1: WM: (2750.930968:58:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75800.561983] ieee80211 phy1: WM: (2760.931670:59:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75810.565518] ieee80211 phy1: WM: (2770.932708:60:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75820.568983] ieee80211 phy1: WM: (2780.933685:61:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75830.572018] ieee80211 phy1: WM: (2790.934234:62:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75840.574990] ieee80211 phy1: WM: (2800.934722:63:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75850.577898] ieee80211 phy1: WM: (2810.935149:64:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75860.581128] ieee80211 phy1: WM: (2820.935882:65:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75870.583713] ieee80211 phy1: WM: (2830.935973:66:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75880.586950] ieee80211 phy1: WM: (2840.936736:67:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75890.590308] ieee80211 phy1: WM: (2850.937591:68:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[75900.593510] ieee80211 phy1: WM: (2860.938293:69:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[... cut ...]
[77781.188527] ieee80211 phy1: WM: ( 446.098041:57:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[77791.192169] ieee80211 phy1: WM: ( 456.099201:58:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[77801.195530] ieee80211 phy1: WM: ( 466.100056:59:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[77811.198590] ieee80211 phy1: WM: ( 476.100635:60:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
[77821.201484] ieee80211 phy1: WM: ( 486.101032:61:MEM-W)whFbmWfdmaCmdEventXmit: xmit fail (out of resource)
```

ryderlee1110, 2022-09-13:
Is this a long standing issue? Or can only be reproduced after certain commits?

RinCat, 2022-09-13:
> Is this a long standing issue? Or can only be reproduced after certain commits?

I can only say a few months ago openwrt snapshots don't have this issue.

ryderlee1110, 2022-09-13:
> > Is this a long standing issue? Or can only be reproduced after certain commits?
> 
> I can only say a few months ago openwrt snapshots don't have this issue.

can you help to bisect those snapshots?

RinCat, 2022-09-13:
> > > Is this a long standing issue? Or can only be reproduced after certain commits?
> > 
> > 
> > I can only say a few months ago openwrt snapshots don't have this issue.
> 
> can you help to bisect those snapshots?

I can try, but the issue happened randomly at max 1-2 times a day. So it will be slow.

ryderlee1110, 2022-09-13:
> > > > Is this a long standing issue? Or can only be reproduced after certain commits?
> > > 
> > > 
> > > I can only say a few months ago openwrt snapshots don't have this issue.
> > 
> > 
> > can you help to bisect those snapshots?
> 
> I can try, but the issue happened randomly at max 1-2 times a day. So it will be slow.

It would be helpful if someone can bisect snapshots/commits as log only tells us something gets stuck.

RinCat, 2022-09-14:
bisect started, cannot estimate required time.

RinCat, 2022-09-18:
I encountered 2 issues that maybe related.

First is happened near https://github.com/openwrt/mt76/commit/506bb0605e3e2fa4c5285e6c8866167691a6d71b, where after:
```
[63036.892782] ieee80211 phy1: WA: Invalid baDirection
[63036.897915] ieee80211 phy1: WA: Invalid baDirection
[63036.992048] ieee80211 phy1: WM: (2883.981691:21:MQM-W)[WR] BSS usage overflows during removing entry
[63037.001232] ieee80211 phy1: WA: ucWlanIdx = 255, fgEnable = 0, ucAc = 15, u4MaxQLen = 2193
[63037.009531] ieee80211 phy1: WA: ==>rxmDeleteStaRec, prStaRecord->fgIsInUse = 1
[63037.823005] ieee80211 phy1: WA: [in] rxmRecvMacDma1R3
[63037.828068] ieee80211 phy1: WA: --> Error RxD Type:0 on MAC_DMA1 RxQ3, send to host!
[63038.119422] ieee80211 phy1: WA: [in] rxmRecvMacDma1R3
```
```
wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
```
The wifi is unable to connect, which is this issue.

Early than that, near https://github.com/openwrt/mt76/commit/eed44048ca575863ebdf1dc73f441cb9604cd388. After sometime I got very slow speed but WiFi connectable. I think it maybe related.

https://github.com/openwrt/mt76/compare/eed44048ca575863ebdf1dc73f441cb9604cd388...506bb0605e3e2fa4c5285e6c8866167691a6d71b
The exact commit is still being tested. I only test around 24h if it not happened, assuming the commit is OK.

everything411, 2022-09-20:
My Xiaomi Redmi AX6S was running a snapshot of openwrt which is built at the end of March (forgot the exact date) before running 22.03.0-rc6. The build from March suffers from the same problem.

ptpt52, 2022-10-09:
I can confirm that before 202207300730 on such an issue, but now the issue comes up to my devices

ptpt52, 2022-10-09:
https://github.com/openwrt/mt76/commit/93e3fce916c62d06892d41bf00d0f4c2926c9a0b

checkout to this commit, no such an issue

@nbd168 
@ryderlee1110

ptpt52, 2022-10-09:
UPDATE:
checkout this commit, still no issue
5ec78e1ec43d1e39edfea1efb9fd4541fa004af0

ptpt52, 2022-10-09:
UPDATE:
checkout to this commit, still no issue
9d601f4eee8f7ca3a64fab9f5e969c62b7d39637

ptpt52, 2022-10-09:
so the issue comes with these commits:
```
pick e49b6063 wifi: mt76: move mt76_rate_power from core to mt76x02 driver code
pick 3f27f6ad wifi: mt76: mt76x02: simplify struct mt76x02_rate_power
pick c07f3d2d wifi: mt76: mt7921: fix antenna signal are way off in monitor mode
pick 9059a5de wifi: mt76: Remove unused inline function mt76_wcid_mask_test()
pick d75f15dd wifi: mt76: mt7915: fix bounds checking for tx-free-done command
pick 06df7e68 wifi: mt76: mt7915: reserve 8 bits for the index of rf registers
pick ad3d0f8d wifi: mt76: mt7915: rework eeprom tx paths and streams init
pick 66065073 wifi: mt76: mt7915: deal with special variant of mt7916
pick b0114a0a wifi: mt76: mt7915: rework testmode tx antenna setting
pick 6dee964e wifi: mt76: connac: introduce mt76_connac_spe_idx()
pick 48c116d9 wifi: mt76: mt7915: add spatial extension index support
pick db6db4de wifi: mt76: mt7915: set correct antenna for radar detection on MT7915D
pick 2b8f56a7 wifi: mt76: mt7915: fix mt7915_mac_set_timing()
pick d554a025 wifi: mt76: mt7915: move wed init routines in mmio.c
pick 61ce40e6 wifi: mt76: mt7915: enable wed for mt7986 chipset
pick 584a96ec wifi: mt76: mt7915: enable wed for mt7986-wmac chipset
pick 172d68b6 mt76: mt76x02: fix vht rate power array overrun
pick 72b87836 Revert "mt76: use IEEE80211_OFFLOAD_ENCAP_ENABLED instead of MT_DRV_AMSDU_OFFLOAD"
pick 8cc34bdc wifi: mt76: mt7921: set MT_DRV_AMSDU_OFFLOAD for USB/SDIO
pick 2182de50 wifi: mt76: fix receiving LLC packets on mt7615/mt7915
pick 269df4b0 wifi: mt76: fix rx checksum offload on mt7615/mt7915/mt7921
```

ptpt52, 2022-10-09:
@nbd168 
@ryderlee1110 

revert commit ad3d0f8db00b81a1abeab9793a9d4064ffa00a04
the issue fixed for me.

ryderlee1110, 2022-10-09:
> @nbd168 @ryderlee1110
> 
> revert commit [ad3d0f8](https://github.com/openwrt/mt76/commit/ad3d0f8db00b81a1abeab9793a9d4064ffa00a04) the issue fixed for me.

Are you sure about that? This patch was just sent couple days ago...

ptpt52, 2022-10-09:
> > @nbd168 @ryderlee1110
> > revert commit [ad3d0f8](https://github.com/openwrt/mt76/commit/ad3d0f8db00b81a1abeab9793a9d4064ffa00a04) the issue fixed for me.
> 
> Are you sure about that? This patch was just sent couple days ago...

it is sure for me. I tested it.

everything411, 2022-10-09:
> > > @nbd168 @ryderlee1110
> > > revert commit [ad3d0f8](https://github.com/openwrt/mt76/commit/ad3d0f8db00b81a1abeab9793a9d4064ffa00a04) the issue fixed for me.
> > 
> > 
> > Are you sure about that? This patch was just sent couple days ago...
> 
> it is sure for me. I tested it.

maybe there are several different bugs causing the same (or similar) error. my device was running a build from March 2022 and  sufferred from low speed. so i built 22.03.0-rc6 and upgraded the firmware but it's speed got worse and the device was unconnectable in about 1 day after a reboot. now i'm using https://github.com/padavanonly/immortalwrtmt7622, which contains MTKSDK drivers, and it works well with low lantacy and high speed.

ptpt52, 2022-10-09:
@ryderlee1110 @nbd168 
I could fixed this issue by this patch:
```diff
diff --git a/mt7915/eeprom.c b/mt7915/eeprom.c
index e2482c65..92336849 100644
--- a/mt7915/eeprom.c
+++ b/mt7915/eeprom.c
@@ -206,6 +206,9 @@ void mt7915_eeprom_parse_hw_cap(struct mt7915_dev *dev,
 
                if (!is_mt7986(&dev->mt76))
                        nss_max = 2;
+
+               if (!nss)
+                       nss = nss_max;
        }
 
        nss = min_t(u8, min_t(u8, nss_max, nss), path);

```

ryderlee1110, 2022-10-09:
> 

This woks - I guess there's an issue in your eeprom, so we need this check just in case.

ryderlee1110, 2022-10-09:
> @ryderlee1110 @nbd168 I could fix this issue by this patch:
> 
> ```diff
> diff --git a/mt7915/eeprom.c b/mt7915/eeprom.c
> index e2482c65..92336849 100644
> --- a/mt7915/eeprom.c
> +++ b/mt7915/eeprom.c
> @@ -206,6 +206,9 @@ void mt7915_eeprom_parse_hw_cap(struct mt7915_dev *dev,
>  
>                 if (!is_mt7986(&dev->mt76))
>                         nss_max = 2;
> +
> +               if (!nss)
> +                       nss = nss_max;
>         }
>  
>         nss = min_t(u8, min_t(u8, nss_max, nss), path);
> ```

This woks - I guess there's an issue in your eeprom, so we need this check just in case.

ptpt52, 2022-10-10:
> > @ryderlee1110 @nbd168 I could fix this issue by this patch:
> > ```diff
> > diff --git a/mt7915/eeprom.c b/mt7915/eeprom.c
> > index e2482c65..92336849 100644
> > --- a/mt7915/eeprom.c
> > +++ b/mt7915/eeprom.c
> > @@ -206,6 +206,9 @@ void mt7915_eeprom_parse_hw_cap(struct mt7915_dev *dev,
> >  
> >                 if (!is_mt7986(&dev->mt76))
> >                         nss_max = 2;
> > +
> > +               if (!nss)
> > +                       nss = nss_max;
> >         }
> >  
> >         nss = min_t(u8, min_t(u8, nss_max, nss), path);
> > ```
> 
> This works - I guess there's an issue in your eeprom, so we need this check just in case. Are you able to submit patch to linux-wireless?

It seems more convenient for you to submit a patch, I suggest you do that.

ernie-t, 2022-11-20:
@ptpt52 @ryderlee1110 @nbd168 - are there any updates? Does anybody submit the patch?

ryderlee1110, 2022-11-20:
https://github.com/openwrt/mt76/commit/0263711ba362e36d6dcdcd2461c1a7ad7cf5098c you can close this ticket.

atanas-vladimirov, 2022-12-14:
Hi, 

Looks like we hit this issue on TP-Link EAP615-Wall v1

```
[1546759.716252] mt7915e 0000:02:00.0: Message 00005aed (seq 10) timeout                                                                                                                      
[1546780.196128] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout                                                                                                                      
[1546800.676003] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout                                                                                                                      
[1546821.155873] mt7915e 0000:02:00.0: Message 00005aed (seq 13) timeout                                                                                                                      
[1546841.635748] mt7915e 0000:02:00.0: Message 00005aed (seq 14) timeout   
```   
                                                                                                   
Running the 22.03.2

```
root@ap10:/etc/config# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
```
As a can see this has been fixed in master, and the question is - do you plan to backport it to 22.03? Or I have to update to a master snapshot (or buitl it myself)?

Thanks,
Atanas

simontretter, 2022-12-16:
Same issue on OpenWRT snapshot r21471-b7a7a97037 with the AW7915-NP1 card.

> Fri Dec 16 15:04:09 2022 kern.err kernel: [ 7217.805605] mt7915e 0000:05:00.0: Message 00005aed (seq 14) timeout
Fri Dec 16 15:04:29 2022 kern.err kernel: [ 7238.285427] mt7915e 0000:05:00.0: Message 00005aed (seq 15) timeout
Fri Dec 16 15:04:50 2022 kern.err kernel: [ 7258.765338] mt7915e 0000:05:00.0: Message 000026ed (seq 1) timeout
Fri Dec 16 15:05:10 2022 kern.err kernel: [ 7279.245276] mt7915e 0000:05:00.0: Message 00005aed (seq 2) timeout
Fri Dec 16 15:05:31 2022 kern.err kernel: [ 7299.725261] mt7915e 0000:05:00.0: Message 00005aed (seq 3) timeout
Fri Dec 16 15:05:51 2022 kern.err kernel: [ 7320.205022] mt7915e 0000:05:00.0: Message 00005aed (seq 4) timeout
Fri Dec 16 15:06:12 2022 kern.err kernel: [ 7340.684938] mt7915e 0000:05:00.0: Message 000026ed (seq 5) timeout
Fri Dec 16 15:06:32 2022 kern.err kernel: [ 7361.164873] mt7915e 0000:05:00.0: Message 00005aed (seq 6) timeout
Fri Dec 16 15:06:53 2022 kern.err kernel: [ 7381.644852] mt7915e 0000:05:00.0: Message 00005aed (seq 7) timeout
Fri Dec 16 15:07:13 2022 kern.err kernel: [ 7402.124726] mt7915e 0000:05:00.0: Message 00005aed (seq 8) timeout
Fri Dec 16 15:07:33 2022 kern.err kernel: [ 7422.604647] mt7915e 0000:05:00.0: Message 000026ed (seq 9) timeout
Fri Dec 16 15:07:54 2022 kern.err kernel: [ 7443.084555] mt7915e 0000:05:00.0: Message 00005aed (seq 10) timeout
Fri Dec 16 15:08:14 2022 kern.err kernel: [ 7463.564713] mt7915e 0000:05:00.0: Message 00005aed (seq 11) timeout
Fri Dec 16 15:08:35 2022 kern.err kernel: [ 7484.044454] mt7915e 0000:05:00.0: Message 00005aed (seq 12) timeout
Fri Dec 16 15:08:55 2022 kern.err kernel: [ 7504.524364] mt7915e 0000:05:00.0: Message 000026ed (seq 13) timeout
Fri Dec 16 15:09:16 2022 kern.err kernel: [ 7525.004309] mt7915e 0000:05:00.0: Message 00005aed (seq 14) timeout

mateuszdraco, 2023-01-18:
Same issue with Netgear WAX202:

```
Wed Jan 18 21:32:59 2023 kern.err kernel: [ 6322.464657] mt7915e 0000:02:00.0: Message 0000aded (seq 6) timeout
Wed Jan 18 21:33:19 2023 kern.err kernel: [ 6342.944737] mt7915e 0000:02:00.0: Message 00005aed (seq 7) timeout
Wed Jan 18 21:33:40 2023 kern.err kernel: [ 6363.424804] mt7915e 0000:02:00.0: Message 0000aded (seq 8) timeout
Wed Jan 18 21:34:00 2023 kern.err kernel: [ 6383.904880] mt7915e 0000:02:00.0: Message 00005aed (seq 9) timeout
Wed Jan 18 21:34:21 2023 kern.err kernel: [ 6404.384952] mt7915e 0000:02:00.0: Message 0000aded (seq 10) timeout
Wed Jan 18 21:34:41 2023 kern.err kernel: [ 6424.865020] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
Wed Jan 18 21:35:02 2023 kern.err kernel: [ 6445.345088] mt7915e 0000:02:00.0: Message 0000aded (seq 12) timeout
Wed Jan 18 21:35:22 2023 kern.err kernel: [ 6465.825154] mt7915e 0000:02:00.0: Message 00005aed (seq 13) timeout
Wed Jan 18 21:35:43 2023 kern.err kernel: [ 6486.305231] mt7915e 0000:02:00.0: Message 0000aded (seq 14) timeout
Wed Jan 18 21:36:03 2023 kern.err kernel: [ 6506.785293] mt7915e 0000:02:00.0: Message 00005aed (seq 15) timeout
Wed Jan 18 21:36:24 2023 kern.err kernel: [ 6527.265366] mt7915e 0000:02:00.0: Message 0000aded (seq 1) timeout
Wed Jan 18 21:36:44 2023 kern.err kernel: [ 6547.745440] mt7915e 0000:02:00.0: Message 00005aed (seq 2) timeout
Wed Jan 18 21:37:05 2023 kern.err kernel: [ 6568.225478] mt7915e 0000:02:00.0: Message 0000aded (seq 3) timeout
Wed Jan 18 21:37:25 2023 kern.err kernel: [ 6588.705554] mt7915e 0000:02:00.0: Message 00005aed (seq 4) timeout
Wed Jan 18 21:37:45 2023 kern.err kernel: [ 6609.185617] mt7915e 0000:02:00.0: Message 0000aded (seq 5) timeout
Wed Jan 18 21:38:06 2023 kern.err kernel: [ 6629.665661] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
Wed Jan 18 21:38:26 2023 kern.err kernel: [ 6650.145721] mt7915e 0000:02:00.0: Message 0000aded (seq 7) timeout
Wed Jan 18 21:38:47 2023 kern.err kernel: [ 6670.625778] mt7915e 0000:02:00.0: Message 00005aed (seq 8) timeout
Wed Jan 18 21:39:07 2023 kern.err kernel: [ 6691.105828] mt7915e 0000:02:00.0: Message 0000aded (seq 9) timeout
Wed Jan 18 21:39:28 2023 kern.err kernel: [ 6711.585871] mt7915e 0000:02:00.0: Message 00005aed (seq 10) timeout
Wed Jan 18 21:39:48 2023 kern.err kernel: [ 6732.065909] mt7915e 0000:02:00.0: Message 0000aded (seq 11) timeout
Wed Jan 18 21:40:09 2023 kern.err kernel: [ 6752.545947] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
Wed Jan 18 21:40:25 2023 user.info : luci: accepted login on / for root from 192.168.0.159
Wed Jan 18 21:40:29 2023 kern.err kernel: [ 6773.026006] mt7915e 0000:02:00.0: Message 00005aed (seq 13) timeout
Wed Jan 18 21:40:39 2023 authpriv.info dropbear[3097]: Child connection from 192.168.0.159:59537
Wed Jan 18 21:40:43 2023 authpriv.notice dropbear[3097]: Password auth succeeded for 'root' from 192.168.0.159:59537
Wed Jan 18 21:40:50 2023 kern.err kernel: [ 6793.506042] mt7915e 0000:02:00.0: Message 00005aed (seq 14) timeout
Wed Jan 18 21:41:10 2023 kern.err kernel: [ 6813.986074] mt7915e 0000:02:00.0: Message 00005aed (seq 15) timeout
Wed Jan 18 21:41:31 2023 kern.err kernel: [ 6834.466108] mt7915e 0000:02:00.0: Message 00005aed (seq 1) timeout
Wed Jan 18 21:41:51 2023 kern.err kernel: [ 6854.946158] mt7915e 0000:02:00.0: Message 00005aed (seq 2) timeout
Wed Jan 18 21:42:12 2023 kern.err kernel: [ 6875.436216] mt7915e 0000:02:00.0: Message 00005aed (seq 3) timeout
Wed Jan 18 21:42:32 2023 kern.err kernel: [ 6895.906246] mt7915e 0000:02:00.0: Message 00005aed (seq 4) timeout
Wed Jan 18 21:42:53 2023 kern.err kernel: [ 6916.386280] mt7915e 0000:02:00.0: Message 0000aded (seq 5) timeout
Wed Jan 18 21:43:13 2023 kern.err kernel: [ 6936.866314] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
Wed Jan 18 21:43:34 2023 kern.err kernel: [ 6957.346346] mt7915e 0000:02:00.0: Message 0000aded (seq 7) timeout
Wed Jan 18 21:43:54 2023 kern.err kernel: [ 6977.826391] mt7915e 0000:02:00.0: Message 00005aed (seq 8) timeout
Wed Jan 18 21:44:15 2023 kern.err kernel: [ 6998.306416] mt7915e 0000:02:00.0: Message 0000aded (seq 9) timeout
Wed Jan 18 21:44:35 2023 kern.err kernel: [ 7018.786455] mt7915e 0000:02:00.0: Message 00005aed (seq 10) timeout
Wed Jan 18 21:44:56 2023 kern.err kernel: [ 7039.266521] mt7915e 0000:02:00.0: Message 0000aded (seq 11) timeout
Wed Jan 18 21:45:16 2023 kern.err kernel: [ 7059.746527] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
Wed Jan 18 21:45:37 2023 kern.err kernel: [ 7080.226573] mt7915e 0000:02:00.0: Message 0000aded (seq 13) timeout
Wed Jan 18 21:45:57 2023 kern.err kernel: [ 7100.706610] mt7915e 0000:02:00.0: Message 00005aed (seq 14) timeout
Wed Jan 18 21:46:03 2023 authpriv.info dropbear[3097]: Exit (root) from <192.168.0.159:59537>: Disconnect received
Wed Jan 18 21:46:17 2023 kern.err kernel: [ 7121.186671] mt7915e 0000:02:00.0: Message 0000aded (seq 15) timeout
Wed Jan 18 21:46:26 2023 authpriv.info dropbear[3124]: Child connection from 192.168.0.159:60900
Wed Jan 18 21:46:30 2023 authpriv.notice dropbear[3124]: Password auth succeeded for 'root' from 192.168.0.159:60900
Wed Jan 18 21:46:38 2023 kern.err kernel: [ 7141.666673] mt7915e 0000:02:00.0: Message 00005aed (seq 1) timeout
Wed Jan 18 21:46:58 2023 kern.err kernel: [ 7162.146713] mt7915e 0000:02:00.0: Message 00005aed (seq 2) timeout
[... cut ...]
Wed Jan 18 21:49:01 2023 kern.err kernel: [ 7285.026919] mt7915e 0000:02:00.0: Message 0000aded (seq 8) timeout
Wed Jan 18 21:49:22 2023 kern.err kernel: [ 7305.506950] mt7915e 0000:02:00.0: Message 00005aed (seq 9) timeout
Wed Jan 18 21:49:42 2023 kern.err kernel: [ 7325.986987] mt7915e 0000:02:00.0: Message 0000aded (seq 10) timeout
Wed Jan 18 21:50:03 2023 kern.err kernel: [ 7346.467031] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
Wed Jan 18 21:50:23 2023 kern.err kernel: [ 7366.947059] mt7915e 0000:02:00.0: Message 0000aded (seq 12) timeout
```

stweil, 2023-01-21:
I'm afraid I have the same issue with Freifunk running on a Zyxel NWA50AX. Is it possible to dump the mt7915 eeprom from the Linux command line? I tried `hexdump -n 4096 -C /dev/mtdblock2`. Is that the right command? How can I see from that dump whether the antenna selection needs the fix mentioned above because of a bad eeprom?

Brain2000, 2023-02-01:
It looks like the original commit was the clean up, but it missed one thing:
old code:

```
     if (!nss_band || nss_band > nss_band_max)
		nss_band = nss_band_max; <-- note that nss_band = nss_band_max

	if (nss_band > nss) {
		dev_warn(dev->mt76.dev,
			 "nss mismatch, nss(%d) nss_band(%d) band(%d) ext_phy(%d)\n",
			 nss, nss_band, phy->band_idx, ext_phy);
		nss = nss_band;  <--- note that nss = nss_band
	}

```

This was all reduced to this:
```
nss = min_t(u8, min_t(u8, nss_max, nss), path);
```

But in the reduction, as all the variables were renamed, nss_band is now nss and nss is now path.

So this logic was essentially removed with the initial commit:
```
	if (nss > path) {
		dev_warn(dev->mt76.dev,
			 "nss mismatch, nss(%d) nss_band(%d) band(%d) ext_phy(%d)\n",
			 path, nss, phy->band_idx, ext_phy);
		path = nss;
	}
```
unless this was intentional

darmach, 2023-03-02:
@csyuanc @nbd168 Can you confirm if above scenario mentioned by @Brain2000 could be the root cause?

Brain2000, 2023-03-03:
@darmach Another possible flaw I didn't notice at first:
```
nss = min_t(u8, min_t(u8, nss_max, nss), path);
```
This will set nss to the minimum value between nss_max, nss, and path.
The previous code set nss to nss_max if it was 0, which is not the minimum value.
```
        if (!nss || nss > nss_max)
		nss = nss_max;
```
That will no longer happen. Perhaps that was also intentional though?

mkyral, 2023-03-13:
Today this happen to me on my Turris Omnia. I had to restart router via ethernet.

```
root@turris:/srv/home/root# grep mt7915e messages |tail
Mar 13 19:39:31 turris kernel: [554176.052299] mt7915e 0000:02:00.0: Message 0000aded (seq 8) timeout
Mar 13 19:39:52 turris kernel: [554196.520900] mt7915e 0000:02:00.0: Message 00002ced (seq 9) timeout
Mar 13 19:40:12 turris kernel: [554217.009554] mt7915e 0000:02:00.0: Message 00002ced (seq 10) timeout
Mar 13 19:40:33 turris kernel: [554237.478208] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
Mar 13 19:40:53 turris kernel: [554257.956872] mt7915e 0000:02:00.0: Message 00002ced (seq 12) timeout
Mar 13 19:41:14 turris kernel: [554278.445516] mt7915e 0000:02:00.0: Message 00002ced (seq 13) timeout
Mar 13 19:41:34 turris kernel: [554298.924205] mt7915e 0000:02:00.0: Message 00002ced (seq 14) timeout
Mar 13 19:41:55 turris kernel: [554319.392842] mt7915e 0000:02:00.0: Message 0000aded (seq 15) timeout
Mar 13 19:42:15 turris kernel: [554339.871491] mt7915e 0000:02:00.0: Message 0000aded (seq 1) timeout
Mar 13 19:42:36 turris kernel: [554360.360153] mt7915e 0000:02:00.0: Message 00002ced (seq 2) timeout

root@turris:/srv/home/root# uname -a
Linux turris 5.15.94 #0 SMP Fri Feb 17 00:25:12 2023 armv7l GNU/Linux

root@turris:/srv/home/root# cat /etc/os-release
NAME="TurrisOS"
VERSION="6.2.4"
ID="turrisos"
ID_LIKE="lede openwrt"
PRETTY_NAME="TurrisOS 6.2.4"
VERSION_ID="6.2.4"
HOME_URL="https://www.turris.cz/"
BUG_URL="https://gitlab.nic.cz/groups/turris/-/issues/"
SUPPORT_URL="https://www.turris.cz/support/"
BUILD_ID="r16824+125-0d4a0250df"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3-d16"
OPENWRT_TAINTS="busybox"
OPENWRT_DEVICE_MANUFACTURER="CZ.NIC"
OPENWRT_DEVICE_MANUFACTURER_URL="https://www.turris.cz/"
OPENWRT_DEVICE_PRODUCT="Turris Omnia"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="TurrisOS 6.2.4 0d4a0250df17a9f7b0fff720fa3224b45d5b0841"

root@turris:/srv/home/root# lspci -v -s 02:00.0
02:00.0 Unclassified device [0002]: MEDIATEK Corp. MT7915E 802.11ax PCI Express Wireless Network Adapter (prog-if 80)
        Subsystem: MEDIATEK Corp. MT7915E 802.11ax PCI Express Wireless Network Adapter
        Physical Slot: 2
        Flags: bus master, fast devsel, latency 0, IRQ 97
        Memory at e0400000 (64-bit, prefetchable) [size=1M]
        Memory at e0500000 (64-bit, prefetchable) [size=16K]
        Memory at e0504000 (64-bit, prefetchable) [size=4K]
        Capabilities: [80] Express Endpoint, MSI 00
        Capabilities: [e0] MSI: Enable+ Count=1/32 Maskable+ 64bit+
        Capabilities: [f8] Power Management version 3
        Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
        Capabilities: [108] Latency Tolerance Reporting
        Capabilities: [110] L1 PM Substates
        Capabilities: [200] Advanced Error Reporting
        Kernel driver in use: mt7915e
lspci: Unable to load libkmod resources: error -12

```

webknjaz, 2023-03-28:
I haven't seen this on my Turris Omnia (TurrisOS 6.3.0 c874aa40c22bec81643af57413c3f9ad613aeced r16833+124-c874aa40c2).
But I do have a few **TOTOLINK X5000R** devices set up to run OpenWRT 21.02.5 (snapshot r16688-fa9a932fdb) as dumb APs and they exhibit the same problem. This also seems to make the devices flakily inaccessible over the wired network too — the HTTP access times out.
I managed to connect over SSH and noticed load average being close to 5.0.
Using `top`, I saw that `hostapd` and `dawn` are in the `D` state which is concerning. Also, `[mt76-tx phy0]` remains in the `SW` state...

hami89, 2023-04-05:
Exact same behavior experienced with my Linksys e8450 (OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-23.069.50944-cd8bea9) running as router of a 3 unit setup with WIFI roaming (other two units are same hardware, but running as dumb APs).
Like @webknjaz wrote, SSH access is working and hostapd and dawn are indeed in D state.

rany2, 2023-04-24:
I know correlation does not imply causation so this might be a freak coincidence but I only started facing this issue within a few hours of booting up after deploying 802.11r and DAWN, before that change it was working reliably for more than week. After uninstalling DAWN, the issue seems gone. I'm not sure what to make of this information but it might help the developers to see what DAWN is doing that causes the chip to freak out.

Brain2000, 2023-04-24:
@rany2 Aha... Dawn ... That's a really good point.

Based on observation, I've been leaning towards something being wrong with multicast, particularly mDNS.
iPhones cause crashes when a lot of them are roaming.
I can manually cause a crash with a laptop in under 10 seconds by sending a large amount of mDNS packets and then either turning on airplane mode or roam (if I stay connected and stop the multicast stream, everything remains fine).

I am going to disable Dawn and see what happens. Clearly Dawn should not cause an issue, and switching it from mDNS to broadcast might be a temporary workaround until the larger issue can be found.

rany2, 2023-04-24:
@Brain2000 DAWN could very well be the issue, it actively sends beacon requests to STAs via a hostapd interface and passively listens to beacons from STAs. I don't think it's MDNS related. 

Of course, DAWN isn't the source but some bug in the firmware or kernel driver. Whatever it is, the router I had was working flawlessly up until I had setup roaming and DAWN. DAWN being uninstalled seems to have gotten rid of the issue reliably.

stweil, 2023-04-24:
My routers get these timeouts without DAWN, but I run Freifunk Gluon with a wireless mesh. I get the timeout much sooner on routers with much wireless traffic.

hami89, 2023-04-24:
My 3-piece system (3 Linksys e8450 - 1 as router + 2 as dumb APs with DAWN) worked flawlessly for almost three weeks, then last week it experienced this issue again.
During those three weeks, I had guests, so we had several computers and phones roaming around the house, streaming media etc. So I see no connection between the actual load or number of clients and this effect.
I did not see any problems during that period, surprisingly, the system failed when only I was home.

I could reconnect to the network through one of the dumb APs, so I was able to connect to the router (controller). I could log into luci, but the page load timed out every time, but I could get in SSH.
Again, hostapd and dawn were in D state (checked in top).

I checked logs and tried to restart networking, without success, so after 5 mins I restarted the whole system as it occurred during a company call and I had to get back,

Anyone interested, I have a dump of logs (some comments and the network restart attempt added in between).
Errors in log started to show up around Wed Apr 19 11:13:12 2023.
[openwrt-logs-prepared.txt](https://github.com/openwrt/mt76/files/11314567/openwrt-logs-prepared.txt)

rany2, 2023-04-24:
@hami89 Interesting. On my end I've noticed that this issues comes up only with 802.11r and DAWN setup. I was reliably getting triggering it with 802.11r and DAWN setup. DAWN without 802.11r doesn't have an issue. It's solely when 802.11r is setup alongside DAWN that this issue gets triggered extremely reliably. 

No idea what to make of this information :/

rany2, 2023-04-24:
At any rate I'm getting this issue in under 2 hours and sometimes within 30 minutes of bootup, so it seems a different situation than mine.

hami89, 2023-04-24:
Previously it occurred twice, within 30 minutes of the boot, the third time it worked for 3+ weeks. Probably it depends on your DAWN setup and the distance from the APs.
When it happened last time, 2 of my three devices were on my desk (because I was configuring them), and I suspected it happened because my laptop was bouncing back and forth between them.
Unfortunately, I don't have the logs from the last time, but I can remember that there was a device "kick" back then seconds before the issue, and I can see it now too. I am suspecting some problem there when a device gets kicked off the network due to low transmit rates.

rany2, 2023-04-25:
@hami89 I just noticed that prior to this crash, I have a lot of spam about rekeying with the message being "Driver cannot safely do that." Could you attest to that being present in your log as well?

rany2, 2023-04-25:
No connection, I just noticed that without DAWN and GTK rekeying disabled I still get this in my log: 
```
[   44.757366] br-guest5ghz: port 2(guest5ghz) entered forwarding state
[ 2938.101592] mt7915e 0000:02:00.0: Message 000025ed (seq 13) timeout
[ 2958.581317] mt7915e 0000:02:00.0: Message 00005aed (seq 14) timeout
[ 2979.061195] mt7915e 0000:02:00.0: Message 0000aded (seq 15) timeout
[ 2999.540864] mt7915e 0000:02:00.0: Message 00005aed (seq 1) timeout
[ 3020.020650] mt7915e 0000:02:00.0: Message 00005aed (seq 2) timeout
[ 3040.500425] mt7915e 0000:02:00.0: Message 000025ed (seq 3) timeout
[ 3060.980235] mt7915e 0000:02:00.0: Message 000025ed (seq 4) timeout
[ 3081.460010] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
[ 3101.939820] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
[ 3122.419658] mt7915e 0000:02:00.0: Message 00005aed (seq 7) timeout
[ 3142.899443] mt7915e 0000:02:00.0: Message 00005aed (seq 8) timeout
[ 3163.379253] mt7915e 0000:02:00.0: Message 00005aed (seq 9) timeout
[ 3183.859057] mt7915e 0000:02:00.0: Message 00005aed (seq 10) timeout
[ 3204.338834] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
[ 3224.818604] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
```

Only difference this time around is that it did not lock out.

rany2, 2023-04-25:
Nope, it just happened yet again but I don't want to disable 802.11r so I guess I'll just suck it up for now. Hopefully someone more talented will figure out why this is happening and fix it. It occurs on both SNAPSHOT and latest stable.

rany2, 2023-04-26:
@hami89 @Brain2000 I've found a solution that worked for me, at least it hasn't crashed for more than 14 hours which is a record (it was consecutively crashing in under 2 hours for me). Reverting to linux-firmware 13dca280f76009ba2c5f25408543a1aaaa062c25's mt7915 firmware fixed it. If you want a script that automates this, first uninstall kmod-mt7915-firmware and run:
```
#!/bin/sh
cd /lib/firmware/mediatek || exit 1
# https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=13dca280f76009ba2c5f25408543a1aaaa062c25
# https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/mediatek/mt7915_rom_patch.bin
id=13dca280f76009ba2c5f25408543a1aaaa062c25
for file in mt7915_rom_patch.bin mt7915_wa.bin mt7915_wm.bin
do
wget -O $file https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/mediatek/$file?id=$id
done
sync; reboot now
```

Of course, don't forget to add these blobs to you /etc/sysupgrade.conf:
```
/lib/firmware/mediatek/mt7915_rom_patch.bin
/lib/firmware/mediatek/mt7915_wa.bin
/lib/firmware/mediatek/mt7915_wm.bin
```

rany2, 2023-04-26:
It crashed just now, but that firmware version does crash less. It took some time for it to crash, unlike the latest.

Might be a mix of some firmware and kernel driver bug?

Brain2000, 2023-04-26:
@rany2 @nbd168 I found a way to reproduce the timeouts. Dawn may be an issue, but the script below will get the wifi into the timeout state within a couple of minutes.

Hook a laptop to the wifi unit, put the code below into multicast.py and run "python multicast.py".
Wait about 30 seconds for it to really saturate the wifi and then hit the airplane mode on the laptop (don't let the transmission stop or it will recover almost every time).

You may have to do this 2-3 times, and then suddenly the timeouts will start appearing in the log and the wifi becomes unusable without a reboot. I can't even rmmod the mt7915e module.
```py
import sys
import socket
import array
from time import sleep

UDP_IP="224.0.0.251"
UDP_PORT="5353"
TTL=2
DATA=b"flajshdflkjashdflkjhasdlkfjhwlueiryluiashdfljhasljkdfhlkajsdhfl ashdfljkashdlfkjhaslkdjfhlaskdfhwhateverandever"

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.setsockopt(socket.IPPROTO_IP, socket.IP_MULTICAST_TTL, TTL)

while True:
	sock.sendto(DATA, (UDP_IP, int(UDP_PORT)))
	sleep(0)
```

rany2, 2023-04-27:
@Brain2000 I'm unable to trigger it that way but I could confirm that I see a TON of "invalid misc" in my iwinfo while doing that and the WiFi does momentarily stop working.

rany2, 2023-04-27:
@Brain2000 I take it back, took me a couple of tries but I got it to crash with your script. Great job!

[136 more replies]