openwrt/mt76 #692: MT7603E WIFI on a MT7621 possible hanging

shown19, 2022-08-27:
I don't know if this is where I should post this issue but I'm guessing that this might be related to this WIFI driver so let me just share it here. I'm using a Newifi D2 router with the latest snapshot build as of this moment. One thing that I noticed is that the 2.4ghz WIFI (MT7603E) is like there was a time that it feels like it's freezing or hanging for like 30secs or 1minute and during that moment, I can't browse to the internet or even ping my router. It's like we're connected but we don't have connections. I don't know if this is also related but I only experience this mostly when the devices connected reaches more than five in 2.4ghz. It sucks when I'm playing online games and suddenly the ping reaches 999 and it took 30secs or more to go back to its normal state so by the time it happens, I switch to the 5ghz connection so I can play again. As for the 5ghz connection, I can't test it further because only 3 devices supported in this connection so everything is smooth and normal, it's just the 2.4ghz just bugging or something.

JFtico, 2022-11-18:
Yep, same on a ZBT-WE3526, none of the major releases (19.07.10, 21.x, 22.03.2) has a decent 2.4GHZ performance. Poor range, lots of hangs, device connected but no internet, etc.
I've posted data to various issues threads, and they don't seem to be addressed.

I got on this platform because it supposedly had 'open' drivers and good support. Oops.

shown19, 2022-11-19:
Learned the hardway, anyway, I gave up on using the wifi of this thing and just used my spare router which has better wifi stability and just mainly used this thing for network management because other one is not an openwrt supported. So yeah, I don't think no software updates could really fixed this issue anymore and my guess is that it's already a hardware limitation.

JFtico, 2022-12-12:
> So yeah, I don't think no software updates could really fixed this issue anymore and my guess is that it's already a hardware limitation.

Oh, it's a software problem, just one that does not seem to get fixed in the Opensource versions. There is this project that uses the MTK native code to build a usable driver for OpenWRT: https://github.com/Azexios/openwrt-r3p-mtk

You'll likely need to make necessary adjustments for your specific hardware.

sinancetinkaya, 2023-01-07:
> Oh, it's a software problem, just one that does not seem to get fixed in the Opensource versions. There is this project that uses the MTK native code to build a usable driver for OpenWRT: [...]

Unfortunately, I gave up on the open source driver as well :(
How can I build openwrt with mtk drivers? Any guide will be really helpful.

ThiloteE, 2023-01-09:
Have you already applied modifications from https://github.com/openwrt/openwrt/pull/11220 and checked if it fixes your problems?

ThiloteE, 2023-01-13:
> I don't know if this is also related but I only experience this mostly when the devices connected reaches more than five in 2.4ghz

Looks like connection deteriorates with multiple clients, hence "multiple streams", which is probably something to be expected.
Check out https://forum.openwrt.org/t/best-practices-and-tools-for-measuring-wifi-performance/146923
and you can see in more details if there are any imbalances between streams, if you test via iperf3, e.g. like this:

Run following command at the serverside: `Iperf3 -s`

Run one of the following commands at the client side:

TCP test, client sends, server receives
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -t 60 -i 10

TCP test with 4 Client Streams, client sends, server receives
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -P 4 -t 60 -i 10

TCP test with 8 Client Streams, client sends, server receives
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -P 8 -t 60 -i 10

TCP test; Reverse mode: server sends, client receives
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -t 60 -i 10 -R

TCP test with 4 Client Streams; Reverse mode: server sends, clients receive
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -P 4 -t 60 -i 10 -R

TCP test with 8 Client Streams; Reverse mode: server sends, clients receive
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -P 8 -t 60 -i 10 -R

Bidirectional TCP test
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -t 60 -i 10 --bidir

Bidirectional TCP test with 4 Client Streams
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -P 4 -t 60 -i 10 --bidir

Bidirectional TCP test with 8 Client Streams
iperf3 -c <IP of Iperf Server> -p <Port of Iperf Server> -P 8 -t 60 -i 10 --bidir

ThiloteE, 2023-01-13:
> 2.4ghz WIFI (MT7603E) is like there was a time that it feels like it's freezing or hanging for like 30secs or 1minute and during that moment, I can't browse to the internet or even ping my router.

1. Have you tried OpenWRT19.07? In [this comment](https://github.com/openwrt/mt76/issues/719#issuecomment-1371335067) some users report issues with mt7603E with newer versions in #719

2. Try this and report back, if it solved your issue:
   > I use snapshot image from 9 january 2023. OpenWrt SNAPSHOT r21728-fc33c41c21 / LuCI Master git-22.361.69865-deed682
    > 
    > And have the same problem. (on this setup, **2ghz, n, ht20, 6 channel**)
    > 
    > I do two configs below and now, 12 hours later, i dont have any disconnection. (will continue testing)
    > 
    > **1 - Disable multicast on phy0-ap0 interface** To do this go to luci and interfaces > devices > phy0-ap0
    > 
    > **2 - Change the wpa2 wifi criptography cipher from [auto] to [aes]**
    > 
    > Ps: This bug dont show any warning, notices or information on log system. i dont know if this bug is a criptography key problem (Tkip or auto switch) or multicast flood on wireless clients.

shown19, 2023-06-21:
Hi sorry all for the very late response, right now I still have the device Newifi-D2 with mt76 chipset and wifi mt7603e but wifi has been disabled for months and just using this only for routing management and so far it's running great except for the WIFI still. Until I got into this thread: https://github.com/openwrt/mt76/issues/450 so my interest is back again and wanted to try my luck if I could be able to achieve better result this time. I am learning how to compile and patch but yes I'm still newbie yet I'm gonna try this method given in the link which is Disabling AMSDU, will update if there's an improvement.

I also got into this another thread: https://github.com/openwrt/mt76/issues/719 and been wondering what if the solution is to enable SMPS but with refine code to work with the MT7603E? I know my knowledge is limited so I don't know if I'm right or it's worth a try.

P.S. I already tried the latest stable version which is 22.03.5, also 23.05.0-rc1 and even snapshots but that doesn't satisfied me. I'm still experiencing hiccups from time to time especially the 2.4ghz, so I'm gonna try the AMSDU disabled patch which I don't know if it's related to the issue or what but let see.

shown19, 2023-06-22:
Update, I compiled the firmware using the latest snapshot build with amsdu disabled patch but sadly I'm still experiencing hiccups and freezing towards the 2.4ghz from time to time so this solution isn't related to the issue after all. I also think that maybe SMPS enabled would help the issue? SMPS is currently disabled on my device but just my guess and I can't do anything about it anymore since my knowledge is limited.

shown19, 2023-06-22:
I'm also believing more now that this might be a hardware limitation or might be faulty since I already tried to flash it with different firmwares like Padavan, Wave-ng, GoCloud, Padawan but same issue experienced.